realsenseai/librealsense #13947: Depth image effect acquired by the D405 depth camera in a dark environment

Tianhao-Wu7, 2025-04-17:
I get poor depth images in dark environments with the D405 depth camera (compared to bright ambient light during the day). The image is a depth image acquired in a dark environment with the D405 depth camera 0.2m from the target.

[image]

MartyG-RealSense, 2025-04-17:
Hi @Tianhao-Wu7  The D405 camera model, due to its compact size, is not equipped with an infrared light emitter component that larger models in the 400 Series have and so unlike those models (D415, D435, D455, etc), the D405 cannot see in the dark.

400 Series cameras can make use of any visible light in the scene to aid the processing of depth though, so if you can place a miniature portable light near to the D405 (perhaps tape-attach a little light to the top of the camera) then your depth image may improve.

Tianhao-Wu7, 2025-04-17:
> Hi [@Tianhao-Wu7](https://github.com/Tianhao-Wu7) The D405 camera model, due to its compact size, is not equipped with an infrared light emitter component that larger models in the 400 Series have and so unlike those models (D415, D435, D455, etc), the D405 cannot see in the dark.
> 
> 400 Series cameras can make use of any visible light in the scene to aid the processing of depth though, so if you can place a miniature portable light near to the D405 (perhaps tape-attach a little light such as a pen light or pen torch to the top of the camera) then your depth image may improve.

Thank you for your reply. Can I ask if I can improve the acquired depth image by installing an IR light emitter device on top of the D405 depth camera.

MartyG-RealSense, 2025-04-17:
I do not have a dark room to test in, but I placed a D405 in my shoe and used a bright hand-torch placed behind the camera to shine light inside the shoe.  There was a noticeable improvement in the depth image when the torch light was present in the dark environment of the inside of the shoe.

**WITHOUT TORCH**

[image]

**WITH TORCH**

[image]

Tianhao-Wu7, 2025-04-17:
> 我没有暗室进行测试，但我在鞋子里放了一台 D405，并使用放置在相机后面的明亮手电筒将光线照射到鞋子内部。当手电筒光出现在鞋内侧的黑暗环境中时，深度图像有明显的改善。
> 
> 是的，RealSense 摄像头还可以利用 IR 灯发出的近乎不可见的光。
> 
> **无手电筒**
> 
> [image]
> 
> **带 TORCH**
> 
> [image]

How should I integrate the IR emitter device with the D405? Is it sufficient to illuminate it as a secondary light source as you have shown?

MartyG-RealSense, 2025-04-17:
It does not need to be integrated.  It can be an external light source.

If you wanted to build your own fully integrated device that incorporates a lamp then you would need to obtain a D401 Depth Module board, a Vision Processor D4 V4 board, and perhaps also construct your own custom casing for the components such as a 3D-printed one.

Tianhao-Wu7, 2025-04-17:
> 它不需要集成。它可以是外部光源。
> 
> 如果您想构建自己的完全集成的定制 D405 设备，其中包含微型红外灯，那么您需要获得 D401 深度模块板、视觉处理器 D4 V4 板，也许还需要为组件（如 3D 打印）构建自己的定制外壳。

Okay, thank you.

closed 2025-04-21 by Tianhao-Wu7