realsenseai/librealsense #13948: D457 RGB camera TIMESTAMP issue

Niel186, 2025-04-17:
What is the difference between the following four timestamps: SENSOR_TIMESTAMP, FRAME_TIMESTAMP, BACKEND_TIMESTAMP, TIME_OF_ARRIVAL.
The actual data I saw using realsesence-view on the local Jetson Orin device is as follows.
According to (https://github.com/IntelRealSense/librealsense/issues/2188), it should satisfy:
Sensor_timestamp << Frame_timestamp
Backend_timestamp << Time_of_arrival
But in reality Backend_timestamp > Time_of_arrival, so I don't know why

[image]

Intel RealSense D457 s.n.:23542200544 | RGB Camera, Color stream
1.Frame Timestamp : 1744872747174.5
2.Clock Domain : Global Time
3.Frame Number : 736
4.Hardware Size : 640 x 480
5.Display Size : 640 x 480
6.Pixel Format : RGB8
7.Hardware FPS : 29.04
8.Viewer FPS : 29.78

Frame Counter : 736

1.Frame Timestamp : 2775115080
2.Sensor Timestamp : 2775080745
3.Actual Exposure : 1660
4.Auto Exposure : 8
5.Time Of Arrival : 1744872747201
6.Backend Timestamp : 1744872773814

MartyG-RealSense, 2025-04-17:
Hi @Niel186  As your camera name is listed as D457 on the top of the stream panel in the above image, does this mean that your camera is on a GMSL / Fakra cable connection instead of USB, please?  If a D457 is on a USB cable connection then it will be named D455 instead.

If you are on a GMSL connection then I can understand why there might be a difference from the expected behaviour for the backend timestamp, since the definition for this timestamp at https://github.com/IntelRealSense/librealsense/issues/2188#issuecomment-409985803 pre-dates the D457's release by years and relates to the **USB** controller.  But the USB controller would not be being used on a GMSL connection.

`Host clock applied to kernel-space buffers (URB) when the data from USB controller is copied to OS kernel. (HW->Kernel transition)`

https://github.com/IntelRealSense/librealsense/issues/2188 is the best available reference for the differences between the different timestamp types.  You might also find a discussion about this reference at https://github.com/IntelRealSense/librealsense/issues/12779 to be helpful.

Niel186, 2025-04-18:
> Hi [@Niel186](https://github.com/Niel186) As your camera name is listed as D457 on the top of the stream panel in the above image, does this mean that your camera is on a GMSL / Fakra cable connection instead of USB, please? If a D457 is on a USB cable connection then it will be named D455 instead.
> 
> If you are on a GMSL connection then I can understand why there might be a difference from the expected behaviour for the backend timestamp, since the definition for this timestamp at [#2188 (comment)](https://github.com/IntelRealSense/librealsense/issues/2188#issuecomment-409985803) pre-dates the D457's release by years and relates to the **USB** controller. But the USB controller would not be being used on a GMSL connection.
> 
> `Host clock applied to kernel-space buffers (URB) when the data from USB controller is copied to OS kernel. (HW->Kernel transition)`
> 
> [#2188](https://github.com/IntelRealSense/librealsense/issues/2188) is the best available reference for the differences between the different timestamp types. You might also find a discussion about this reference at [#12779](https://github.com/IntelRealSense/librealsense/issues/12779) to be helpful.

Thanks for your reply. I use GMSL data cable connection. In this connection mode, how are the values ​​of Sensor_timestamp, Frame_timestamp
Backend_timestamp, Time_of_arrival different from those when connected via USB? What is the expected performance under GMSL? Thanks

MartyG-RealSense, 2025-04-22:
Hi @Niel186  Thanks very much for your patience.  My Intel RealSense colleagues have provided advice about this issue.

The backend timestamp comes from the UVC driver when using a USB cable connection, and from the Jetson VI driver when using a GMSL (MIPI) connection on Nvidia Jetson.

The issue of time of arrival < backend (when it should be backend < time of arrival) is a known issue to the RealSense team that is under investigation.

Niel186, 2025-04-22:
[image]

Niel186, 2025-04-22:
[image]

Thank you for your reply. In addition, you can see that Frame Timestamp and Sensor Timestamp are in line with expectations. I would like to know how the two Timestamps are calculated and set in the RealSense SDK, and how to obtain them (code snippets for writing and reading).
Looking forward to your reply, thank you very much!

MartyG-RealSense, 2025-04-22:
The official documentation about metadata timestamps can be found at the link below.

https://dev.intelrealsense.com/docs/frame-metadata

Timestamps can be retrieved using the **get_timestamp()** instruction, as demonstrated by a Python script at https://github.com/IntelRealSense/librealsense/issues/12779 and a C++ one at https://github.com/IntelRealSense/librealsense/issues/4702

https://github.com/IntelRealSense/librealsense/issues/4525 has further information about get_timestamp().

Timestamps are generated automatically, so you do not need to write them.  If support for **hardware metadata** is enabled in the RealSense SDK then 'hardware timestamps' will be provided.  If hardware metadata support is not enabled then some types of timestamp will be unavailable, and 'software timestamps' that are calculated based on the computer's clock are provided instead.

Metadata support is enabled automatically on Linux when installing the SDK from packages or when building it from source code with the libuvc backend = true or RSUSB backend = true installation methods.  Source code builds where libuvc or RSUSB backend is not true need to have a kernel patch script applied to the kernel to enable metadata support.

On Windows, metadata support can be enabled simply by clicking a button on a pop-up message in the RealSense Viewer tool that appears when the Viewer launches.

Niel186, 2025-04-22:
Thank you for your detailed introduction. I am more interested in the process of Timestamps automatic generation.
It can be simply understood as follows:
The timestamp is generated by the camera driver (UVC driver or VI driver), then realSenceSDK obtains it through the V4L2 architecture, and finally realSenceSDK writes it into the corresponding metadata for the user application to call.
Is my simple understanding of this process correct?

closed 2025-05-11 by MartyG-RealSense