realsenseai/librealsense #13950: L515 depth image all black and sometimes repeated

RainingJoe, 2025-04-18:
- Question:
    Hello, I am using L515 to do some robot-related development. I encountered two difficult problems related to the depth image. The first problem is that the depth image sometimes does not update, and the other problem is that the depth map is occasionally black, that is, the max depth distance value is 0.

[image]

[image]

- Required Info
   Camera Model -- L515
   Firmware Version -- 1.5.8.1
   Operation System -- Win 11
   Platform  -- PC
   SDK Version -- 2.50.0
   Language -- C++
   

- Laboratory environment
    Our laboratory has Optitrack Motive installed. When the problem occurs, Optitrack Motive is working . I wonder if it is possible that Optitrack Motive affects Realsense L515?

- Test code
-`#include <librealsense2/rs.hpp>
#include <iostream>
#include <cstdlib>
#include <vector>
#include <thread>
#include <map>
#include <unordered_map>
#include <string>
#include <zmq.hpp>
#include <chrono>
#include <mutex>
#include <queue>
#include <condition_variable>

#define REAL_MODE 1
#define DEBUG_MODE 0

template <typename T>
class SyncQueue {
public:
    void Push(const T& value) {
        {
            std::lock_guard<std::mutex> lock(mutex_);
            queue_.push(value);
        }
        cond_.notify_one(); 
    }

    T Pop() {
        std::unique_lock<std::mutex> lock(mutex_);
        cond_.wait(lock, [this] { return !queue_.empty(); });
        T val = queue_.front();
        queue_.pop();
        return val;
    }

    bool TryPop(T& value) {
        std::lock_guard<std::mutex> lock(mutex_);
        if (queue_.empty()) return false;
        value = queue_.front();
        queue_.pop();
        return true;
    }

    bool Empty() const {
        std::lock_guard<std::mutex> lock(mutex_);
        return queue_.empty();
    }

    size_t Size() const {
        std::lock_guard<std::mutex> lock(mutex_);
        return queue_.size();
    }

private:
    std::queue<T> queue_;
    mutable std::mutex mutex_;
    std::condition_variable cond_;
};

void RecvCamera(rs2::pipeline pipe, std::vector<std::string> topic_names, std::unordered_map<std::string, std::shared_ptr<SyncQueue<std::vector<uint8_t>>>>& topic_queues) {
    rs2::align align(RS2_STREAM_COLOR);
    auto depth_start = std::chrono::high_resolution_clock::now();
    auto depth_end = depth_start;
    auto color_start = std::chrono::high_resolution_clock::now();
    auto color_end = depth_start;
    std::vector<uint8_t> pre_depth_data;
    int repeat_cnt = 0;
    while (true) {
        rs2::frameset frames = pipe.wait_for_frames();
        auto aligned_frames = align.process(frames);

        auto depth_frame = aligned_frames.get_depth_frame();
        auto color_frame = aligned_frames.get_color_frame();
        if (!depth_frame || !color_frame) continue;

        // Depth
        int depth_width = depth_frame.get_width();
        int depth_height = depth_frame.get_height();
        const uint8_t* depth_ptr = reinterpret_cast<const uint8_t*>(depth_frame.get_data());
        std::vector<uint8_t> depth_data(depth_ptr, depth_ptr + depth_width * depth_height * 2);
        auto max_it = std::max_element(depth_data.begin(), depth_data.end());
        auto min_it = std::min_element(depth_data.begin(), depth_data.end());
        if (*max_it == 0) {
            std::cout << "black pic occur" << std::endl;
        }
        if (depth_data == pre_depth_data) {
            repeat_cnt++;
            if (repeat_cnt == 10)
                std::cout << "depth data not update" << std::endl;
            else {
                repeat_cnt = 0;
            }
        }
        pre_depth_data = depth_data;
        // Color
        int color_width = color_frame.get_width();
        int color_height = color_frame.get_height();
        const uint8_t* color_ptr = reinterpret_cast<const uint8_t*>(color_frame.get_data());
        std::vector<uint8_t> color_data(color_ptr, color_ptr + color_width * color_height * 3);
        topic_queues[topic_names[0]]->Push(depth_data);
        topic_queues[topic_names[1]]->Push(color_data);
    }
}

void SendData(zmq::context_t& context, std::string& topic_name, std::string port, std::shared_ptr<SyncQueue<std::vector<uint8_t>>>& que) {
    zmq::socket_t publisher(context, zmq::socket_type::pub);
    publisher.bind("tcp://192.168.10.16:" + port);
    std::cout << "ZMQ Publisher started on tcp://192.168.10.16:" + port << std::endl;
    auto start = std::chrono::high_resolution_clock::now();
    auto end = start;
    while (true) {
        auto payload = que->Pop();
        std::vector<uint8_t> full_data(payload.begin(), payload.end());
        zmq::message_t message(full_data.data(), full_data.size());
        try {
            publisher.send(message, zmq::send_flags::none);
            //std::cout << "ZMQ Send success payload size = " << payload.size() << std::endl;

            end = std::chrono::high_resolution_clock::now();
            auto duration = std::chrono::duration_cast<std::chrono::milliseconds>(end - start);
            //std::cout << "Time elapsed: " << duration.count() << " ms" << std::endl;
            start = end;
        }
        catch (const zmq::error_t& e) {
            std::cerr << "ZMQ Send failed: " << e.what() << std::endl;
            continue;
        }
        continue;
    }
}

int main(int argc, char* argv[]) try {
    std::unordered_map<std::string, std::string> camera_config = {
        {"f1421915", "down"},
        {"f1382268", "up"},
        {"f1422477", "left"},
        {"f1420637", "right"}
    };
    std::unordered_map<std::string, std::string> port_map = {
        {"/camera_realsense_left/depth_image", "5555"},
        {"/camera_realsense_left/color_image", "5556"},
        {"/camera_realsense_right/depth_image", "5557"},
        {"/camera_realsense_right/color_image", "5558"},
        {"/camera_realsense_up/depth_image", "5559"},
        {"/camera_realsense_up/color_image", "5560"},
        {"/camera_realsense_down/depth_image", "5561"},
        {"/camera_realsense_down/color_image", "5562"}
    };
    std::vector<std::string> topic_list = {
        "/camera_realsense_left/depth_image",
        "/camera_realsense_left/color_image",
        "/camera_realsense_right/depth_image",
        "/camera_realsense_right/color_image",
        "/camera_realsense_up/depth_image",
        "/camera_realsense_up/color_image",
        "/camera_realsense_down/depth_image",
        "/camera_realsense_down/color_image"
    };
    std::unordered_map<std::string, std::shared_ptr<SyncQueue<std::vector<uint8_t>>>> topic_queues;
    for (auto& topic : topic_list) {
        topic_queues[topic] = std::make_shared<SyncQueue<std::vector<uint8_t>>>();
    }
#if REAL_MODE
    rs2::context ctx;
    auto devices = ctx.query_devices();
    if (!devices.size()) {
        std::cerr << "No device available!" << std::endl;
        return EXIT_FAILURE;
    }

    std::vector<std::thread> threads;
    zmq::context_t context(1);

    for (auto&& device : devices) {
        try {
            std::string sn = device.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER);
            std::cout << "after sn: " << sn << std::endl;
            std::string name = camera_config[sn];
            std::string port = port_map[sn];
            rs2::pipeline pipe;
            rs2::config cfg;
            cfg.enable_device(sn);
            cfg.enable_stream(RS2_STREAM_COLOR, 1280, 720, RS2_FORMAT_BGR8, 30);
            cfg.enable_stream(RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_Z16, 30);
            pipe.start(cfg);
            std::cout << "after config: " << name << std::endl;
            std::vector<std::string> topic_names = {
                "/camera_realsense_" + name + "/depth_image",
                "/camera_realsense_" + name + "/color_image"
            };
            std::cout << "after hash: " << topic_names[0] << std::endl;
            threads.emplace_back(RecvCamera, std::move(pipe), topic_names, topic_queues);
            std::cout << "Started pipeline for device: " << sn << std::endl;
        }
        catch (const std::exception& e) {
            std::cerr << "Device init error: " << e.what() << std::endl;
        }
    }
    for (auto topic : topic_list) {
        threads.emplace_back(SendData, std::ref(context), topic, port_map[topic], topic_queues[topic]);
    }

    for (auto& t : threads) {
        if (t.joinable()) t.join();
    }
#endif

    return EXIT_SUCCESS;
}
catch (const rs2::error& e) {
    std::cerr << "RealSense error: " << e.what() << std::endl;
    return EXIT_FAILURE;
}
catch (const std::exception& e) {
    std::cerr << e.what() << std::endl;
    return EXIT_FAILURE;
}`

MartyG-RealSense, 2025-04-18:
Hi @RainingJoe  Optitrack Motive apparently makes use of infrared cameras.  The L515 camera model is sensitive to interference from external IR, so if the problem only seems to occur when Optitrack Motive is being used then it is possible that OM's infrared output is blinding the L515's vision.  

Using the L515's 'Low Ambient Light' camera configuration preset can take account of the presence of IR light, so you could try setting this preset in C++ code to see whether the black images still occur.

```
auto sensor = profile.get_device().first<rs2::depth_sensor>();
sensor.set_option(rs2_option::RS2_OPTION_VISUAL_PRESET,
rs2_l500_visual_preset::RS2_L500_VISUAL_PRESET_LOW_AMBIENT_LIGHT);
```

RainingJoe, 2025-04-19:
Ok, I will try it

RainingJoe, 2025-04-21:
Now my code has been set up like this, but black image still occur. It really only appears when motive is turned on. Is there any other way to avoid the black image problem here?

[image]

MartyG-RealSense, 2025-04-21:
Thanks very much for the update.

You could confirm whether or not IR interference is an issue by turning the L515 to face in the opposite direction so that it does not see the Motive's IR emissions.

Since depth data also does not update sometimes when the Motive is active, if both the L515 and Motive are attached to the same computer then you could investigate whether having two cameras active simultaneously is burdening the computer's available resources.

As you appear to be on a Windows computer, you could test this by using the Ctrl-Alt-Delete keyboard shortcut when both cameras are active and selecting the 'Task Manager' option.  When the Task Manager window pops up, you might need to select the 'Processes' tab if you are on Windows 10.  The Processes information should be shown by default on Windows 11.

[image]

If you look at the 'CPU' readout then you can see whether it has a high percentage value when both cameras are active, which could indicate that the computer is struggling with handling two cameras simultaneously.  An over-burdened CPU could result in glitches such as missed frames.

RainingJoe, 2025-04-25:
Hello~ MartyG, our motive device and L515 camera are not on the same computer. We use only one L515 camera instead. The resource consumption of L515 is low. Now when collecting data, we try to make the camera's angle of view face downward, not facing the motive camera above our head. The frequency of black images is obviously reduced compared to before, but it may still appear if the stress test is carried out for a long time. I wonder if there are other possible solutions?

MartyG-RealSense, 2025-04-25:
If you ran one of the SDK example programs for a while instead of your own RealSense script then you could observe whether or not the SDK example also experiences black frames after a while. If it does not then it could indicate that there is a problem in your script, such as instability introduced by a memory leak (the progressive consumption of the computer's available memory capacity over a period of time).