realsenseai/librealsense #13953: Problems using the SDK for the python programming language

Tianhao-Wu7, 2025-04-20:
1. How to set depth units in python's SDK?
2. How to use post-processing filters in python's SDK?
3. How to call histogram equalization in python's SDK?
4. How to call already stored presets in python's SDK?

MartyG-RealSense, 2025-04-20:
Hi @Tianhao-Wu7  Thanks very much for your questions.  Provided below are some examples of Python references for the subjects that you asked about.

1.  Changing depth units: https://github.com/IntelRealSense/librealsense/issues/10976#issuecomment-1271236193

2.  Using post-processing filters: https://github.com/IntelRealSense/librealsense/issues/11246

An official Intel post-processing tutorial for Python can also be found here:

https://github.com/IntelRealSense/librealsense/blob/jupyter/notebooks/depth_filters.ipynb

With post-processing, you first define the filter in your script and then apply it to the data with a **.process()** instruction.

3.  Using histogram equalization: https://github.com/IntelRealSense/librealsense/issues/7089#issuecomment-672105974

Set the **histogram_equalization_enabled** to 1.0 instead of 0.0 to enable equalization.

https://github.com/IntelRealSense/librealsense/issues/3363 also has information about setting the color scheme for depth data colorization in Python if you wish to do so.

4.  Calling the SDK's in-built visual presets: https://github.com/IntelRealSense/librealsense/issues/10014

```
 HIGH_ACCURACY = 3
 HIGH_DENSITY = 4
 MEDIUM_DENSITY = 5
 self.profile.get_device().sensors[0].set_option(rs.option.visual_preset, HIGH_DENSITY)
```

Tianhao-Wu7, 2025-04-21:
When the depth unit setting is not performed, the D405 prints the acquired depth unit of 0.0001, and after the depth unit setting is performed, the print acquires the depth unit of 0.01. The setting process is shown in the code. When I set the depth unit to 1 in Intel Realsense Viewer, the depth unit obtained by python printing is 0.01.

    print(f"depth_units 1", depth_sensor.get_depth_scale())
    depth_sensor.set_option(rs.option.depth_units, 0.01)
    print(f"depth_units 2", depth_sensor.get_depth_scale())

Tianhao-Wu7, 2025-04-21:
How can I use my own visual preset json file in python?
like this？
    preset = r"F:\\HumanEar\\librealsense-master\\earlc_preset\\Earhole_local.json"
    depth_sensor.set_option(rs.option.visual_preset, preset)

MartyG-RealSense, 2025-04-21:
The depth unit scale of the D405 model is already 0.01 by default (the other models in the 400 Series camera range have a default of 0.001) and so a value of 0.01 does not need to be set with scripting.  D405 has a different scale because it is designed for very close range depth sensing.

On other 400 Series models, '1' is considered an invalid "out of bounds" value, with '0.1' being the largest value permitted.  So on D405, inputting '1' may be causing that too-large value to be rejected and the D405's default of 0.01 to be automatically set instead.

*************

It is not possible to load your own json preset using `rs.option.visual_preset` as this only works for the built-in presets.  Instead, a custom json can be loaded by using a **string** to import the file.  This method is more complex and prone to problems than the one for loading the built-in presets.  There is a Python discussion about loading custom jsons at https://github.com/IntelRealSense/librealsense/issues/10376

closed 2025-04-21 by Tianhao-Wu7