realsenseai/librealsense #13963: Inconsistency when using the depth and color python package D435i

stevedanomodolor, 2025-04-24:
I will using the align-depth2color.py python example because it happens there also. I modified the example just to extract color, and i noticed that when i set the rs config for only color image, and extract only color image, i noticed that the fps is not consistent, what i mean is that it drops really low every x seconds. When this happens, I am using less cpu(i will give more context on why this is important)

The strange thing, when i also set the config for depth sensor and just extract color information, so basically call the following:

wait for frames
get color
np array---- color get data

its becomes stables and uses more cpu. I am missing something?

MartyG-RealSense, 2025-04-24:
Hi @stevedanomodolor  Does performance of align_depth2color.py improve if you disable the RGB option **Auto-Exposure Priority** using the Python code below somewhere after the pipe start line?

```
profile = pipeline.start(config)

profile.get_device().query_sensors()[1].set_option(rs.option.auto_exposure_priority, 0.0)
```

This query_sensors() instruction sets the sensor to an index number of '1' so that it accesses the RGB sensor instead of the depth sensor (which is index '0') so that the **auto_exposure_priority** setting can be set to 0 (disabled).

When auto-exposure is enabled and auto-exposure priority is disabled, the RealSense SDK will try to force the depth and color FPS to be constant instead of permitting the FPS of one of the streams to fall.

stevedanomodolor, 2025-04-24:
It doesn't improve.[image]

MartyG-RealSense, 2025-04-24:
Thank you for testing.

Which computer / computing board are you running the program on, please?  For example, is it a PC desktop / laptop or a Raspberry Pi single-board computer?

stevedanomodolor, 2025-04-24:
> Thank you for testing.
> 
> Which computer / computing board are you running the program on, please?  For example, is it a PC desktop / laptop or a Raspberry Pi single-board computer?

The weird things is that when i configure the depth too, it is stable and uses all possible core. 

Another thing that i am not sure if its related but would like ask is why when i have a python thread that does the data retrieval and when i call get data from the main program i collect the data stored reduces the fps by half. 

I send a Code example of this

MartyG-RealSense, 2025-04-24:
I am not able to run and test the code of RealSense users on my own computer, unfortunately.

It would make sense that CPU usage would be lower when FPS drops (perhaps because frames are being skipped), because a lower FPS means that there is less processing burden on the CPU.  When FPS is high, CPU % usage will also rise because the CPU has more work to do because of the higher number of frames being produced.

stevedanomodolor, 2025-04-24:
python thread grabbing data and it stored in a dictionary. The loop itself goes at the requires fps but when i grab the data it lowers the fps
```
frames = []
            while self.is_alive:
                raw_frames = self.camera.wait_for_frames()

                if self.retrieve and len(frames) < self.max_frames:
                    frames.append(align(raw_frames))

                elif len(frames) == self.max_frames:
                    self.retrieve = False

                    # Get color frame
                    color_frame = frames[-1].get_color_frame()

                    # Get texture and set points
                    texture = np.asanyarray(color_frame.get_data())
                    .....
```

stevedanomodolor, 2025-04-24:
forget about the second issue, it is my fault. I am still having the issue with the fps drop as i mentioned initially.

MartyG-RealSense, 2025-04-24:
I note that you are using the **append()** instruction in your script.  I would recommend avoiding using append if possible, as in a RealSense Python script it often causes the program to stop receiving frames after only the first 15 frames.  If you do need to use append then a workaround is to store frames in memory with the Keep() instruction.  This can be be done simply by inserting the Keep() instruction immediately after the wait_for_frames() line.

```
raw_frames = self.camera.wait_for_frames()
raw_frames.keep()
```

stevedanomodolor, 2025-04-24:
> I am not able to run and test the code of RealSense users on my own computer, unfortunately.
>
> It would make sense that CPU usage would be lower when FPS drops (perhaps because frames are being [...]

The second issue is not a problem anymore.
The issue with the frame drop when not setting the depth config is still there even after disabling the exposure you mentioned.

stevedanomodolor, 2025-04-24:
The following are images of the plot of the fps per loop to get frames
Without config depth
[image]

Zoom on the dropes

[image]

CPU 
[image]

With config depth
No frame drops

[image]

CPU [image]

Btw do i have to put a loop rate limiter or something?

The loop 

[image]

MartyG-RealSense, 2025-04-24:
You should not need to implement any restrictions on the loop.  And you have a PC with a good hardware specification.

If you manage to achieve a working solution through a method that does not entirely make logical sense then it can be worth just going with it if the workaround is not seriously affecting performance.

Having said that, if you are experiencing worse performance when not specifying a config, I wonder if it is because without a cfg instruction for the streams, the program will automatically apply 848x480 resolution at 30 FPS for depth and 1280x720 at 30 FPS for color, instead of 640x480 at 30 FPS for both.

stevedanomodolor, 2025-04-24:
> I note that you are using the **append** instruction in your script.  I would recommend avoiding using append if possible, as in a RealSense Python script it often causes the program to stop [...]

I will try this tomorrow and tell you if it improves the frame rate.

stevedanomodolor, 2025-04-24:
> You should not need to implement any restrictions on the loop. And you have a PC with a good hardware specification.
>
> If you manage to achieve a working solution through a method that does not [...]

Is it possible to disable the depth stream entire?

MartyG-RealSense, 2025-04-24:
Yes, you can create RGB-only RealSense programs by only having a color cfg line and not a depth one too, so that the program will only enable the color stream when run.

But as **align_depth2color.py** is a depth to color alignment script, disabling depth would break that particular program.

stevedanomodolor, 2025-04-24:
> Yes, you can create RGB-only RealSense programs by only having a color cfg line and not a depth one too, so that the program will only enable the color stream when run.
> 
> But as **align_depth2color.py** is a depth to color alignment script, disabling depth would break that particular program.

I mean just using the cfg for color because for me, as i said settings
 only cfg causes the frame drop. Do you have any example that just sets color that i can test that could help me reduce the frame drop without setting the depth crg.

stevedanomodolor, 2025-04-24:
Maybe a python version of https://stackoverflow.com/questions/62056622/how-to-turn-off-depth-sensor-completely-in-realsense-d435

stevedanomodolor, 2025-04-25:
> Maybe a python version of https://stackoverflow.com/questions/62056622/how-to-turn-off-depth-sensor-completely-in-realsense-d435

The disabling of the depth stream does not work.

MartyG-RealSense, 2025-04-25:
You cannot set a custom configuration for one stream type and not another that you also plan to enable, unfortunately.  If you have a cfg line only for color then the depth stream will not be enabled.

You could try having a depth cfg line but setting its resolution and FPS to a very low configuration so that it imposes hardly any burden on the computer's CPU.  Your D435i should support 424x240 at 6 FPS.

stevedanomodolor, 2025-04-25:
> You cannot set a custom configuration for one stream type and not another that you also plan to enable, unfortunately. If you have a cfg line only for color then the depth stream will not be [...]

this seem to work for now. If you do come up with a solution that doesnt require having depth , let me know but thank you for the support.

MartyG-RealSense, 2025-04-25:
You are very welcome.  I'm pleased that I could help!

MartyG-RealSense, 2025-05-01:
Hi @stevedanomodolor  Do you require further assistance with this case at the moment, please?  Thanks!

stevedanomodolor, 2025-05-01:
No, the workaround works.

closed 2025-05-01 by stevedanomodolor