realsenseai/librealsense #13964: how to change resolution quickly?

ctxqlxs, 2025-04-28:
When using Intel RealSense SDK (librealsense) to switch resolutions dynamically by stopping and restarting the pipeline (pipe.stop() and pipe.start(cfg)), the pipe.stop() call takes several seconds to complete. 

This issue impacts applications that require frequent resolution changes, as the delay can degrade user experience and application responsiveness.

**Steps to Reproduce**:
Start a pipeline with an initial resolution (e.g., 640x480).
Stop the pipeline using pipe.stop().
Reconfigure the pipeline with a new resolution (e.g., 1280x720).
Restart the pipeline using pipe.start(cfg).
**Expected Behavior**:
The pipe.stop() call should complete quickly (within milliseconds) to allow for fast resolution switching.
**Actual Behavior**:
The pipe.stop() call takes several seconds to complete, causing a noticeable delay before the pipeline can be restarted with the new configuration.

MartyG-RealSense, 2025-04-28:
Hi @ctxqlxs  Does the delay still occur if you comment out the sleep line (which puts the CPU to sleep for 5 seconds):

`std::this_thread::sleep_for(std::chrono::seconds(5));`

The impression I get from the script is that the problem is not that Stop() is taking several seconds to occur, but that once the pipeline is started and the CPU is put to sleep for 5 seconds, the script is prevented from continuing on to the script line with the Stop() command until the time has elapsed.

ctxqlxs, 2025-04-29:
worse result！
[image]
**Result**
pipe stop time: 3185ms

ctxqlxs, 2025-04-29:
**change the code**:
```
 for (int i = 0; i < 10; ++i) {
    rs2::pipeline pipe;
    rs2::config cfg;

    // 初始配置：640x480, 30 FPS
    cfg.enable_stream(RS2_STREAM_COLOR, 1280, 720, RS2_FORMAT_BGR8, 30);
    cfg.enable_stream(RS2_STREAM_DEPTH, 1280, 720, RS2_FORMAT_Z16, 30);

    // 启动管道
    pipe.start(cfg);
//  std::this_thread::sleep_for(std::chrono::seconds(5));
    auto start = std::chrono::high_resolution_clock::now();
    pipe.stop();
    auto end = std::chrono::high_resolution_clock::now();
    long long  elapsed = std::chrono::duration_cast<std::chrono::milliseconds>(end - start).count();
    std::cout<< i << "th|  pipe stop time: " << elapsed << " ms" << std::endl;

    rs2::config new_cfg;
    new_cfg.enable_stream(RS2_STREAM_COLOR, 424, 240, RS2_FORMAT_BGR8, 90);
    new_cfg.enable_stream(RS2_STREAM_DEPTH, 424, 240, RS2_FORMAT_Z16, 90);

    pipe.start(new_cfg);
//  std::this_thread::sleep_for(std::chrono::seconds(5));

    // 停止管道
    pipe.stop();
  }
```
**Result**:
```
0th|  pipe stop time: 3020 ms
1th|  pipe stop time: 3153 ms
2th|  pipe stop time: 4119 ms
3th|  pipe stop time: 3220 ms
4th|  pipe stop time: 3052 ms
5th|  pipe stop time: 3052 ms
6th|  pipe stop time: 3151 ms
7th|  pipe stop time: 3153 ms
8th|  pipe stop time: 3053 ms
9th|  pipe stop time: 2986 ms
```

ctxqlxs, 2025-04-30:
**without any  config(rs2::config), code:**
`auto start = std::chrono::high_resolution_clock::now();
    rs2::pipeline pipe;
    pipe.start();
    pipe.stop();

    auto end = std::chrono::high_resolution_clock::now();
    long long  elapsed = std::chrono::duration_cast<std::chrono::milliseconds>(end - start).count();
    std::cout<< i << "th|  pipe stop time: " << elapsed << " ms" << std::endl;`
**results**
`0th|  pipe stop time: 4581 ms
1th|  pipe stop time: 4425 ms
2th|  pipe stop time: 4422 ms
3th|  pipe stop time: 4460 ms
4th|  pipe stop time: 4439 ms
5th|  pipe stop time: 4450 ms
6th|  pipe stop time: 4459 ms
7th|  pipe stop time: 4372 ms
8th|  pipe stop time: 4390 ms
9th|  pipe stop time: 4444 ms`

**without any  config(rs2::config), exclude start. code:**
`    rs2::pipeline pipe;
    pipe.start();

    auto start = std::chrono::high_resolution_clock::now();
    pipe.stop();
    auto end = std::chrono::high_resolution_clock::now();
    long long  elapsed = std::chrono::duration_cast<std::chrono::milliseconds>(end - start).count();
    std::cout<< i << "th|  pipe stop time: " << elapsed << " ms" << std::endl;`
**results**
`0th|  pipe stop time: 3771 ms
1th|  pipe stop time: 3816 ms
2th|  pipe stop time: 3816 ms
3th|  pipe stop time: 3783 ms
4th|  pipe stop time: 3775 ms
5th|  pipe stop time: 3752 ms
6th|  pipe stop time: 3769 ms
7th|  pipe stop time: 3773 ms
8th|  pipe stop time: 3810 ms
9th|  pipe stop time: 3853 ms`

**but if we enable rgb stream,code**:
`rs2::config cfg;

    cfg.enable_stream(RS2_STREAM_COLOR, 1280, 720, RS2_FORMAT_BGR8, 30);
//    cfg.enable_stream(RS2_STREAM_DEPTH, 1280, 720, RS2_FORMAT_Z16, 30);

//  std::this_thread::sleep_for(std::chrono::seconds(5));
    rs2::pipeline pipe;
    pipe.start(cfg);

    auto start = std::chrono::high_resolution_clock::now();
    pipe.stop();
    auto end = std::chrono::high_resolution_clock::now();
    long long  elapsed = std::chrono::duration_cast<std::chrono::milliseconds>(end - start).count();
    std::cout<< i << "th|  pipe stop time: " << elapsed << " ms" << std::endl;`

**results**:
`0th|  pipe stop time: 945 ms
1th|  pipe stop time: 946 ms
2th|  pipe stop time: 946 ms
3th|  pipe stop time: 945 ms
4th|  pipe stop time: 945 ms
5th|  pipe stop time: 946 ms
6th|  pipe stop time: 946 ms
7th|  pipe stop time: 946 ms
8th|  pipe stop time: 946 ms
9th|  pipe stop time: 944 ms`

so the stream effect the stop!

ctxqlxs, 2025-05-06:
ok.

ctxqlxs, 2025-05-08:
sensor stop sometimes crash..

[image]

ctxqlxs, 2025-05-08:
yeah, i known this. because release mode crashed without any infomation, so i use the debug dll.

MartyG-RealSense, 2025-05-08:
What happens if you move **pipe.stop();** within the closing bracket of the {  } block that sensor.stop() is enclosed in?

```
 sensor.stop();
//      sensor.close();

pipe.stop();
}
    
end = std::chrono::high_resolution_clock::now();
elapsed = std::chrono::duration_cast<std::chrono::milliseconds>(end - start).count();
std::cout << i << "th|  pipe stop time: " << elapsed << " ms" << std::endl;
}
```

ctxqlxs, 2025-05-09:
**Err:**
```
RealSense error calling rs2_pipeline_stop(pipe:0000026D951DDA00):
    stop() cannot be called before start()
```

MartyG-RealSense, 2025-05-09:
You identified at https://github.com/IntelRealSense/librealsense/issues/13964#issuecomment-2840533030 that if the depth stream is disabled then the stop-start time is small, indicating that the depth stream is the cause of the delay.  And changing the depth resolution will not make a difference to the delay, as shown by the problem still occurring when the cfg configurations are not used.  Is that assessment correct, please?

ctxqlxs, 2025-05-10:
yeah, it's right. close the depth stream get smaller delay.

MartyG-RealSense, 2025-05-10:
Does depth still lag if you amend your pipe start line to make it ab **auto** statement?

```
 for (int j = 0; j < 10; ++j) {
      auto frames = pipe.wait_for_frames();
    }
```

MartyG-RealSense, 2025-05-17:
Hi @ctxqlxs  Did the advice in the comment above make a difference?

Do you require further assistance with this case, please?  Thanks!

closed 2025-05-24 by MartyG-RealSense