realsenseai/librealsense #13972: Error Intel RealSense Viewer?

maryjvergara, 2025-04-29:
| | |
|---|---|
|**librealsense**|2.56.3 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
GL_INVALID_VALUE - C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\common\opengl3.cpp:572

```

MartyG-RealSense, 2025-04-30:
Hi @maryjvergara  Which RealSense camera model do you have, please?  Thanks!

maryjvergara, 2025-04-30:
Hi! I have the RealSense D435. I was able to solve the issue by installing and running an earlier version of Intel RealSense Viewer and by resetting the graphic display settings to default. Thanks!

MartyG-RealSense, 2025-04-30:
Yes, that's what I was going to suggest.  I asked for the model because that workaround would not usually work with a camera model newer than D435.  As your issue is solved. I will close this case.  Thanks very much for the update!

closed 2025-04-30 by MartyG-RealSense