realsenseai/librealsense #13975: D457 - motion module is not properly working

christopherpapageorgiou, 2025-05-01:
Camera Model: Intel RealSense D455
Firmware Version: 5.16.0.1
Librealsense Version: 2.56.3
Operating System: Ubuntu 22.04.5 LTS (Jammy)
Kernel: 5.15.148-tegra
Platform: Jetson Orin (aarch64, ARMv8)
Jetpack 6.2
Python Version: 3.10.12

Stream Config:
    Color: 640×480 @ 15 FPS
    Depth: 640×480 @ 15 FPS
    IMU: accel @ 100 Hz, gyro @ 200 Hz

Hi everyone,

I’m running into intermittent IMU streaming issues with an Intel RealSense D457 (recognized as a D455) connected to a Jetson Orin via USB-C to USB-A. While the RGB and depth streams generally work reliably, the IMU (accelerometer and gyroscope) behaves inconsistently.

Sometimes everything works fine — I can see live motion data through tools like rs-motion or the RealSense Viewer. But just as often, the motion stream either doesn’t start at all or appears frozen (e.g., pitch, roll, and yaw values don’t update). This happens even though the RGB and depth feeds continue without issue.

To troubleshoot, I’ve tried separating the motion stream from RGB/depth by starting them on separate pipelines. I’ve also experimented with using custom acquisition code in both C++ and Python, sometimes successfully accessing all three streams — but the success is hit-or-miss. Performing a hardware reset through the RealSense Viewer occasionally restores full functionality, but even that isn’t always reliable.

Has anyone else experienced something similar with the D455/D457 on Jetson platforms?
Are there any reliable workarounds or insights into what might be causing this inconsistency with the IMU stream?

MartyG-RealSense, 2025-05-02:
Hi @christopherpapageorgiou  Does reliability improve with your pyrealsense2 two-separate-pipeline script if you incorporate a hardware reset into it by adding the code below before the pipe start line?

```
ctx = rs.context()
devices = ctx.query_devices()
for dev in devices:
dev.hardware_reset()
```

Does performance improve if you unplug the micro-sized end of the USB cable from the base of the camera, turn the connector around the other way and re-insert it?  Or is the camera not detected in that other connector direction?

christopherpapageorgiou, 2025-05-02:
Hi MartyG, thanks for your reply. I usually turn the connector around when I connect the camera and I see through realsense-viewer that the usb connection is 2.1 instead of 3.2. 

I’ve also tried resetting the camera through a script, but unfortunately, that hasn’t worked. The only method that reliably resets the camera is using the hardware reset option within the RealSense Viewer. 

Occasionally, even before I start streaming, I’ll open the RealSense Viewer and see an error related to the motion module in the terminal. Performing a hardware reset through the viewer is the only thing that resolves it.

MartyG-RealSense, 2025-05-02:
Yes, that is usual for a D457 that the USB connection changes from 3.2 to 2.1 when the connector orientation is placed in a particular direction.  I just wanted to confirm what the behaviour of your particular camera was in regards to how the connection is affected.  Thanks very much for the information!

In the RealSense Viewer, you should avoid enabling all three streams simultaneously (depth, RGB, Motion Module) because for some RealSense users this causes one of the streams (often the RGB one but occasionally the IMU) to stop working until the computer is rebooted.  

The trigger for this problem when it occurs is the enabling of the Motion Module, so changing the order in which streams are enabled does not make a difference.  There is no fix for the three-stream issue in the Viewer (in Python the fix is the two-separate-pipeline approach, and in C++ it is using callbacks), so either only two streams should be enabled (e.g IMU + depth, or IMU + RGB) or the Motion Module on its own.

If a **Motion Module Failure** error occurs with the Motion Module in the Viewer when attempting to start it, enabling the Motion Module a second time may cause it to start working normally on the second attempt.  if not, try a hardware reset like you usually do.

As the IMU of your camera can function correctly sometimes, it seems more likely to be a software issue than a hardware problem with the IMU component.  

As you have a D457, you likely had to supply your own USB cable as the D457 comes with a GMSL cable.  The cable is able to produce a 3.2 speed connection, so that suggests that the cable is okay too.  How many meters long is the cable, please?

MartyG-RealSense, 2025-05-02:
If your cable is 1 meter then reliability should be fine as long as it is a high quality cable (which is important for RealSense cameras) and is not one designed simply for device charging / sync instead of data transfer.

The two separate pipelines Python script at https://github.com/IntelRealSense/librealsense/issues/5628#issuecomment-575943238 is the best one I know of for a three-stream depth-RGB-IMU setup.

For C++, the official SDK example program **rs-capture** is a good one for testing all three streams.  rs-capture can display depth, RGB and IMU simultaneously but does not experience the three-stream issue that the RealSense Viewer (which is also a C++ program) has.

https://github.com/IntelRealSense/librealsense/blob/master/examples/capture/rs-capture.cpp

Rather than defining specific streams, rs-capture simply enables all stream types that are supported by a particular camera model, including IMU streams if the model is equipped with an IMU.

closed 2025-05-17 by MartyG-RealSense