realsenseai/librealsense #13980: Support Request for Framos D415e with Windows 11 24H2 – Deadlock Issue

deepakramalingam, 2025-05-05:
Hi Team,

We are using the Framos version mentioned below for the 3D camera features in the Philips DXR System.

Recently, we upgraded our Windows OS to **Windows 11 Version 24H2 (OS Build 26100.3476)** and have since observed a recurring deadlock issue.

The deadlock occurs approximately once every two system restarts, and we currently have no insight into the root cause on our end.

        const rs2::context c;
        nDevices = static_cast<int>(c.query_devices().size());

It occurs on the very first instance of calling **rs2_create_context()** . If it is successful, c.query_devices() returns proper value else there is no response from real sense library after calling rs2::context

Could you please confirm whether the following hardware and software combination is officially supported? If so, could you also provide the appropriate version of the Framos IntelliSense library compatible with Windows 11 24H2, or let us know if there are any plans to address the deadlock issue for this configuration?

**Camera Information:**

* **Name**: FRAMOS D415e
* **Firmware Version**: 05.13.00.50
* **Device Version**:

  * HW: 1.1
  * FW: 1.8.0.0
* **API Version**: 2.50.10

As this issue is currently a blocker, we would appreciate your prompt response.

**Note:**
Please let me know if you need any further information.

Thank you,
Deepak R

---

Would you like help drafting a follow-up if you don’t hear back in a few days?

MartyG-RealSense, 2025-05-05:
Hi @deepakramalingam  Intel are unfortunately not able to provide technical support for the FRAMOS range of RealSense-compatible 'e' models of industrial camera, as FRAMOS are their manufacturer.  Please contact FRAMOS technical support at the website link below to ask them your question.

https://www.framos.com/en/resources/support

MartyG-RealSense, 2025-05-11:
Hi @deepakramalingam  Did you contact FRAMOS as suggested, please?  Thanks!

closed 2025-05-17 by MartyG-RealSense