realsenseai/librealsense #13983: D4 Vision Processor getting Bricked (USB won't recognise)

GirishiF, 2025-05-06:
**Required Info:**
_Camera Model {D450}
Firmware Version (Open RealSense Viewer --> Click info)
Operating System & Version {Win (8.1/10) / Linux (Ubuntu 14/16/17)
Kernel Version (Linux Only) (e.g. 4.14.13)
Platform PC/Raspberry Pi/ NVIDIA Jetson / Qualcomm/etc..
SDK Version {legacy / 2.. }
Language {C/C#/LabVIEW/NodeJS/OpenCV/pcl/python/unity }
Segment {Robot/Smartphone/VR/AR/others }_

Hello @MartyG-RealSense , 
Through trial and error, we have found that for our D450 modules, the fw version 5.15.00.002 works most reliably. For all/most other firmware versions we see this error:

Intel Realsense D455 error calling rs2_load_json(dev: ..., json_content: ..., content_size: 3448)

However sometimes we see this error also with the version 5.15.00.002. When we try to update the firmware version using the fw-update-tool, we continue to see the same error. Repeated updates of the firmware eventually leads to rs device not found error likely because the Vision Board is getting bricked.

Can you please help us with:
the correct fix for rs2_load_json issue
how to fix the issue of rs device not found after repeated fw updates

MartyG-RealSense, 2025-05-12:
You are very welcome.  Thanks very much for the update!

GirishiF, 2025-05-19:
That will be all @MartyG-RealSense! Thanks for your consistent support.

closed 2025-05-19 by MartyG-RealSense