realsenseai/librealsense #13993: Building Inside Docker

AnIVIol, 2025-05-10:
I am trying to build `librealsense` inside a Docker container based on `ros-noetic-core` Dockerfile. There is an issue I am running into.

My host machine is running `Ubuntu 22.04` on kernel version`6.8.0-59-generic`. The docker container is running `Ubuntu 20.04`. 

While following installation steps, I ran `./scripts/setup_udev_rules.sh` and received:
```
Setting-up permissions for RealSense devices
Cannot open device /dev/video0, exiting.
Cannot open device /dev/video0, exiting.
Failed to send reload request: No such file or directory
udev-rules successfully installed
``` 
Which was fixed by passing `--volume="/dev/:/dev/"`. Which itself is a security hazard but I can worry about it later.

Then comes `./scripts/patch-realsense-ubuntu-lts-hwe.sh` which gives:
```
Remove all RealSense cameras attached. Hit any key when ready

Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package linux-headers-6.8.0-59-generic
E: Couldn't find any package by glob 'linux-headers-6.8.0-59-generic'
E: Couldn't find any package by regex 'linux-headers-6.8.0-59-generic'
```

I am guessing it tries to find `linux-headers-6.8.0-59-generic` in 'Ubuntu 20.04' repositories but can't find it. 
Is there anything I can do to get a FULL librealsense build (emphasis on full since the existing Dockerfile in the repo supposedly builds it without graphical examples and IMU support afaik) running inside a Docker container.

MartyG-RealSense, 2025-05-11:
Hi @AnIVIol  If you aim to build librealsense in a Docker container and are not concerned about using ROS even though your Dockerfile is based around ROS1 Noetic, it may be worth instead trying an official Dockerfile for librealsense provided by Intel at the link below.

https://github.com/IntelRealSense/librealsense/tree/master/scripts/Docker

In regard to RSUSB versus kernel patching, https://github.com/IntelRealSense/librealsense/issues/5212#issuecomment-552184604 has a good comparison of the advantages and disadvantages of an RSUSB based installation vs kernel patching.  Scroll down to the section of the linked-to comment headed **What are the advantages and disadvantages of using libuvc vs patched kernel modules?**

That information is over five years old though, and RSUSB has received a number of improvement updates in the librealsense SDK over the subsequent years, so the performance difference between them should be closer than it used to be.  The main area where an RSUSB installation may be slower to respond than a kernel-patched one is when changing the values of option settings.

AnIVIol, 2025-05-11:
Thanks for the quick reply @MartyG-RealSense 
I looked at the Dockerfile. I wanted to confirm one thing, it does not seem to rely on kernel patching and is build with `FORCE_RSUSB_BACKEND=false`.
It only passes the USB device group to the container. 

How does that seem to work? I am sort of lost.

MartyG-RealSense, 2025-05-11:
Thanks very much for the kind words!

An RSUSB = true build of librealsense bypasses the kernel, so it is not dependent on a particular Linux version or kernel version and does not require a kernel patch script to be applied.  It installs dependencies over an internet connection instead of the user having to install dependencies manually.

AnIVIol, 2025-05-11:
Hmm I understand that, but Dockerfile has not specified a flag, hence it means by default RSUSB = false as per documentation. Therefore both kernel path is absent and RSUSB patch is absent. Hence my concern. I am sorry if I am overlooking something.

MartyG-RealSense, 2025-05-11:
librealsense can function without the kernel patch being applied.  The patch enables support for hardware metadata timestamps, so if hardware timestamps are not supported then 'software timestamps' based on the clock of the computer will be provided instead.

When RSUSB is not set to true (it is false by default), the librealsense build will use the default V4L2 backend, also known as the native backend.

AnIVIol, 2025-05-11:
Thanks alot for your time @MartyG-RealSense , hope this conversation helps clarify things for people in the future.

closed 2025-05-11 by AnIVIol

MartyG-RealSense, 2025-05-11:
You are very welcome, @AnIVIol - I'm pleased that I could help!