realsenseai/librealsense #14000: can't get IMU data from D455

dimatusz, 2025-05-12:
Hi,

I can't get the IMU data from D455. `pipeline.wait_for_frames()` times out with `RuntimeError: Frame didn't arrive within 5000`.
I searched around but either there is no solution or the ones that are there don't work. 
I also tried the code below with another camera unit and another PC with Ubuntu and got the same error.
I can see the IMU data with the Viewer. I can also retrieve RGB and depth frames with Python and similar code, no issues there. 
Can you please help with this?

Camera: D455
Serial number: 234322308675
USB type descriptor: 3.2
Camera firmware version: 5.16.0.1
OS: Windows 11 Pro
pyrealsense2==2.55.1.6486
Python 3.11.3

```
import pyrealsense2 as rs
import numpy as np
from time import sleep

def motion_data_to_npa(motion):
    return np.asarray([motion.x, motion.y, motion.z])

pipeline = rs.pipeline()
conf = rs.config()
conf.enable_stream(rs.stream.accel, rs.format.motion_xyz32f, 200)
conf.enable_stream(rs.stream.gyro, rs.format.motion_xyz32f, 200)
pipeline.start(conf)

try:
    while True:
        frames = pipeline.wait_for_frames()

        accel_frame = frames.first_or_default(rs.stream.accel)
        gyro_frame = frames.first_or_default(rs.stream.gyro)

        accel_data = accel_frame.as_motion_frame().get_motion_data()
        gyro_data = gyro_frame.as_motion_frame().get_motion_data()
        
        accel = motion_data_to_npa(accel_data)
        gyro = motion_data_to_npa(gyro_data)
        
        print("accelerometer: ", accel)
        print("gyro: ", gyro)
        sleep(0.5)

finally:
    pipeline.stop()
```

MartyG-RealSense, 2025-05-12:
Thank you for testing.  And is there any difference if you insert a **Keep()** instruction into your first script immediately after the wait_for_frames() line in order to store frames in memory?

```
frames = pipeline.wait_for_frames()
frames.keep()
```

dimatusz, 2025-05-12:
Still the same result; it throws the exception on the previous line: `frames = pipeline.wait_for_frames()`

MartyG-RealSense, 2025-05-12:
And the same problem if you run the script at https://github.com/IntelRealSense/librealsense/issues/5628#issuecomment-575943238 that creates two pipelines and puts IMU on one of them on its own and depth + RGB on the other?

dimatusz, 2025-05-13:
Interestingly, that code works. However, only when at least one of the videos: depth or RBG streams is enabled too. 

The conclusion is that IMU pipeline may not be enough on it's own. The camera has to be configured to stream the images, and then it can stream the IMU too but in a separate pipeline. We can't have a pipeline just for the IMU data.

Even more interestingly, my code works on a Jetson computer. So this is platform dependent.

Thanks!

closed 2025-05-13 by dimatusz

MartyG-RealSense, 2025-05-13:
You are very welcome.  I'm pleased to hear that you achieved positive results.  Thanks very much for the update!

There was a previous case at https://github.com/IntelRealSense/librealsense/issues/8492 with a few Python IMU scripts on it that all had problems on one computer but all worked fine on another.