realsenseai/librealsense #14002: Unable to patch ubuntu 20.04 LTS kernel 5.15

lucasvickers, 2025-05-14:
| **Required Info**  | | Notes |
| ------------- | ------------- | ------------- |
| Camera Model | D455 | |
| Firmware Version | NA | |
| Operating System & Version | Ubuntu 20.04.6 LTS | |
| Kernel Version | 5.15.0-139-generic | |
| Platform | Nuc i7 16GB RAM | |
| SDK Version | 2.55.1 (from package.xml) | Downloaded from git May 14th |

### Issue Description
Patching of kernel fails

```
$ sudo ./scripts/patch-realsense-ubuntu-lts-hwe.sh 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
bc is already the newest version (1.07.1-2build1).
build-essential is already the newest version (12.8ubuntu1.1).
git is already the newest version (1:2.25.1-1ubuntu3.14).
linux-headers-5.15.0-139-generic is already the newest version (5.15.0-139.149~20.04.1).
linux-headers-generic is already the newest version (5.4.0.215.208).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Package required libusb-1.0-0-dev:  - found
Package required libssl-dev:  - found
Create patches workspace in ubuntu-focal folder

Package required libelf-dev:  - found
Package required elfutils:  - found
Package required bison:  - found
Package required flex:  - found
Package required dwarves:  - found
Fetching Ubuntu LTS tag Ubuntu-hwe-5.15-5.15.0-139.149_20.04.1  to the local kernel sources folder
remote: Enumerating objects: 1, done.
remote: Counting objects: 100% (1/1), done.
remote: Total 1 (delta 0), reused 1 (delta 0)
Unpacking objects: 100% (1/1), 812 bytes | 812.00 KiB/s, done.
The kernel has modified files:
	modified:   drivers/media/usb/uvc/uvc_ctrl.c
	modified:   drivers/media/usb/uvc/uvc_driver.c
	modified:   drivers/media/usb/uvc/uvcvideo.h
	modified:   drivers/media/v4l2-core/v4l2-ioctl.c
	modified:   include/uapi/linux/videodev2.h
Proceeding will reset all local kernel changes. Press 'n' within 3 seconds to abort the operation
Do you want to proceed? [Y/n]
Resetting local changes in ubuntu-focal folder
 HEAD is now at 279eeb424 UBUNTU: Ubuntu-hwe-5.15-5.15.0-139.149~20.04.1
Switching to LTS tag Ubuntu-hwe-5.15-5.15.0-139.149_20.04.1
HEAD is now at 279eeb424 UBUNTU: Ubuntu-hwe-5.15-5.15.0-139.149~20.04.1
Applying patches for focal-hwe-5.15 line
Applying realsense-uvc patch
patching file drivers/media/usb/uvc/uvc_driver.c
patching file drivers/media/usb/uvc/uvcvideo.h
patching file drivers/media/v4l2-core/v4l2-ioctl.c
patching file include/uapi/linux/videodev2.h
Applying realsense-metadata patch
patching file drivers/media/usb/uvc/uvc_driver.c
Hunk #1 succeeded at 3266 (offset 102 lines).
patching file drivers/media/usb/uvc/uvcvideo.h
Hunk #1 succeeded at 221 (offset 34 lines).
Applying realsense-powerlinefrequency-fix patch
patching file drivers/media/usb/uvc/uvc_ctrl.c
Hunk #1 succeeded at 367 with fuzz 1 (offset 10 lines).
Increase UVC_URBs in uvcvideo
patching file drivers/media/usb/uvc/uvcvideo.h
Hunk #1 succeeded at 217 (offset 44 lines).
Prepare kernel modules configuration
#
# configuration written to .config
#
[... 80 lines cut ...]
Skipping BTF generation for /home/user/code/librealsense/ubuntu-focal/drivers/media/v4l2-core/v4l2-flash-led-class.ko due to unavailability of vmlinux
  LD [M]  /home/user/code/librealsense/ubuntu-focal/drivers/media/v4l2-core/videobuf-dma-sg.ko
  BTF [M] /home/user/code/librealsense/ubuntu-focal/drivers/media/v4l2-core/videobuf-dma-sg.ko
Skipping BTF generation for /home/user/code/librealsense/ubuntu-focal/drivers/media/v4l2-core/videobuf-dma-sg.ko due to unavailability of vmlinux
  LD [M]  /home/user/code/librealsense/ubuntu-focal/drivers/media/v4l2-core/videobuf-core.ko
  LD [M]  /home/user/code/librealsense/ubuntu-focal/drivers/media/v4l2-core/v4l2-dv-timings.ko
  BTF [M] /home/user/code/librealsense/ubuntu-focal/drivers/media/v4l2-core/videobuf-core.ko
Skipping BTF generation for /home/user/code/librealsense/ubuntu-focal/drivers/media/v4l2-core/videobuf-core.ko due to unavailability of vmlinux
  BTF [M] /home/user/code/librealsense/ubuntu-focal/drivers/media/v4l2-core/v4l2-dv-timings.ko
Skipping BTF generation for /home/user/code/librealsense/ubuntu-focal/drivers/media/v4l2-core/v4l2-dv-timings.ko due to unavailability of vmlinux
  LD [M]  /home/user/code/librealsense/ubuntu-focal/drivers/media/v4l2-core/videodev.ko
  BTF [M] /home/user/code/librealsense/ubuntu-focal/drivers/media/v4l2-core/videodev.ko
Skipping BTF generation for /home/user/code/librealsense/ubuntu-focal/drivers/media/v4l2-core/videodev.ko due to unavailability of vmlinux
  LD [M]  /home/user/code/librealsense/ubuntu-focal/drivers/media/v4l2-core/videobuf-vmalloc.ko
  BTF [M] /home/user/code/librealsense/ubuntu-focal/drivers/media/v4l2-core/videobuf-vmalloc.ko
Skipping BTF generation for /home/user/code/librealsense/ubuntu-focal/drivers/media/v4l2-core/videobuf-vmalloc.ko due to unavailability of vmlinux
make: Leaving directory '/home/user/code/librealsense/ubuntu-focal'
Patched kernels modules were created successfully

Replacing videodev  -
	Applying the patched module ... modprobe: ERROR: could not insert 'videodev': Operation not permitted
Failed to insert the patched module. Operation is aborted, the original module is restored
Verify that the current kernel version is aligned to the patched module version
The original  videodev  module was reloaded
```

MartyG-RealSense, 2025-05-15:
Hi @lucasvickers  Can you confirm the method that you used to install the librealsense SDK, please?  In the information at the top of the discussion it says **2.55.1 (from package.xml)**.   Does this mean that you installed the SDK from DKMS packages?  If you did then it is not necessary to apply a patch script to the kernel, as the patch is already bundled inside the DKMS package.

lucasvickers, 2025-05-15:
Hello,

I pulled from GitHub yesterday and was following instructions at https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

I only referenced package.xml as I wasn’t sure how else to reference the version number.

MartyG-RealSense, 2025-05-15:
There is a past case at https://github.com/IntelRealSense/librealsense/issues/11572 where another RealSense user 
experienced the **ERROR: could not insert 'videodev': Operation not permitted** error when attempting to patch kernel 5.15 on Ubuntu 20.04.

In that case I recommended not using the patch script and instead adding the flag `-DFORCE_RSUSB_BACKEND=true` to the CMake build command in the installation.md instructions.  When librealsense is built from source code with RSUSB = true, applying the patch script to the kernel is not required.

When the hwe patch script for kernel 5.13 and 5.15 was introduced, it was intended for use with Ubuntu 22.04 Jammy rather than 20.04 Focal.

lucasvickers, 2025-05-15:
Thanks. I need multi camera support, will RUSB impact that? If so can you give me an exact compatible Ubuntu and kernel version? I can choose my OS.

MartyG-RealSense, 2025-05-15:
RSUSB works best with single-camera applications.  It is not ideal to use it with multicam applications.

It is possible for a source code build of the librealsense SDK to work without applying a kernel patch script, though camera **hardware metadata** timestamps will not be provided and the SDK will instead provide 'software timestamps' based on the computer's clock.

If you would prefer to use a kernel patch script, kernel 5.11 and patch script **/patch-realsense-ubuntu-lts.sh** would be an option for Ubuntu 20.04.

Otherwise, if you can switch OS to 22.04 or 24.04 then use kernel 5.15 and the **patch-realsense-ubuntu-lts-hwe.sh** script.

lucasvickers, 2025-05-15:
Sorry for the confusion but if I understand correctly I’m using 

Ubuntu 20.04.6 LTS 5.15.0-139-generic

And you’re asking me to use 22.04 or 24.04 then use kernel 5.15

Aren’t I already doing that?

MartyG-RealSense, 2025-05-15:
You stated that you can choose your OS, which I took to mean that whilst you are currently using Ubuntu 20.04, you have the ability to change to a newer OS such as 22.04 or 24.04 if necessary.  My apologies if I misunderstood your statement about choosing an OS.

I would again recommend seeing how librealsense performs with kernel 5.15 on Ubuntu 20.04 without the kernel patch to determine whether it will be necessary to apply a patch.  If errors do not occur then it may be okay to use an unpatched kernel.

lucasvickers, 2025-05-15:
Ubuntu 24.04.2 LTS and kernel 6.8.12-060812-generic yields the error:
> Unsupported distribution noble, kernel version 6.8.12-060812-generic . The patches are maintained for Ubuntu 16/18/20 LTS

I selected 6.8 due to the list at https://github.com/IntelRealSense/librealsense/releases/

I don't see 5.X as an option in Mainline for 24.04, so I'm not sure it's available.

Do I need to try 22.04, or is a 6.X kernel supported?

Full output:
```
$ ./scripts/patch-realsense-ubuntu-lts-hwe.sh
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
linux-headers-6.8.12-060812-generic is already the newest version (6.8.12-060812.202501300202).
build-essential is already the newest version (12.10ubuntu1).
git is already the newest version (1:2.43.0-1ubuntu7.2).
bc is already the newest version (1.07.1-3ubuntu4).
bc set to manually installed.
The following additional packages will be installed:
  linux-headers-6.8.0-60 linux-headers-6.8.0-60-generic
The following NEW packages will be installed:
  linux-headers-6.8.0-60 linux-headers-6.8.0-60-generic linux-headers-generic
0 upgraded, 3 newly installed, 0 to remove and 1 not upgraded.
Need to get 18.1 MB of archives.
After this operation, 115 MB of additional disk space will be used.
Get:1 http://us.archive.ubuntu.com/ubuntu noble-updates/main amd64 linux-headers-6.8.0-60 all 6.8.0-60.63 [13.9 MB]
Get:2 http://us.archive.ubuntu.com/ubuntu noble-updates/main amd64 linux-headers-6.8.0-60-generic amd64 6.8.0-60.63 [4,158 kB]
Get:3 http://us.archive.ubuntu.com/ubuntu noble-updates/main amd64 linux-headers-generic amd64 6.8.0-60.63 [10.5 kB]   
Fetched 18.1 MB in 12s (1,520 kB/s)                                                                                    
Selecting previously unselected package linux-headers-6.8.0-60.
(Reading database ... 200910 files and directories currently installed.)
Preparing to unpack .../linux-headers-6.8.0-60_6.8.0-60.63_all.deb ...
Unpacking linux-headers-6.8.0-60 (6.8.0-60.63) ...
Selecting previously unselected package linux-headers-6.8.0-60-generic.
Preparing to unpack .../linux-headers-6.8.0-60-generic_6.8.0-60.63_amd64.deb ...
Unpacking linux-headers-6.8.0-60-generic (6.8.0-60.63) ...
Selecting previously unselected package linux-headers-generic.
Preparing to unpack .../linux-headers-generic_6.8.0-60.63_amd64.deb ...
Unpacking linux-headers-generic (6.8.0-60.63) ...
Setting up linux-headers-6.8.0-60 (6.8.0-60.63) ...
Setting up linux-headers-6.8.0-60-generic (6.8.0-60.63) ...
Setting up linux-headers-generic (6.8.0-60.63) ...
Package required libusb-1.0-0-dev:  - found
Package required libssl-dev:  - found
Unsupported distribution noble, kernel version 6.8.12-060812-generic . The patches are maintained for Ubuntu 16/18/20 LTS
```

lucasvickers, 2025-05-15:
FWIW Ubuntu 24.04.2 LTS and kernel 6.8.12-060812-generic seems stable so far, I'll test dual camera next week.

In the meantime it would be helpful to have a specific OS and Kernel combination that is both achievable and known to successfully patch, in case I need to go down that route.

Thanks for the help!

MartyG-RealSense, 2025-05-16:
I'm pleased to hear you found an OS and kernel combination that seems to be stable.  I look forward to hearing dual-camera test results.

It is difficult to give specific kernel recommendations.  RealSense users use a wide range of kernels, and whilst I can say whether a 'major' class of kernel version number is supported by the SDK (such as 6.8), there may be differences in performance between the 'minor' version numbers of that major class, such as 6.8.11 versus 6.8.12.

I would recommend testing with whatever kernel is installed by default when you do a fresh install of Ubuntu.  For example, 24.04 uses kernel 6.8 as its default for a new install.  Patching for 6.8 is supported in librealsense beta 2.56.2 onwards.

lucasvickers, 2025-05-24:
Sorry for the slow update. I've been streaming 2 cameras nonstop for ~2 days and will continue to do so over the weekend.
Seems stable to me.

One thing to note is that it was not clear to me that patching was optional. Based on how I've read the documentation, seemed like you had to patch or compile an alternative way. Perhaps this can be clearer?

Thanks!

MartyG-RealSense, 2025-05-24:
Patching is the recommended procedure for a non-libuvc / RSUSB source code build.  The occasions when **not** applying a patch is necessary in order to resolve a problem are rare, as in most cases using libuvc / RSUSB is a suitable officially recommended alternative and going the unpatched route is only required when using multiple cameras or when the application is a mission-critical one where an libuvc / RSUSB build of the SDK will not respond as fast as a kernel-dependent 'native' build of the SDK.

So an unpatched source code build is not a recommended installation strategy, but past support case history has shown that going unpatched is worth trying as a last resort in cases where libuvc / RSUSB cannot be used by a project.

closed 2025-05-27 by lucasvickers