realsenseai/librealsense #14007: D435i camera, the obtained depth values are far larger than the actual real-world values.

Andybert, 2025-05-16:
import numpy as np                        # fundamental package for scientific computing
import matplotlib.pyplot as plt           # 2D plotting library producing publication quality figures
import pyrealsense2 as rs                 # Intel RealSense cross-platform open-source API
print("Environment Ready")

# Setup:
pipe = rs.pipeline()
config = rs.config()

pipeline_wrapper = rs.pipeline_wrapper(pipeline)
pipeline_profile = config.resolve(pipeline_wrapper)
device = pipeline_profile.get_device()

config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

profile = pipeline.start(config)

depth_sensor = profile.get_device().first_depth_sensor()
depth_multiplier = 6
depth_sensor.set_option(rs.option.depth_units, 0.001000 / depth_multiplier)
depth_sensor.set_option(rs.option.emitter_always_on, 1.0)
depth_scale = depth_sensor.get_depth_scale()

align_to rs.stream.color
align = rs.align(align_to)

clipping_distance_in_meters = 2
clipping_distance = clipping_distance_in_meters /depth_scale

################### filter #############################
spatial = rs.spatial_filter()
spatial.set_option(rs.option.filter_magnitude, 5)
spatial.set_option(rs.option.filter_smooth_alpha, 1)
spatial.set_option(rs.option.filter_smooth_delta, 50)
spatial.set_option(rs.option.holes_fill, 3)

temporal = rs.temporal_filter()

#Hole Filling
hole_filling = rs.hole_filling_filter()

depth_to_disparity = rs.disparity_transform(True)
disparity_to_depth = rs.disparity_transform(False)
################### filter #############################

try:
	while True:
		frames = pipeline.wait_for_frames() # create a pipeline object
		aligned_frames = align.process(frames) # align the depth frame to color frame

		aligned_depth_frame = aligned_frames.get_depth_frame() # get aligned frames

		frame = depth_to_disparity.process(aligned_depth_frame)
		frame = spatial.process(aligned_depth_frame)
		frame = temporal.process(aligned_depth_frame)
		frame = disparity_to_depth.process(aligned_depth_frame)
		frame = hole_filling.process(aligned_depth_frame)

		np_image = np.asanyarray(frame.get_datga()).astype(float)
		dists = np_image * depth_scale
		dists[dists>clipping_distance_in_meters] = 0

finally:
	pipeline.stop()

The hardware model is D435i. I want to use the above code to read the actual distance (unit: m), but the returned result is far greater than the actual distance. How should I modify it? When viewed with Intel RealSense Viewer, it is normal.

MartyG-RealSense, 2025-05-16:
Hi @Andybert  I would recommend first trying to remove the **depth_multiplier** parameter from the line that sets the depth scale, as shown below.  Dividing the scale by the multiplier will throw off the depth values.

`depth_sensor.set_option(rs.option.depth_units, 0.001000)`

For example, the real-world depth value in meters is calculated by multiplying the raw pixel depth value (from the array) by the scale value.  So if a pixel depth value of 6500 was multiplied by the D435i's default scale of 0.001 then that would give a real-world distance of 6.5 meters.

Andybert, 2025-05-19:
> Hi [@Andybert](https://github.com/Andybert) I would recommend first trying to remove the **depth_multiplier** parameter from the line that sets the depth scale, as shown below. Dividing the scale by [...]

Thanks for the reply. I modified the code according to your suggestion. The modified code is as follows, but the measured distance is more than ten times larger than the actual distance (dists = np_image * depth_scale). The depth_scale value obtained via depth_scale = depth_sensor.get_depth_scale() is 0.0010000000474974513.

```
import numpy as np                        # fundamental package for scientific computing
import matplotlib.pyplot as plt           # 2D plotting library producing publication quality figures
import pyrealsense2 as rs                 # Intel RealSense cross-platform open-source API
print("Environment Ready")

# Setup:
pipe = rs.pipeline()
config = rs.config()

pipeline_wrapper = rs.pipeline_wrapper(pipe)
pipeline_profile = config.resolve(pipeline_wrapper)
device = pipeline_profile.get_device()

config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

profile = pipe.start(config)

depth_sensor = profile.get_device().first_depth_sensor()
depth_sensor.set_option(rs.option.depth_units, 0.001000)
depth_sensor.set_option(rs.option.emitter_always_on, 1.0)
depth_scale = depth_sensor.get_depth_scale()

align_to rs.stream.color
align = rs.align(align_to)

clipping_distance_in_meters = 2
clipping_distance = clipping_distance_in_meters /depth_scale

################### filter #############################
spatial = rs.spatial_filter()
spatial.set_option(rs.option.filter_magnitude, 5)
spatial.set_option(rs.option.filter_smooth_alpha, 1)
spatial.set_option(rs.option.filter_smooth_delta, 50)
spatial.set_option(rs.option.holes_fill, 3)

temporal = rs.temporal_filter()

#Hole Filling
hole_filling = rs.hole_filling_filter()

depth_to_disparity = rs.disparity_transform(True)
disparity_to_depth = rs.disparity_transform(False)
################### filter #############################

try:
	while True:
		frames = pipe.wait_for_frames() # create a pipe object
		aligned_frames = align.process(frames) # align the depth frame to color frame

		aligned_depth_frame = aligned_frames.get_depth_frame() # get aligned frames

		frame = depth_to_disparity.process(aligned_depth_frame)
		frame = spatial.process(aligned_depth_frame)
		frame = temporal.process(aligned_depth_frame)
		frame = disparity_to_depth.process(aligned_depth_frame)
		frame = hole_filling.process(aligned_depth_frame)

		np_image = np.asanyarray(frame.get_datga()).astype(float)
		dists = np_image * depth_scale
		dists[dists>clipping_distance_in_meters] = 0

finally:
	pipe.stop()
```

MartyG-RealSense, 2025-05-19:
Thanks very much for the information.  Is the depth value still large if you use the **get_distance()** command to obtain and print the real-world depth in meters instead of multiplying the depth value by the scale?

```
import pyrealsense2 as rs
import numpy as np

pipeline = rs.pipeline()

# Start streaming
pipeline.start()

frames = pipeline.wait_for_frames()
depth = frames.get_depth_frame()

width = depth.get_width()
height = depth.get_height()

dist = depth.get_distance(int(width/2), int(height/2))
print(dist)

# Stop streaming
pipeline.stop()
```

Andybert, 2025-05-19:
> Thanks very much for the information. Is the depth value still large if you use the **get_distance()** command to obtain and print the real-world depth in meters instead of multiplying the depth [...]

@MartyG-RealSense Thank you for your latest suggestion. I ran the following code (the original code reported an error at frames = pipeline.wait_for_frames(): "Frame didn't arrive within 5000"). The returned depth values are now correct.

```
import numpy as np    
import pyrealsense2 as rs   

# Setup:
pipe = rs.pipeline()
config = rs.config()

pipeline_wrapper = rs.pipeline_wrapper(pipe)
pipeline_profile = config.resolve(pipeline_wrapper)
device = pipeline_profile.get_device()

config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

profile = pipe.start(config)

depth_sensor = profile.get_device().first_depth_sensor()

frames = pipeline.wait_for_frames()
depth = frames.get_depth_frame()

width = depth.get_width()
height = depth.get_height()

dist = depth.get_distance(int(width/2), int(height/2))
print(dist)

# Stop streaming
pipeline.stop()
```

MartyG-RealSense, 2025-05-19:
I'm pleased to hear that you obtained a correct result from the get_distance() method.

Andybert, 2025-05-19:
> I'm pleased to hear that you obtained a correct result from the get_distance() method.

@MartyG-RealSense But what I need is to obtain the complete depth map rather than just a single point. Is there any way to achieve this besides iterating over each pixel using the get_distance() function?

MartyG-RealSense, 2025-05-19:
The SDK example program **python-tutorial-1-depth** uses get_distance() and prints "a simple text-based representation of the image, by breaking it into 10x20 pixel regions and approximating the coverage of pixels within one meter coverage".

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/python-tutorial-1-depth.py

[image]

Andybert, 2025-05-19:
> The SDK example program **python-tutorial-1-depth** uses get_distance() and prints "a simple text-based representation of the image, by breaking it into 10x20 pixel regions and approximating the [...]

@MartyG-RealSense Is there a tutorial that can achieve obtaining precise and complete distance information similar to Intel RealSense Viewer?

Andybert, 2025-05-19:
> The example **opencv_viewer_example** will provide a Viewer-like display of depth. It does not display the depth values as text though, only color-shades them.
> 
> https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/opencv_viewer_example.py
> 
> [image]

@MartyG-RealSense Apologies for my lack of clarity. My task is monocular depth estimation, and I aim to use the D435i camera to obtain actual distances as training labels for a monocular depth estimation model. Therefore, I need to acquire an accurate and complete 2D matrix of real-world distances. Specifically, this would be a 2D matrix the same size as the input image, where each value represents the actual distance (in meters) at that pixel location.

MartyG-RealSense, 2025-05-19:
How about a script that saves depth as an array of scaled matrices to an **.npy** format file.

https://github.com/IntelRealSense/librealsense/issues/4934#issuecomment-537705225

An .npy can be imported into non-RealSense tools, such as TensorFlow and PyTorch.

Andybert, 2025-05-19:
> How about a script that saves depth as an array of scaled matrices to an **.npy** format file.
> 
> [#4934 (comment)](https://github.com/IntelRealSense/librealsense/issues/4934#issuecomment-537705225)
> 
> An .npy can be imported into non-RealSense tools, such as TensorFlow and PyTorch.

In this example, the real-world distance conversion is also implemented directly using the following code, which is the same as my original code. However, the real-world distance obtained by this method is far larger than the true real-world distance:
python
```
depth_scale = depth_sensor.get_depth_scale()  

depth_image = np.asanyarray(aligned_depth_frame.get_data())  
scaled_depth_image = depth_image * depth_scale  
```

MartyG-RealSense, 2025-05-19:
The only other approach I can think of to achieve an array of text values of depth whose positions in the array correspond to the original coordinates would be to save a single frame from the camera to a **bag** format file using the save_single_frameset() instruction, like in this script:

https://github.com/soarwing52/RealsensePython/blob/master/separate%20functions/single_frameset.py

And then convert the bag file to a textual **csv** file, which can be opened in a wordprocessor, database or spreadsheet.

The RealSense SDK has a bag to csv conversion tool called rs-convert.  If your librealsense installation includes the tools and examples programs then it should be located in the **usr/local/bin** directory of Ubuntu.

https://github.com/IntelRealSense/librealsense/tree/master/tools/convert

Andybert, 2025-05-20:
@MartyG-RealSense I used the recording function of Intel RealSense Viewer to obtain a .bag file, then used the extract_from_bag function in this link[(https://github.com/IntelRealSense/librealsense/issues/4934#issuecomment-537705225)] to get an .npy file of depth information. However, I noticed that the maximum value in it is over 60. Does this mean the .npy file does not store actual distance values in meters?

MartyG-RealSense, 2025-05-20:
RealSense outputs its depth measurements in the distance unit of meters.  

60 might be an erroneous value, like the very large ones that you were getting earlier.

You could try implementing a Threshold post-processing filter that automatically excludes depth values that are outside a minimum and maximum range.  There is an example of use of the threshold filter in Python at https://github.com/IntelRealSense/librealsense/issues/8170#issuecomment-761930301

Andybert, 2025-05-21:
> RealSense outputs its depth measurements in the distance unit of meters.
> 
> 60 might be an erroneous value, like the very large ones that you were getting earlier.
> 
> You could try implementing a Threshold post-processing filter that automatically excludes depth values that are outside a minimum and maximum range. There is an example of use of the threshold filter in Python to set a maximum at [#8170 (comment)](https://github.com/IntelRealSense/librealsense/issues/8170#issuecomment-761930301)

@MartyG-RealSense I apologize. I found that the distance obtained using the extract_from_bag function in this link is approximately 10 times the actual value. Could there be an issue with the depth_scale value obtained via depth_scale = depth_sensor.get_depth_scale()?

MartyG-RealSense, 2025-05-21:
You could try loading one of the official RealSense sample bag files into the script to see whether the large values still occur.  If they do not then it could suggest that the large depth values are within your own bag file.

https://github.com/IntelRealSense/librealsense/blob/master/doc/sample-data.md