realsenseai/librealsense #14008: 使用vs2022做C#开发，realsense2.dll引入不了

xinbada19851109, 2025-05-16:
使用vs2022做C#开发，realsense2.dll引入不了

[image]
[image]

MartyG-RealSense, 2025-05-16:
Hello @xinbada19851109  In a past case with the same problem at https://github.com/IntelRealSense/librealsense/issues/5022 the RealSense user in that case suggests going to the Solution Configuration section of Visual Studio and setting it to Release instead of Debug.

********

你好@xinbada19851109 在过去的案例中，https://github.com/IntelRealSense/librealsense/issues/5022 也出现过同样的问题，当时的 RealSense 用户建议转到 Visual Studio 的解决方案配置部分，并将其设置为 Release 而不是 Debug。

MartyG-RealSense, 2025-05-24:
Do you require further assistance with this case, please?  Thanks!

请问您还需要进一步的帮助吗？谢谢！

MartyG-RealSense, 2025-05-30:
Case closed due to no further comments received.

由于未收到进一步评论，案件已结案。

closed 2025-05-30 by MartyG-RealSense