realsenseai/librealsense #14009: In android development, how can RGB video recording be achieved to obtain mp4 files

1programmer2dfg3, 2025-05-17:
I need to obtain an mp4 file with sound. How can I record it? Is there a specific code demo

MartyG-RealSense, 2025-05-17:
Hi @1programmer2dfg3  RealSense applications do not usually record sound as they have no need for audio data.  There is an Android project at the link below for recording RealSense data to mp4 that may be a helpful reference for you though.

https://apkpure.com/rs-recorder/com.neuroeventlabs.rscamera

The project can be downloaded as an APK package from the above link, and the source code for it is here:

https://gitlab.com/neuroeventlabs/tools/realsense-android

1programmer2dfg3, 2025-05-18:
I tested this demo here and found many errors. The installation would crash and there were also many code errors. Is there a simpler method
I don't want a preview of the picture. I need to achieve video recording directly in the background

MartyG-RealSense, 2025-05-18:
There was an official RealSense Android application on the Play Store that could record and playback.  It does not seem to be on the Store now but the link below is an archive of the store page that provides the details about its functions.

https://web.archive.org/web/20240106053737/https://play.google.com/store/apps/details?id=com.intel.realsense.camera

[image]

The RealSense Android wrapper has a source-code version of this tool here:

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/android/tools/camera

I have not used the app myself.  From the store information though, it sounds as if it records depth and color data to a **bag** file rather than an MP4 video file though.

*********

I could not locate any examples of mp4 grabbing with RealSense script code, unfortunately.  The link below has a video frame grabber for the **.avi** video format that is apparently RealSense compatible via its integration of the **RealSenseFrameGrabber** and **RealSense2FrameGrabber** Java classes.

https://github.com/bytedeco/javacv

MartyG-RealSense, 2025-05-24:
Hi @1programmer2dfg3  Do you require further assistance with this case, please?  Thanks!

1programmer2dfg3, 2025-05-24:
Yes, I need, I haven't solved it yet. The demo provided above can't solve it 。Thanks!
 

丁福根
***@***.***

MartyG-RealSense, 2025-05-25:
The RealSense SDK has a bag recording example program for the Android wrapper here:

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/android/examples/recording

I could not find any other RealSense Android recording solutions, unfortunately.  You could consider using the **Screen Recorder** tool that is in Android 11 onwards to capture RealSense streams displayed on the screen, or use a Play Store screen recording application if your Android version is older than 11.

https://www.pcmag.com/how-to/how-to-record-the-screen-on-your-android-phone

MartyG-RealSense, 2025-06-08:
Hi @1programmer2dfg3  Was my advice in the comment above helpful or do you need further advice, please?  Thanks!

MartyG-RealSense, 2025-06-21:
The only official SDK demonstration of recording for Android is the bag recording in the Android wrapper's Camera tool that I mentioned earlier, unfortunately.  It does not provide an mp4 recording function.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/android/tools/camera/src/main