realsenseai/librealsense #14014: When I open both rgb and depth maps simultaneously in the viewer software, the picture will be very laggy

dut-zjh, 2025-05-19:
When I open both rgb and depth maps simultaneously in the viewer software, the picture will be very laggy. There wasn't such a situation before. Could you help me

dut-zjh, 2025-05-19:
[image]

dut-zjh, 2025-05-19:
https://github.com/user-attachments/assets/4227f8be-e2e6-418d-a984-3a3677ad8aa6

MartyG-RealSense, 2025-05-19:
Hi @dut-zjh  When you say lag, do you mean the delay in the RGB image starting up?  There does not seem to be a delay once the RGB stream has started.

On the D405 camera model, the delay should not be a problem.  This is because it does not have an RGB sensor component and instead obtains the RGB image from the depth sensor.  This means that the RGB image is always perfectly aligned and synchronized with the depth map.

As the default resolution of D405 is 848x480 though, you could try enabling depth and color at 848x480 instead of 1280x720 to see if it makes a difference to the start-up delay.

dut-zjh, 2025-05-19:
No. What I mean is that the FPS is very low and doesn't reach the specified value. You can watch the video I recorded after calling it with python. There is a big difference between the two

https://github.com/user-attachments/assets/f54b46b9-9e85-42f8-bd5c-8fee38ce8c57

https://github.com/user-attachments/assets/a695b3c2-0ee2-46b2-bbd6-59c53d6875bf

MartyG-RealSense, 2025-05-19:
The RealSense Viewer applies a range of post-processing filters by default, whilst a Python script applies no filters by default (they have to be deliberately programmed into the script).  Post-processing filters place a processing burden on the computer's CPU as they are calculated on the computer and not on the camera hardware.

So if you disable the Post Processing section of the Viewer side-panel then you will get a better like-for-like comparison between the performance of the Viewer and the Python program.

Please also click the 'i' icon at the top of the Viewer's depth and RGB stream panels to overlay the FPS value over the top of the images.

dut-zjh, 2025-05-19:
I did as you said. Although the software shows that the fps is around 60, the picture is still visibly lagging and not smooth. Why is that? I wonder if you can understand my confusion

https://github.com/user-attachments/assets/75cb8b13-780e-4e0a-8ea0-d303d85e3237

MartyG-RealSense, 2025-05-19:
The depth image may be smoother and fluctuate less if you enable the post-processing filters again and expand open the list of filters, then go to the **Temporal** filter and expand its settings open.  In the 'Filter Smooth Alpha' setting, reduce its value from the default setting of 0.4 to '0.1'.  This should make the image more stable.

closed 2025-05-19 by dut-zjh