realsenseai/librealsense #14022: No Frames Received On windows 10 (It was work well on Jetson though)

KRJames-lab, 2025-05-23:
| | |
|---|---|
|**librealsense**|2.54.2 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense L515|
|**Serial Number**|f0172221|
|**Firmware Version**|1.5.8.1|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|L500|
|**Asic Serial Number**|0003a8a54860|
|**Firmware Update Id**|0003a8a54860|

Hi, I'm having some problems using the L515. **I was working on my project on the Jetson with the L515, and the Realsense Viewer worked fine.** Then, I needed to use the L515 on Windows 10. So, I installed the Realsense Viewer, but the "No Frames Received" error occurred.

The way I downloaded Realsense viewer.
[image]

And the result
[image]

### What I've tried
1. Allow the camera to be used in the desktop app -> Allowed RealSense(TM) Viewer
2. Change version. 2.56.3, 2.50.0, 2.47.0
3. Change USB port
4. Try on a labtop
5. Changed power option. USB Selective Sleep Mode Settings (That means Do Not Use)
[image]
6. I'm not sure why, but the sample works well.
[image]
[image]

Please let me know how to solve this problem...!

MartyG-RealSense, 2025-05-23:
Hi @KRJames-lab  The final SDK version to officially support L515 was 2.50.0.  Afterwards, it could work unofficially up until 2.52.1 but was not tested for correct working.

Could you first try unplugging the micro-sized end of the USB cable from the base of the camera, reversing the connector's orientation and plugging it back into the L515 camera, please (USB-C cables are two-way insertion at the micro-sized end).  Does doing so make any difference?

KRJames-lab, 2025-05-26:
**1. Are you launching the RealSense Viewer using the launch shortcut that is placed on the Windows desktop by the Intel.RealSense.SDK-WIN10 installation program?**

No, I lunched RealSense Viewer with Window search.
[image]

**2. Do you have any other RealSense programs active when the Viewer is opened? For example, if one of the example programs such as rs-align was running when the Viewer was launched then it would prevent the Viewer from using streams, as only one program at a time can use them.**

No, I did't think so. How can I check the other RealSense programs acitve? Should I check on Task Manager?
This is the list of Background processes, and I’m sure there aren’t any other processes in the Apps section.
[image]

MartyG-RealSense, 2025-05-26:
1.  Launching the Viewer from Windows search will be the same as using a desktop shortcut, so that's okay.

2.  The way to be sure that no other RealSense programs are running is to launch the Viewer as your first action after the computer has been started up.

KRJames-lab, 2025-05-26:
I see! Thanks to you, I’ll be able to move forward with the project smoothly. Thank you!

MartyG-RealSense, 2025-05-26:
You are very welcome.  I'm pleased that I could help.  Good luck with your project!

MartyG-RealSense, 2025-06-08:
Hi @KRJames-lab  Do you require further assistance with this case, please?  Thanks!

KRJames-lab, 2025-06-09:
No, I'm good for now, thanks!

MartyG-RealSense, 2025-06-09:
You are very welcome.  As you do not require further assistance, I will close this case.  Thanks again!