realsenseai/librealsense #14025: jetpack 6.2, d455 not recognized in realsense view.

bruce577, 2025-05-26:
On Jetson AGX Orin with Ubuntu 22.04, when I run RealSense Viewer, the RealSense D455 device is not detected.
When I run rs-enumerate-devices, it says “No device detected.”
However, USB recognition and kernel logs both confirm that the device is connected properly.
What should I do?

MartyG-RealSense, 2025-05-26:
Hi @bruce577  A common solution for non-camera detection when an IMU-equipped RealSense camera is used on a Jetson with JetPack 6 is to build the librealsense SDK from source code using the **libuvc backend** installation method.  

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

The libuvc backend method provides a pre-made build script called **libuvc_installation.sh** that can be found at the link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

When installing on a Jetson, I recommend changing the CMake build instruction on line 46 of the script to the instruction below:
 
`cmake ../ -DFORCE_LIBUVC=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DBUILD_WITH_CUDA=on`

*******

For some RealSense users of Jetson and JP6, installing with the libuvc method still does not resolve the problem.  If that happens to you then taking an additional step of downgrading the camera's firmware driver to version **5.13.0.50** often finally resolves the issue.

As you have a D455 you can use this method.  I should warn other readers of this case with RealSense models newer than D455 (D435f, D435if, D455f, D456, D457) that they should not downgrade their firmware to 5.13.0.50 or the camera will stop functioning.

******

Before trying a libuvc installation though, you could first see if you camera is detected when launching programs in **sudo** admin permissions mode:

`sudo realsense-viewer`

`sudo rs-enumerate-devices`

bruce577, 2025-05-26:
Hello, thanks to your help, my D455 camera is now recognized in the RealSense Viewer. However, when I try to turn on the RGB camera, I get the following error:
synthetic-stream.cpp:48 - Exception was thrown during callback: loc_devIhEESt10shared_ptrIT_EiEUlPhE_
The stereo module and motion module work fine. Do you know what might be causing this?

bruce577, 2025-05-27:
It will not run if i enable RGB camera only.
Running it in Python will also show up.
Tracking (last most recent call):
파일 "/home/kwon-islab/ros2_ws/view_realsense.py", 14번째 줄, <module>
frames = pipeline.wait_for_frames ()
Runtime Error: Frame did not arrive within 5000

Should I reinstall sdk??

MartyG-RealSense, 2025-05-27:
I note that you have your RealSense Python script within the **ros2_ws** directory.  Do you also have the librealsense SDK installed at this location, please?  If you do then librealsense should be installed at a different location not within the ROS2 catkin workspace directory or it can cause problems.

bruce577, 2025-05-27:
[image]

Like this? sorry,,,,

MartyG-RealSense, 2025-05-27:
To set the RGB from Infrared mode, expand open the Stereo Module and click on the black box beside **Infrared** to turn it blue and select it.  Open the format drop-down menu for Infrared and select BGR8.  Then enable the Stereo Module (not the RGB section).

bruce577, 2025-05-27:
[image]

Yes, thank you. The camera will now run.
I wonder if the original realsense viewer doesn't print out the original screen?

MartyG-RealSense, 2025-05-27:
The Viewer should be able to display the RGB image when the **RGB** category is enabled.  Does it still show No Frames Received if you select a lower resolution such as 640x480 before enabling the RGB?

In your Python script you can enable the RGB from infrared mode with a cfg configuration line.  For example:

`cfg.enable_stream(rs.stream.infrared, 848, 480, rs.format.bgr8, 30)`

MartyG-RealSense, 2025-05-27:
Does the error still occur if the FPS speed is reduced in the script from 30 to 15 ?

bruce577, 2025-05-28:
Yes yes, when I select YUYV on RGB camera, the screen comes up, but I can't find any other color stream.

MartyG-RealSense, 2025-05-28:
Could you try unplugging the micro-sized end of the USB cable from the base of the camera, turning the connector around the other way and plugging it back into the camera, please (USB-C cables are two-way insertion at the micro-sized end).  Does doing so make any difference?

bruce577, 2025-05-30:
Hello, I wanted to get the rgb value and pointcloud value, so I kept trying. If it says I can't get the rgb camera frame, can't I print out the RGB value?
Is jetpack 6.2 realsense d455 not compatible.. there is also realsense d435.

MartyG-RealSense, 2025-05-31:
IMU-equipped RealSense camera models have issues with JetPack 6 because JP6 removed a component called **hiddraw** that RealSense IMU cameras depended upon.  Non-IMU camera models such as D435 are not affected and function normally.

Another Jetson user at https://github.com/IntelRealSense/librealsense/issues/14038 had the same RGB problem with the same error a couple of days ago.  Their camera was also an IMU-equipped one, a D435i.  In their situation, all RGB formats were unavailable including YUYV.

If the RGB image shows No Frames Received then there may not be RGB values available to retrieve.

Usually, installing librealsense with libuvc backend and using firmware 5.13.0.50 if necessary resolves the majority of cases involving JetPack 6 issues.  The other methods for dealing with JP6 issues are to downgrade the JetPack to version 5.1.3, or to install the RealSense MIPI driver that provides support for IMU cameras when using JP6.

https://github.com/IntelRealSense/realsense_mipi_platform_driver

What happens if you run the Python **opencv_pointcloud_viewer.py** pointcloud example program?

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/opencv_pointcloud_viewer.py

kaola-zero, 2025-06-05:
> Hello, thanks to your help, my D455 camera is now recognized in the RealSense Viewer. However, when I try to turn on the RGB camera, I get the following error: synthetic-stream.cpp:48 - Exception [...]

Hello, I have encountered the same problem as you. When using the D455 camera in Jetpack 6.2, I still cannot find the D455 device in realsense_view. I have tried on a computer, but it displays both the device and the image. How did you solve this problem? Thank you very much for your reply!

MartyG-RealSense, 2025-06-24:
Hi @bruce577 and @kaola-zero  Do either of you require further assistance with this case, please?  Thanks!

TeixeiraDasilva-Helena, 2025-06-25:
I have the same issue. The RGB and Stereo streams work fine but IMU data will not stream. Using a Jetson AGX Orin with Jetpack 6.2 and a Realsense d455. I tried the fix with the mipi driver but that is only supported through Jetpack 6.0.

MartyG-RealSense, 2025-07-05:
Hi @TeixeiraDasilva-Helena  Were you able to test the libuvc backend SDK installation method suggested in the comment above, please?  Thanks!

closed 2025-07-27 by MartyG-RealSense