realsenseai/librealsense #14030: Intel RealSense Viewer only detect USB 2.1 for D405 camera

ValentinLeglise, 2025-05-27:
Hello Intel,

I am encountering an issue by using your software Intel RealSense Viewer with my D405 camera. In the datasheet of this product, it is indicated that this camera supports USB 3.0. Also, with the camera, a 3.2 USB cable is provided.

When i use the camera with this cable on the software, it detects USB 2.1 instead of 3.2. I have tried multiple cable but the problem is still the same. Does this camera really support USB 3.0 ? Can I change the USB detection manually ?

Thanks in advance for your help.
Valentin.

MartyG-RealSense, 2025-05-27:
Hi @ValentinLeglise  RealSense cameras can be used with USB 3.0 ports on a computer, but the preference is for USB 3.1 / 3.2.  The information below is from the D405's listing on the official RealSense Store.

[image]

Do you have the ability to plug the camera into a USB 3.1 hub instead of directly into the computer to test whether this allows the camera to be detected as USB 3.2, please?  A USB hub has its own USB controller for the ports on it, separate from the controller that the ports on the computer use.

ValentinLeglise, 2025-05-27:
Hi. I just want to add more information. My customer (who discovered this issue), has multiple camera and only one unit have this problem. All other cameras seems to easily recognize the USB 3.1. Maybe it is an hardware problem of the camera ?

MartyG-RealSense, 2025-05-27:
There are no Viewer settings that relate to the configuration of the USB port.

Are all of your customer's camera units the D405 model or a mix of different models?

Has the customer tested the D405 on its own or do they always have multiple cameras attached in addition to the D405?

If your customer runs the **rs-enumerate-devices** program and launches it with the command below to provide a short summary of the camera information, is the USB type of the affected D405 camera shown as 2.1 or 3.2?

`rs-enumerate-devices -s`

ValentinLeglise, 2025-05-27:
All the units he has are all D405 model camera. He has tested every camera but only one unit has this USB issue.

MartyG-RealSense, 2025-05-27:
Please ask the customer if it is possible for them to downgrade the firmware driver of just that one affected camera to version **5.13.0.50** to see whether it corrects the issue.  Doing so sometimes corrects problems that are otherwise unsolvable.

5.13.0.50 is the minimum version that D405 supports, so they should not install a version older than 5.13.0.50 or the camera will stop functioning.

closed 2025-06-15 by MartyG-RealSense