realsenseai/librealsense #14033: Failed to unload module uvcvideo error while applying kernel patch for ubuntu 24.04

Vishnu-AP, 2025-05-28:
I am trying to build librealsense from source following the instructions in installation.md from the tag v2.56.3

While applying the kernel patch this is the error I get:

Patched kernels modules were created successfully

sudo: modprobe: command not found
Failed to unload module uvcvideo. error type 1 . Operation is aborted

Important things to note:
1. I am running ubuntu 24.04 in a docker container and my goal is to build ros2-realsense inside the container so that I can use the d435i.
2. I am using the tag v2.56.3 because I want to build ros2-realsense from the branch 4.56.3 which is compatible with ros2 jazzy

MartyG-RealSense, 2025-05-28:
Hi @Vishnu-AP  Please try the approach of not using the kernel patch script and instead adding the flag **-DFORCE_RSUSB_BACKEND=true** to your CMake build instruction so that kernel patching is not required, as this flag enables librealsense to bypass the kernel.

Vishnu-AP, 2025-05-29:
Hi @MartyG-RealSense.

I just tried with the DFORCE_RSUSB_BACKEND=true option and it works, the librealsense package builds as expected. But when I try running rs-fw-update or rs-enumerate-devices, I do not see the device being detected. Is this something to do with the firmware? I am running 5.13.0.50 
But for me to update the firmware, the camera should be detected right? Even in the realsense-viewer I dont see anything

MartyG-RealSense, 2025-05-29:
If you are using librealsense 2.55.1 then firmware 5.13.0.50 should be okay.  Otherwise, if you are using one of the newer beta versions (2.56.1, 2.56.2, 2.56.3) then I recommend updating the firmware to 5.16.0.1.

Is the camera detected if you launch programs in sudo admin permissions mode?

`sudo rs-enumerate-devices`

`sudo realsense-viewer`

Vishnu-AP, 2025-05-29:
I was able to figure out why I was seeing no outputs, actually I forgot to uninstall the ros-jazzy-librealsense2 and ros-jazzy-realsense2 binaries. Uninstalling them solved my issue and the realsense was detected. 
Okay I will update the firmware to 5.16.0.1, but if I decide to use the camera in a SDKv2.55.1 (there are still some systems using that) Would it still work with the 5.16.0.1 version?

Vishnu-AP, 2025-05-30:
Okay great, thank you! I will be closing the issue since it has been resolved

closed 2025-05-30 by Vishnu-AP