realsenseai/librealsense #14035: 关于False Nearest Neighbors, FNN的问题

xyl265, 2025-05-29:
请问我想使用你的fnn.m加入我的matlab中改选取项目中的哪个代码，求教

MartyG-RealSense, 2025-05-29:
Translation: Question about False Nearest Neighbors, FNN.  I want to use your fnn.m and add it to my MATLAB.  Which code in the project should I choose?  Please advise.

Hello @xyl265  The RealSense SDK does not have a False Nearest Neighbor detection function.  If your goal is to exclude from the image pixels with inaccurate / invalid depth values then you could try setting a camera configuration preset called High Accuracy.  It would be difficult to implement this in the RealSense MATLAB wrapper but possible to do so.

How you may be able to achieve it is with the procedure below.

1.  Visit the Visual Presets page and download the preset file called HighAccuracyPreset.json

https://dev.intelrealsense.com/docs/d400-series-visual-presets#preset-table

2.  Use the code in the RealSense SDK's advanced_mode_example-m example program to load the HighAccuracyPreset.json file and so apply the High Accuracy preset that excludes inaccurate looking depth values.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/matlab/advanced_mode_example.m

***************

您好 @xyl265。RealSense SDK 没有错误最近邻检测功能。如果您的目标是从图像中排除深度值不准确/无效的像素，可以尝试设置一个名为“高精度”的摄像头配置预设。虽然在 RealSense MATLAB 封装器中实现此功能比较困难，但并非不可能。

您可以按照以下步骤操作。

1. 访问 Visual Presets 页面并下载名为 HighAccuracyPreset.json 的预设文件

https://dev.intelrealsense.com/docs/d400-series-visual-presets#preset-table

2. 使用 RealSense SDK 的 advanced_mode_example-m 示例程序中的代码加载 HighAccuracyPreset.json 文件，并应用“高精度”预设，从而排除看起来不准确的深度值。

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/matlab/advanced_mode_example.m

MartyG-RealSense, 2025-06-08:
Hi @xyl265  Do you require further assistance with this case, please?  Thanks!

您好 @xyl265，请问您需要进一步的帮助吗？谢谢！

MartyG-RealSense, 2025-06-15:
Case closed due to no further comments received.

由于未收到进一步评论，案件已结案。

closed 2025-06-15 by MartyG-RealSense