realsenseai/librealsense #14041: Camera not connected

ky047, 2025-06-02:
Hello,
I tried connecting the D435 camera to my computer, but it does not appear in the Viewer.
I tested a different camera with the same cable, and it worked fine.
I also replaced the cable and got the same result — the D435 still does not connect.

Given this, can I assume that the D435 camera is malfunctioning?
Is there any other method I can use to verify the status of this camera?

Thank you,
Giyoung Jung

MartyG-RealSense, 2025-06-02:
Hi @ky047  Is the D435 detected if you run the rs-enumerate-devices tool and retrieve a short summary of the camera's information with the command `rs-enumerate-devices -s`

MartyG-RealSense, 2025-06-09:
If you are using Windows (as you mention the Device Manager), let's try uninstalling and re-installing the drivers of all your cameras from the Windows Device Manager to eliminate the possibility that the installed drivers are damaged.

1.   Unplug all cameras from the computer. 

2. Find and open the Device Manager interface by typing **device** into the text box at the bottom corner of your Windows screen.

[image]

3.  In the Device Manager window, go to the **View** menu option and select **Show hidden devices** to list the Depth and RGB driver pairs for each of the RealSense cameras that have previously been attached to the computer.  So if you have two cameras, there should be two pairs of Depth and RGB drivers on your particular computer for your two individual cameras.  

[image]

[image]

4.  Right-click on a Depth or RGB driver and select the **Uninstall device** menu option.   Repeat until all of the pairs of Depth and RGB drivers have been uninstalled.

[image]

5.  Plug the cameras into the computer.  The drivers should automatically reinstall into the Device Manager.  Then open the RealSense Viewer to see whether they can be detected.

ky047, 2025-06-10:
[image]
[image]
Thank you for providing the guidance.
However, we followed the steps you mentioned, but the camera still does not connect.
The picture below shows the result after we removed and reconnected the camera driver.
At this point, it seems that the camera may be malfunctioning.

Best regards,
Giyoung Jung

MartyG-RealSense, 2025-06-11:
Can you check the camera privacy settings in Windows please to ensure that permission for applications to access the camera is set to On.  If other cameras work then the permissions are probably already on but it is worth checking to be sure.  Instructions for doing so can be found at https://github.com/IntelRealSense/librealsense/issues/3381#issuecomment-503424789

Also please try unplugging the micro-sized end of the USB cable from the side of the camera, turning it around the other way and re-inserting it if you have not tried this already (USB-C cables are two-way insertion at the micro-sized end).

Has this camera ever been detected since it was purchased?

If the privacy settings or reversing the USB connector do not make a difference and if the camera was purchased less than a year ago then I would recommend contacting the retailer that the camera was purchased from to ask about a replacement under warranty.

ky047, 2025-06-11:
Thank you for your answer.

I checked the personal settings, and the camera is turned on.

I haven’t checked the cable yet. I will try that next.

If the issue still persists after checking the cable, can I assume that the camera may be faulty?

Best regards,
Giyoung Jung