realsenseai/librealsense #14047: rk3588   Frame didn't arrive within 5000

chenwenfeng, 2025-06-05:
I followed the tutorial at https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md to install the driver successfully. However, when I tried to use pyrealsense2 to read image data, I encountered the following error:
frames = pipeline.wait_for_frames()
RuntimeError: Frame didn't arrive within 5000

my env:

Ubuntu 20.04.6 LTS
kernel 5.10.160

my test code:

  pipeline = rs.pipeline()
  config = rs.config()
  config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)
  profile = pipeline.start(config)
  try:
    frames = pipeline.wait_for_frames()
    color_frame = frames.get_color_frame()
    if not color_frame:
        raise RuntimeError("error")
    color_image = np.asanyarray(color_frame.get_data())
    fname = os.path.join(os.path.dirname(os.path.abspath(__file__)), 'web/static/images/test.png')
    cv2.imwrite(fname, color_image)
  finally:
    pipeline.stop()

MartyG-RealSense, 2025-06-05:
Hi @chenwenfeng  Does your Python program function correctly if you launch it in **sudo** admin permissions mode?  For example:

`sudo python3 test.py`

chenwenfeng, 2025-06-05:
> Hi [@chenwenfeng](https://github.com/chenwenfeng) Does your Python program function correctly if you launch it in **sudo** admin permissions mode? For example:嗨，如果您在 **sudo** 管理员权限模式下启动您的 Python 程序，它是否正常运行？例如：
> 
> `sudo python3 test.py`

yes , same error

MartyG-RealSense, 2025-06-05:
And the same problem occurs if you remove 'config' from the brackets of the start line so that the camera uses its default configuration for color instead of your custom cfg configuration?

`profile = pipeline.start()`

chenwenfeng, 2025-06-05:
> And the same problem occurs if you remove 'config' from the brackets of the start line so that the camera uses its default configuration for color instead of your custom cfg configuration?
> 
> `profile = pipeline.start()`

I tried again, but I’m still getting the same error.

MartyG-RealSense, 2025-06-05:
What are your librealsense SDK and camera firmware driver versions, please?

chenwenfeng, 2025-06-05:
> What are your librealsense SDK and camera firmware driver versions, please?

I downloaded the source code from https://github.com/IntelRealSense/librealsense/archive/master.zip and compiled it using the following command:cmake ../ -DFORCE_LIBUVC=true -DCMAKE_BUILD_TYPE=release -DBUILD_PYTHON_BINDINGS=ON -DPYTHON_EXECUTABLE=$(which python3) 
The result of the Python package build was a file named: pyrsutils.cpython-38-aarch64-linux-gnu.so.2.55.1

If you need any other information, please let me know where I can find it.

chenwenfeng, 2025-06-05:
> Thanks very much. If you used the libuvc backend installation procedure then your librealsense SDK version will be 2.55.1, as shown by your pyrealsense2 version also being 2.55.1.
> 
> If you use `rs-enumerate-devices -s` then it should show your firmware driver version.

Device Name                   Serial Number       Firmware Version
Intel RealSense D435          834412070885        5.16.0.1

MartyG-RealSense, 2025-06-05:
Thank you very much again.

5.16.0.1 is the correct firmware driver version for librealsense 2.55.1.  However, if a camera is not fully working correctly after installing with the libuvc backend procedure then it is recommended that their firmware is downgraded to the earlier version **5.13.0.50** (which can also work with 2.55.1).  Doing so often resolves problems with a libuvc installation and as you have a D435, you can use this method (newer RealSense models such as D435f and D435if cannot).  

The downgrade can be performed in the RealSense Viewer tool using the instructions below.

Download the Viewer bin file for 5.13.0.50 to your computer from the link below.
https://dev.intelrealsense.com/docs/firmware-releases#d400-series-firmware-downloads

[image]

Go to the **More** option near the top of the Viewer's options side-panel and select Update Firmware from its drop-down menu.

A file navigation box will pop up.  Navigate to the location on your computer where you downloaded the .bin file to and select the file.  The firmware update should now automatically begin.

chenwenfeng, 2025-06-06:
> The dmesg log message ** Failed to query (GET_CUR) UVC control 1** indicates a failure on the kernel driver level to read the requested parameter, for instance if the device was busy with internal operation or that the incoming requests queue is full. In most cases this message can be disregarded as the SDK has built-in auto-retry mechanism to compensate for the temporalities.
> 
> When librealsense is installed with the **libuvc backend** method like you did though, the SDK bypasses the kernel and so should not be affected by kernel related problems.
> 
> Are you using an Nvidia Jetson Nano computing board, please? Kernel 5.10 is most often used on that system.
> 
> Are you able to use the RealSense Viewer tool? If you are, can RGB color be streamed in the Viewer?

I’m using the Lubancat 4 development board with the RK3588S2 chip, running Ubuntu Server 20.04.

On my Windows PC, the RealSense Viewer works fine, and I’m also able to successfully read both depth and RGB images using my program.

MartyG-RealSense, 2025-06-06:
The Lubancat 4 board has USB 3.0 ports.  Whilst it is possible to use RealSense cameras with USB 3.0, what tends to happen is that the camera connection uses the slower USB 2 speed instead.   RealSense cameras should ideally be used with a computer with USB 3.1 ports.

Do frames arrive if 6 FPS is used instead of 30?

`config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 6)`

chenwenfeng, 2025-06-06:
I also tested with:config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 6)
but it still doesn’t work — the same error occurs.

Regarding what was mentioned about USB 3.1 being required: does this mean the Lubancat 4 board cannot run the RealSense D435 at all?

Is there any possible workaround or solution for using the D435 on the Lubancat 4 (RK3588S2) platform?

chenwenfeng, 2025-06-06:
Thank you very much. Since the Lubancat 4 board is running the server version of Ubuntu, it’s not convenient to use the RealSense Viewer for testing.

Next week, I plan to try using a powered USB hub to see if it helps with the issue.

MartyG-RealSense, 2025-06-06:
Thanks very much too.  I look forward to your next report.  Good luck!

In the meantime, if you can only run text-based programs (no graphics) on Ubuntu Server then you could try the **rs-hello-realsense** example program if you have the examples installed, as it is text based.

chenwenfeng, 2025-06-09:
Update on my testing results:
	1.	I ran a C++ test program and was able to successfully read both RGB and depth data.
	2.	However, when using Python with pyrealsense2, I can only retrieve depth data — the RGB stream fails to start or returns no frames.
	3.	I also tested with a powered USB hub, but the device could not be recognized. This might be due to the poor quality of the hub itself.

Given that the C++ program can access both RGB and depth streams, I believe this rules out any hardware or firmware-related issues. My current guess is that the problem lies in the pyrealsense2 Python bindings or how they are interacting with the environment on the Lubancat 4 board.

my c++ test code:

#file  hello.cpp

#include <librealsense2/rs.hpp> 
#include <iostream>        

int main() try {
    rs2::pipeline p;
    p.start();

    while (true) {
        rs2::frameset frames = p.wait_for_frames();
        
        rs2::depth_frame depth = frames.get_depth_frame();
        
        float dist_to_center = depth.get_distance(
            depth.get_width() / 2, 
            depth.get_height() / 2
        );
        
        std::cout << "The camera is facing an object " 
                  << dist_to_center << " meters away \r";
    }
    return EXIT_SUCCESS;
} catch (const rs2::error& e) {
    std::cerr << "RealSense error: " << e.what() << std::endl;
    return EXIT_FAILURE;
}

#file rs-capture-rgb.cpp

#include <librealsense2/rs.hpp>
#include <opencv2/opencv.hpp>  // 引入 OpenCV

int main() try {
    rs2::pipeline p;
    rs2::config cfg;
    cfg.enable_stream(RS2_STREAM_COLOR, 640, 480, RS2_FORMAT_BGR8, 30); 
    p.start(cfg);

    auto frames = p.wait_for_frames();
    auto color_frame = frames.get_color_frame();

    cv::Mat image(cv::Size(640, 480), CV_8UC3, (void*)color_frame.get_data(), cv::Mat::AUTO_STEP);

    cv::imwrite("rgb_image.png", image); 
    std::cout << "Image saved as rgb_image.png" << std::endl;

    return EXIT_SUCCESS;
} catch (const rs2::error& e) {
    std::cerr << "RealSense error: " << e.what() << std::endl;
    return EXIT_FAILURE;
}

chenwenfeng, 2025-06-09:
I deleted all the previous environment setups and recompiled everything from scratch. Now the tests are working correctly. It was probably caused by leftover files from the previous builds, since I found pyrealsense2 in both /usr/local/lib/python3.8/dist-packages and /usr/lib/python3.8/dist-packages.

Sorry for taking up so much of your time.

MartyG-RealSense, 2025-06-15:
Case closed due to solution achieved and no further comments received.

closed 2025-06-15 by MartyG-RealSense

zhang-qihao, 2025-07-12:
> I deleted all the previous environment setups and recompiled everything from scratch. Now the tests are working correctly. It was probably caused by leftover files from the previous builds, since I [...]

May I ask how you deleted all the previous environment setups? I'm also using a LubanCat4 development board, but I still can't use pyrealsense2 to retrieve frames. 

I got an error when running the code.
```python
import pyrealsense2 as rs

pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 15)
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 15)

profile = pipeline.start(config)

```
```bash
Traceback (most recent call last):
  File "realsense_project/test.py", line 8, in <module>
    profile = pipeline.start(config)
RuntimeError: No device connected
```

More details have been provided in the issue #14129.

I look forward to your reply. Thank you very much! 🙏