realsenseai/librealsense #14055: librealsense on RK3588 (Kernel 5.10) not support gray image of D435i

tom-jerr, 2025-06-09:
Hello community,
We are currently trying to compile and use librealsense from source on a CoolPi 4B(RK3588) to support our Intel RealSense D435i camera.

Our progress so far is as follows:
We have successfully compiled the library on our system.
The RGB Camera and Depth Camera functionalities are working correctly, and we can stream data from them without issues.
However, we are facing a problem: we are unable to receive the dual infrared (grayscale) image streams from the stereo cameras.
We understand that enabling these streams typically requires applying Intel's kernel patches to the system. The challenge is that our CoolPi 4B runs on a 5.10 kernel, and the official librealsense patches do not seem to support this version.
This leads us to two main questions for the community:
1. Is there a community-supported solution or patch available for the 5.10 kernel? We are looking for any guidance or workarounds that would allow us to enable the full functionality of the D435i, specifically the stereo IR streams.

2. As an alternative, could the community recommend a stereo camera that is well-suited for use with VINS-Fusion? We are looking for a camera that has reliable support and a smoother integration process on ARM-based platforms like the CoolPi or Raspberry Pi.

We would appreciate any help or advice you can offer.
Thank you

MartyG-RealSense, 2025-06-09:
Hi @tom-jerr  You should be able to receive infrared streams on a system with an unpatched kernel.  What would prevent the dual left and right infrared streams from displaying is if you camera was on a USB 2 connection.  The right-side infrared stream is only accessible on a USB 3 connection. On USB 2, only left-side infrared can be obtained.

The Cool Pi 4 board apparently has 2x USB 2.0 ports and 2x USB 3.0 ports.  USB 3.1 is recommended for RealSense cameras, and sometimes a RealSense camera will be detected as USB 2.1 when plugged into a USB 3.0 port.

For a Rockchip (RK) system, it is recommended that librealsense is compiled from source code with CMake with the flag `-DFORCE_RSUSB_BACKEND=true` included in the CMake build instruction.  An RSUSB = true build of librealsense bypasses the kernel and so does not need to have a kernel patch applied.

When RSUSB is not set to true It is possible to use librealsense with an unpatched kernel but hardware metadata timestamps will not be generated and the timestamps will instead be 'software' ones that are based on the computer's clock.

If you are considering using Raspberry Pi in future then I strongly recommend using a Pi 5, as RealSense cameras have restricted performance on a Pi 4 board.

tom-jerr, 2025-06-10:
Thank you for your explanation.
It appears that our USB 3.0 port is not being correctly identified as a USB 3.0 connection. Could you please provide any potential solutions or recommendations?

MartyG-RealSense, 2025-06-10:
In realsense-viewer, you can find whether the USB connection is being detected as USB 2 or USB 3 by looking at the number beside the camera name at the top of the options side-panel.  It will either say 2.1 or 3.2.  When it is 2.1, Infrared (the left infrared image) should be able to be streamed, but not Infrared 2 (the right-side infrared image).

In the RealSense ROS wrapper, your connection is detected as 3.2 (USB 3), as shown on the **Device USB type** line of the launch log.  However, both of the infrared streams are disabled by default in the ROS wrapper.  You can enable them by adding the commands below to your roslaunch instruction:

`enable_infra1:=true enable_infra2:=true`

Both of the infrared topics should then be published.

tom-jerr, 2025-06-12:
Thanks for your help, now we have solved this problem. We enabled them by adding the commands below to roslaunch instruction: enable_infra1:=true enable_infra2:=true, then we can see the topic of infra1 and infra2.

closed 2025-06-12 by tom-jerr