realsenseai/librealsense #14061: error: ‘uint8_t’ does not name a type

wallop560, 2025-06-10:
```
/home/le-birdo/.cache/yay/librealsense/src/librealsense-2.55.1/src/core/frame-interface.h:27:19: error: ‘uint8_t’ does not name a type
   27 |     virtual const uint8_t * get_frame_data() const = 0;
      |                   ^~~~~~~
/home/le-birdo/.cache/yay/librealsense/src/librealsense-2.55.1/src/core/frame-interface.h:6:1: note: ‘uint8_t’ is defined in header ‘<cstdint>’; this is probably fixable by adding ‘#include <cstdint>’
    5 | #include <librealsense2/h/rs_frame.h>
  +++ |+#include <cstdint>
    6 | #include <memory>
make[2]: *** [CMakeFiles/realsense2.dir/build.make:79: CMakeFiles/realsense2.dir/src/core/matcher-factory.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1023: CMakeFiles/realsense2.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
==> ERROR: A failure occurred in build().
```
i get this error whenever i try to build it from the aur.
all i am doing is ``yay -S librealsense``

MartyG-RealSense, 2025-06-10:
Hi @wallop560  The error **error: ‘uint8_t’ does not name a type** has not been previously reported in relation to the librealsense SDK.  Performing a more general (non-RealSense) research of the problem suggested that the error can result from **OpenMP** not being enabled when building.

in librealsense, when OpenMP is set to true it enables YUY to RGB color conversion and depth-color alignment to take advantage of multiple CPU cores for their processing.

A RealSense user at https://github.com/IntelRealSense/librealsense/issues/10519#issuecomment-1129710315 built librealsense from AUR, which used a librealsense build flags configuration where OpenMP was set to true (it is false by default in librealsense), so this adds weight to the possibility that an absence of OpenMP is the cause.

The configuration that they used to build librealsense from AUR, with the `-DBUILD_WITH_OPENMP` build configuration flag set to 'on', is here:

https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=librealsense#n58

**********

When using a non-Ubuntu flavor of Linux, I would recommend building librealsense from source code using the **libuvc backend** installation method, which provides a pre-made build script called 'libuvc_installation.sh'.  When installing with this method, the build is not dependent on a particular Linux type or kernel type and does not need to have a patch script applied to the kernel.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

https://github.com/IntelRealSense/librealsense/blob/master/scripts/libuvc_installation.sh

sawirt, 2025-06-21:
Hi I am actually having the exact same issue. Based on the comments on the AUR page, this is currently an issue. I tried to build with the build configuration you mentioned too, however the error mentioned by OP persists until the include `#include <cstdint>` is added in a couple of header files. Changing all of these is a bit of a pain, so I was wondering if I am missing something?

MartyG-RealSense, 2025-06-22:
Hi @sawirt  In May 2023 at the link below, `#include<cstdint>` was added to a RealSense header at **common/sw-update/http-downloader.h** to prevent uint_t errors from occurring.  So it looks as though the edit that you performed was indeed the appropriate course of action.

https://github.com/IntelRealSense/librealsense/pull/11852/commits/39225ebb8f98cf78600e04cd39fed6ae2c299092

sawirt, 2025-06-22:
I see. If anyone will stumble across the same issue in the future what I did is:

1. Clone the repo form AUR `https://aur.archlinux.org/packages/librealsense`
2. `cd` into the repo
3. Taking the diff
```
diff -ruN a/common/opengl3.h b/common/opengl3.h
--- a/common/opengl3.h
+++ b/common/opengl3.h
@@ -10,6 +10,7 @@
 #include <unordered_map>
 #include <vector>
 #include <memory>
+#include <cstdint>

 void _check_gl_error(const char *file, int line);
 void clear_gl_errors();
diff -ruN a/src/core/debug.h b/src/core/debug.h
--- a/src/core/debug.h
+++ b/src/core/debug.h
@@ -4,6 +4,7 @@

 #include "extension.h"
 #include <vector>
+#include <cstdint>

 namespace librealsense
 {
diff -ruN a/src/core/frame-additional-data.h b/src/core/frame-additional-data.h
--- a/src/core/frame-additional-data.h
+++ b/src/core/frame-additional-data.h
@@ -8,7 +8,7 @@
 #include <memory>
 #include <array>
 #include <cstring>  // memcpy
-
+#include <cstdint>

 namespace librealsense {

diff -ruN a/src/core/frame-interface.h b/src/core/frame-interface.h
--- a/src/core/frame-interface.h
+++ b/src/core/frame-interface.h
@@ -5,6 +5,7 @@
 #include <librealsense2/h/rs_frame.h>
 #include <memory>
 #include <iosfwd>
+#include <cstdint>

 namespace librealsense {
diff -ruN a/src/core/stream-profile-interface.h b/src/core/stream-profile-interface.h
--- a/src/core/stream-profile-interface.h
+++ b/src/core/stream-profile-interface.h
@@ -10,7 +10,7 @@
 #include <memory>
 #include <vector>
 #include <ostream>
-
+#include <cstdint>

 namespace librealsense {

```
to create a patch by copying the content to a file `cstdint-fix.patch` in the cloned libsense repo.
4. add the patch to the `source` array of the `PKGBUILD` file:
```
...
source=("https://github.com/IntelRealSense/librealsense/archive/refs/tags/v${pkgver}.tar.gz"
    "https://librealsense.intel.com/Releases/RS4xx/FW/D4XX_FW_Image-${_RS4XX_VER}.bin"
    "https://librealsense.intel.com/Releases/SR300/FW/SR3XX_FW_Image-${_SR300_VER}.bin"
    "https://librealsense.intel.com/Releases/L5xx/FW/L51X_FW_Image-${_L51X_VER}.bin"
    "https://librealsense.intel.com/Releases/L5xx/FW/L53X_FW_Image-${_L53X_VER}.bin"
    "realsense-viewer.desktop"
    "cstdint-fix.patch") # <---
...
```
5. set the sha256sums of the patch in the same `PKGBUILD` file to `'SKIP'`:
```
....
sha256sums=('54546d834ff5d8b35d9955319ad2e428f6d9ae4c61b932d1bd716ed81ad135f7'
            'a481376ac2d072de1d057fe73d74fcc00ab5da17aa63fa92c18bb8f65adf909c'
            'c4ac2144df13c3a64fca9d16c175595c903e6e45f02f0f238630a223b07c14d1'
            '87a9a91b613d9d807b2bfc424abe9cac63cad75dfc04718592c44777cb0b4452'
            'b837b2cff2b270b89eed3c0b212ab4108389a20b6e07c19dd5957918ff9ce7e0'
            '59281f91e7d471a7dde1cf7207eddd8624e05218cc4301ee52e4c453a0c8ab21'
            'SKIP') # <---
...
```
6. Again in the `PKGBUILD` file adding to the `prepare()` function:
```
prepare() {
  cd "${srcdir}/${pkgname}-${pkgver}"
  patch -p1 < ./cstdint-fix.patch # <---
  sed -i 's|, GROUP:="plugdev"||g' "config/99-realsense-libusb.rules"
  sed -i 's|, GROUP="plugdev"||g' "config/99-realsense-libusb.rules"

  mkdir -p build/common/fw/
  cp "../D4XX_FW_Image-${_RS4XX_VER}.bin" build/common/fw/
  cp "../SR3XX_FW_Image-${_SR300_VER}.bin" build/common/fw/
  cp "../L51X_FW_Image-${_L51X_VER}.bin" build/common/fw/
  cp "../L53X_FW_Image-${_L53X_VER}.bin" build/common/fw/
}
```
7. Updating the checksums by running `updpkgsums` in the terminal.
8. Running `makepkg -sric` to install with pacman.

This is my first time building a package on arch, so I am not sure if this is the appropriate way to do it, but `realsense-viewer` seems to work as intended.
I hope this may help someone if they have the same issue.

closed 2025-07-15 by MartyG-RealSense