realsenseai/librealsense #14062: crash when connecting to model D405, firmware update

NocusNibbes, 2025-06-11:
| | |
|---|---|
|**librealsense**|2.49.0 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
Program terminated due to an unrecoverable SEH exception:
Access Violation!
```

MartyG-RealSense, 2025-06-11:
Hi @NocusNibbes  The D405 model is not compatible with RealSense SDK 2.49.0.  Official support for it began in version 2.51.1, though it was able to work unofficially with 2.50.0.

MartyG-RealSense, 2025-06-24:
Hi @NocusNibbes  Do you require further assistance with this case, please?  Thanks!

NocusNibbes, 2025-06-25:
> What happens if you visit the link below and download and run the file
> *Intel.RealSense.Viewer.exe*
>
> https://github.com/IntelRealSense/librealsense/releases/tag/v2.55.1
>
> image.png (view on web)

Dear

best regards,
Nanno Mulder

> *Marty *
The viewer works now for the D405 and the D415.
Thanks for your persistent help

Greets,
Nanno Mulder.

*-RealSense* left a comment (IntelRealSense/librealsense#14062)

MartyG-RealSense, 2025-06-25:
You are very welcome.  It's excellent to hear that it is working for you.  I'm pleased that I could help!