realsenseai/librealsense #14068: Errors occurred when building the PCL

rayee922, 2025-06-12:
I followed the guide to build realsense from the source code and added -DBUILD_PCL_EXAMPLES=true. But I received the following error prompt. Could you check it for me? Thank you so much.

/usr/bin/ld: CMakeFiles/rs-pcl.dir/rs-pcl.cpp.o: in function `ImGuiSetStyleColors()':
rs-pcl.cpp:(.text+0x17f1): undefined reference to `ImGui::SetNextWindowBgAlpha(float)'
/usr/bin/ld: rs-pcl.cpp:(.text+0x17f6): undefined reference to `ImGui::GetStyle()'
/usr/bin/ld: rs-pcl.cpp:(.text+0x1802): undefined reference to `ImGui::GetStyle()'
/usr/bin/ld: rs-pcl.cpp:(.text+0x180e): undefined reference to `ImGui::GetStyle()'
/usr/bin/ld: CMakeFiles/rs-pcl.dir/rs-pcl.cpp.o: in function `window::window(int, int, char const*)::{lambda(GLFWwindow*, int, int, int, int)#4}::_FUN(GLFWwindow*, int, int, int, int)':
rs-pcl.cpp:(.text._ZZN6windowC4EiiPKcENUlP10GLFWwindowiiiiE2_4_FUNES3_iiii[_ZZN6windowC4EiiPKcENUlP10GLFWwindowiiiiE2_4_FUNES3_iiii]+0x2f): undefined reference to `ImGui::GetCurrentContext()'
/usr/bin/ld: rs-pcl.cpp:(.text._ZZN6windowC4EiiPKcENUlP10GLFWwindowiiiiE2_4_FUNES3_iiii[_ZZN6windowC4EiiPKcENUlP10GLFWwindowiiiiE2_4_FUNES3_iiii]+0x47): undefined reference to `ImGui_ImplGlfw_KeyCallback(GLFWwindow*, int, int, int, int)'
/usr/bin/ld: CMakeFiles/rs-pcl.dir/rs-pcl.cpp.o: in function `window::window(int, int, char const*)::{lambda(GLFWwindow*, double, double)#3}::_FUN(GLFWwindow*, double, double)':
rs-pcl.cpp:(.text._ZZN6windowC4EiiPKcENUlP10GLFWwindowddE1_4_FUNES3_dd[_ZZN6windowC4EiiPKcENUlP10GLFWwindowddE1_4_FUNES3_dd]+0x2a): undefined reference to `ImGui::GetCurrentContext()'
/usr/bin/ld: rs-pcl.cpp:(.text._ZZN6windowC4EiiPKcENUlP10GLFWwindowddE1_4_FUNES3_dd[_ZZN6windowC4EiiPKcENUlP10GLFWwindowddE1_4_FUNES3_dd]+0x41): undefined reference to `ImGui_ImplGlfw_CursorPosCallback(GLFWwindow*, double, double)'
/usr/bin/ld: CMakeFiles/rs-pcl.dir/rs-pcl.cpp.o: in function `window::window(int, int, char const*)::{lambda(GLFWwindow*, double, double)#2}::_FUN(GLFWwindow*, double, double)':
rs-pcl.cpp:(.text._ZZN6windowC4EiiPKcENUlP10GLFWwindowddE0_4_FUNES3_dd[_ZZN6windowC4EiiPKcENUlP10GLFWwindowddE0_4_FUNES3_dd]+0x2a): undefined reference to `ImGui::GetCurrentContext()'
/usr/bin/ld: rs-pcl.cpp:(.text._ZZN6windowC4EiiPKcENUlP10GLFWwindowddE0_4_FUNES3_dd[_ZZN6windowC4EiiPKcENUlP10GLFWwindowddE0_4_FUNES3_dd]+0x41): undefined reference to `ImGui_ImplGlfw_ScrollCallback(GLFWwindow*, double, double)'
/usr/bin/ld: CMakeFiles/rs-pcl.dir/rs-pcl.cpp.o: in function `window::window(int, int, char const*)::{lambda(GLFWwindow*, int, int, int)#1}::_FUN(GLFWwindow*, int, int, int)':
rs-pcl.cpp:(.text._ZZN6windowC4EiiPKcENUlP10GLFWwindowiiiE_4_FUNES3_iii[_ZZN6windowC4EiiPKcENUlP10GLFWwindowiiiE_4_FUNES3_iii]+0x2a): undefined reference to `ImGui::GetCurrentContext()'
/usr/bin/ld: rs-pcl.cpp:(.text._ZZN6windowC4EiiPKcENUlP10GLFWwindowiiiE_4_FUNES3_iii[_ZZN6windowC4EiiPKcENUlP10GLFWwindowiiiE_4_FUNES3_iii]+0x3f): undefined reference to `ImGui_ImplGlfw_MouseButtonCallback(GLFWwindow*, int, int, int)'
collect2: error: ld returned 1 exit status
make[2]: *** [wrappers/pcl/pcl/CMakeFiles/rs-pcl.dir/build.make:185: Release/rs-pcl] Error 1
make[1]: *** [CMakeFiles/Makefile2:1300: wrappers/pcl/pcl/CMakeFiles/rs-pcl.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....

MartyG-RealSense, 2025-06-13:
Hi @rayee922  Before building librealsense from source code, did your first install PCL using the commands below, please?

```
sudo add-apt-repository ppa:v-launchpad-jochen-sprickerhof-de/pcl
sudo apt-get update
sudo apt-get install libpcl-all
```

This is from the PCL wrapper installation instructions for Linux at the link below.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/pcl#linux

rayee922, 2025-06-13:
Yes, it would say:
CMake Warning (dev) at wrappers/pcl/pcl-color/CMakeLists.txt:7 (find_package):
  Policy CMP0074 is not set: find_package uses <PackageName>_ROOT variables.
  Run "cmake --help-policy CMP0074" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  CMake variable PCL_ROOT is set to:

    /usr

  For compatibility, CMake is ignoring the variable.
This warning is for project developers.  Use -Wno-dev to suppress it.

Then when I was building the librealsense from the source code, the error message I mentioned above would appear again.

rayee922, 2025-06-13:
sudo add-apt-repository ppa:v-launchpad-jochen-sprickerhof-de/pcl
sudo apt-get update
sudo apt-get install libpcl-all
[sudo] password for ruifeng: 
Repository: 'deb https://ppa.launchpadcontent.net/v-launchpad-jochen-sprickerhof-de/pcl/ubuntu/ jammy main'
Description:
This only exists for history reasons. Please use the PCL version included in Ubuntu:

sudo apt install libpcl-dev

---

The Point Cloud Library (or PCL) is a large scale, open project for point cloud processing.
The PCL framework contains numerous state-of-the art algorithms including filtering, feature estimation, surface reconstruction, registration, model fitting and segmentation.
PCL is released under the terms of the BSD license and is open source software. It is free for commercial and research use. We are financially supported by Willow Garage, NVidia, and Google.
More info: https://launchpad.net/~v-launchpad-jochen-sprickerhof-de/+archive/ubuntu/pcl
Adding repository.
Press [ENTER] to continue or Ctrl-c to cancel.
Found existing deb entry in /etc/apt/sources.list.d/v-launchpad-jochen-sprickerhof-de-ubuntu-pcl-jammy.list
Adding deb entry to /etc/apt/sources.list.d/v-launchpad-jochen-sprickerhof-de-ubuntu-pcl-jammy.list
Found existing deb-src entry in /etc/apt/sources.list.d/v-launchpad-jochen-sprickerhof-de-ubuntu-pcl-jammy.list
Adding disabled deb-src entry to /etc/apt/sources.list.d/v-launchpad-jochen-sprickerhof-de-ubuntu-pcl-jammy.list
Adding key to /etc/apt/trusted.gpg.d/v-launchpad-jochen-sprickerhof-de-ubuntu-pcl.gpg with fingerprint 7A71F7294C7EEF1202C92D3D0A05879F75AE447E
Hit:1 http://gb.archive.ubuntu.com/ubuntu jammy InRelease
Hit:2 http://gb.archive.ubuntu.com/ubuntu jammy-updates InRelease              
Hit:3 http://gb.archive.ubuntu.com/ubuntu jammy-backports InRelease            
Hit:4 https://librealsense.intel.com/Debian/apt-repo jammy InRelease           
Hit:5 https://packages.microsoft.com/repos/code stable InRelease               
Hit:6 http://security.ubuntu.com/ubuntu jammy-security InRelease               
Hit:7 https://dl.google.com/linux/chrome/deb stable InRelease                  
Ign:8 https://ppa.launchpadcontent.net/v-launchpad-jochen-sprickerhof-de/pcl/ubuntu jammy InRelease
Err:9 https://ppa.launchpadcontent.net/v-launchpad-jochen-sprickerhof-de/pcl/ubuntu jammy Release
  404  Not Found [IP: 2620:2d:4000:1::81 443]
Hit:10 http://packages.osrfoundation.org/gazebo/ubuntu-stable jammy InRelease  
Hit:11 http://packages.ros.org/ros2/ubuntu jammy InRelease                     
Reading package lists... Done
E: The repository 'https://ppa.launchpadcontent.net/v-launchpad-jochen-sprickerhof-de/pcl/ubuntu jammy Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
W: http://packages.osrfoundation.org/gazebo/ubuntu-stable/dists/jammy/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
Hit:1 http://gb.archive.ubuntu.com/ubuntu jammy InRelease
Hit:2 http://gb.archive.ubuntu.com/ubuntu jammy-updates InRelease
Hit:3 http://gb.archive.ubuntu.com/ubuntu jammy-backports InRelease
Hit:4 https://librealsense.intel.com/Debian/apt-repo jammy InRelease
Hit:5 https://packages.microsoft.com/repos/code stable InRelease               
Ign:6 https://ppa.launchpadcontent.net/v-launchpad-jochen-sprickerhof-de/pcl/ubuntu jammy InRelease
Err:7 https://ppa.launchpadcontent.net/v-launchpad-jochen-sprickerhof-de/pcl/ubuntu jammy Release
  404  Not Found [IP: 2620:2d:4000:1::81 443]
Hit:8 http://security.ubuntu.com/ubuntu jammy-security InRelease               
Hit:9 https://dl.google.com/linux/chrome/deb stable InRelease                  
Hit:10 http://packages.ros.org/ros2/ubuntu jammy InRelease          
Hit:11 http://packages.osrfoundation.org/gazebo/ubuntu-stable jammy InRelease
Reading package lists... Done
E: The repository 'https://ppa.launchpadcontent.net/v-launchpad-jochen-sprickerhof-de/pcl/ubuntu jammy Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
W: http://packages.osrfoundation.org/gazebo/ubuntu-stable/dists/jammy/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package libpcl-all

MartyG-RealSense, 2025-06-14:
Have you previously had any problems with installing packages with **sudo apt-get** commands?

It looks as though you likely installed librealsense using used the method at the link below that was recommended in the PCL wrapper instructions.  Is that correct, please?

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

If you did use those instructions, does it make a difference if you build librealsense from source code using the **libuvc backend** method described here:

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

This installation method provides a pre-made build script called 'libuvc_installation.sh'.  You would need to edit line 46 of the script to add the `-DBUILD_PCL_EXAMPLES=true` flag to the CMake build instruction on that line.

https://github.com/IntelRealSense/librealsense/blob/master/scripts/libuvc_installation.sh#L46

rayee922, 2025-06-19:
I can run **rs-pcl-color** and obtain a .pcd file (Captured_Frame1.pcd). 
However, when I run **rs-pcl**, there is no image and it will show "No response".

rayee922, 2025-06-29:
Sorry, I'm working on another project. Please keep this case open, thanks!

MartyG-RealSense, 2025-06-29:
It's no problem to keep the case open.  Thanks very much for the update!

rayee922, 2025-07-24:
Hi,

I'm trying to generate a colored point cloud using ros2 bag, but when I record it, it prompts:

[WARN] [1753368140.143267271] [camera.D435i]: No stream match for pointcloud chosen texture Process-Color

I used these commands: ros2 launch realsense2_camera rs_launch.py camera_name:=D435i rgb_camera. color_profile:=1280x720x30  depth_module.depth_profile:=1280x720x30 enable_color:=true enable_depth:=true align_depth. enable:=true pointcloud.  enable:=true enable_sync:=true

[INFO] [launch]: All log files can be found below /home/ruifeng/.ros/log/2025-07-24-15-03-28-712549-Study-29201
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [29202]
[realsense2_camera_node-1] [INFO] [1753365808.847550639] [camera.D435i]: RealSense ROS v4.55.1
[realsense2_camera_node-1] [INFO] [1753365808.847614121] [camera.D435i]: Built with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1753365808.847628737] [camera.D435i]: Running with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1753365809.254966055] [camera.D435i]: Device with serial number 250122079158 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1753365809.255039324] [camera.D435i]: Device with physical ID 2-4-4 was found.
[realsense2_camera_node-1] [INFO] [1753365809.255061897] [camera.D435i]: Device with name Intel RealSense D435IF was found.
[realsense2_camera_node-1] [INFO] [1753365809.255450754] [camera.D435i]: Device with port number 2-4 was found.
[realsense2_camera_node-1] [INFO] [1753365809.255477552] [camera.D435i]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1753365809.255585315] [camera.D435i]: getParameters...
[realsense2_camera_node-1] [INFO] [1753365809.256059190] [camera.D435i]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1753365809.256085317] [camera.D435i]: Device Name: Intel RealSense D435IF
[realsense2_camera_node-1] [INFO] [1753365809.256102264] [camera.D435i]: Device Serial No: 250122079158
[realsense2_camera_node-1] [INFO] [1753365809.256122670] [camera.D435i]: Device physical port: 2-4-4
[realsense2_camera_node-1] [INFO] [1753365809.256154978] [camera.D435i]: Device FW version: 5.16.0.1
[realsense2_camera_node-1] [INFO] [1753365809.256171583] [camera.D435i]: Device Product ID: 0x0B3A
[realsense2_camera_node-1] [INFO] [1753365809.256181036] [camera.D435i]: Sync Mode: On
[realsense2_camera_node-1] [WARN] [1753365812.329431047] [camera.D435i]: re-enable the stream for the change to take effect.
[realsense2_camera_node-1] [INFO] [1753365812.329824754] [camera.D435i]: Set ROS param depth_module.infra_profile to default: 848x480x30
[realsense2_camera_node-1] [INFO] [1753365812.428211508] [camera.D435i]: Set ROS param gyro_fps to default: 200
[realsense2_camera_node-1] [INFO] [1753365812.428256374] [camera.D435i]: Set ROS param accel_fps to default: 100
[realsense2_camera_node-1] [INFO] [1753365812.432115212] [camera.D435i]: Stopping Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1753365812.432260346] [camera.D435i]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1753365812.550312402] [camera.D435i]: Starting Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1753365812.617949426] [camera.D435i]: Open profile: stream_type: Depth(0), Format: Z16, Width: 1280, Height: 720, FPS: 30
[realsense2_camera_node-1]  24/07 15:03:32,618 WARNING [140688826033728] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1] [INFO] [1753365812.619551943] [camera.D435i]: Starting Sensor: RGB Camera
[realsense2_camera_node-1]  24/07 15:03:32,628 WARNING [140688826033728] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1] [INFO] [1753365812.636557238] [camera.D435i]: Open profile: stream_type: Color(0), Format: RGB8, Width: 1280, Height: 720, FPS: 30
[realsense2_camera_node-1] [INFO] [1753365812.637989349] [camera.D435i]: RealSense Node Is Up!
[realsense2_camera_node-1]  24/07 15:03:32,940 WARNING [140688826033728] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  24/07 15:03:32,951 WARNING [140688826033728] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  24/07 15:03:32,961 WARNING [140688826033728] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  24/07 15:03:33,072 WARNING [140688826033728] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  24/07 15:03:33,284 WARNING [140688826033728] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  24/07 15:03:33,294 WARNING [140688826033728] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  24/07 15:04:31,142 WARNING [140688826033728] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[... cut ...]
It is the record topic: ros2 bag record/camera/D435i/color/image_raw/camera/D435i/the depth/color/points /camera/D435i/aligned_depth_to_color/image_raw  /camera/D435i/color/camera_info  /camera/D435i/aligned_depth_to_color/camera_info /camera/D435i/depth/camera_info /camera/D435i/depth/image_rect_raw  /camera/D435i/color/camera_info

Finally, I used open3d to save the point cloud image, but the resulting image only had one color and did not have the original color image. Is this situation due to frame drops or is it because the depth image and the color image are not aligned?

[image]

MartyG-RealSense, 2025-07-24:
The message `No stream match for pointcloud chosen texture Process-Color` indicates that color frames are missing, possibly because the frames are being dropped.

`Resource temporarily unavailable, number: 11` could indicate that the computer's CPU is having difficulty with keeping up with processing the volume of data arriving from the camera at the computer's USB port.

It is not necessary to enable align_depth unless you require the aligned topics to be published, because the point cloud will automatically map depth and color together if both streams are enabled.

Does performance improve if you use a simpler launch command with a lower resolution and FPS speed and also reset the camera hardware during launch with **initial_reset**

`ros2 launch realsense2_camera rs_launch.py camera_name:=D435i rgb_camera. color_profile:=640x480x30 depth_module.depth_profile:=640x480x6 enable_color:=true enable_depth:=true pointcloud. enable:=true initial_reset:=true`

MartyG-RealSense, 2025-07-24:
If the pointcloud is being published correctly in RViz then it is safe to ignore the **resource temporarily unavailable** warnings even if they are generating continuously in the log.

closed 2025-09-07 by MartyG-RealSense