realsenseai/librealsense #14070: RealSense IMU/Motion Module: Failed to open device sensor

milan-r-shah, 2025-06-16:
| Required Info                         |                                                     |
|----------------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 5.12.14.50 | 
| Operating System & Version  | Linux (Ubuntu 20) | 
| Kernel Version (Linux Only)   |  5.15.0 | 
| Platform                                 | Computer  |
| SDK Version                          | 2.44.0 | 
| Language                              | C++ | 
| Segment			        | {Robot/Smartphone/VR/AR/others } | 

### Issue Description
On my machine, the `realsense-viewer` works without any issues. In the viewer, I'm able to see depth image, RGB image, as well as Accelerometer and Gyroscope data.
However, when I try to access the IMU, depth camera, and RGB camera data, using my own code, I get the following error:
```
WARNING: RealSense IMU/Motion Module could not be opened: 'Failed to open device sensor. /sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.5/0003:8086:0B3A.000E/HID-SENSOR-200073.3.auto/iio:device1 Last Error: No such file or directory'
```
If I don't enable the IMU/Motion Module, I don't get the aforementioned error!

Notes:
1. I'm using the D430 camera module, not the D435 camera. 
2. I have seen this error before, as referenced in #12452

MartyG-RealSense, 2025-06-17:
Hi @milan-r-shah  If you are using the D430 module, is your Vision Processor D4 board the **V3** model equipped with an IMU?  A D430 with a V3 board attached should be detected as a D430i.  It would not be detected as a D435i as the D430 module does not have an RGB sensor attached - in the D435 and D435i camera models the RGB sensor is a separate component that is attached via a cable.  

So I'm not certain where an RGB image in the Viewer would be coming from - perhaps an RGB built-in webcam if your computer is a laptop.

Are depth and gyro / accel able to be displayed if you run the RealSense SDK's C++ example program **rs-capture** if your librealsense installation has the example programs installed?

MartyG-RealSense, 2025-06-18:
Are you enabling depth, RGB and IMU with **cfg** configuration instructions, please?  If you are, is the IMU able to start if you remove the reference for your cfg lines from the pipe start instruction?  For example, changing **pipe.start(cfg)** to **pipe.start()**

If the camera firmware driver can detect the presence of the IMU then when **pipe.start()** is used the camera should enable the default stream profiles for depth, RGB and IMU automatically.  On a D430 with RGB and IMU, that would be 848x480 depth, 1280x720 RGB, 63 Accel and 200 Gyro.

As your Accel is detected as 63, this indicates that your Vision Processor board is likely quite old, as from mid 2022 onwards the IMU component was changed to one that has 100 as its minimum Accel instead of 63.

milan-r-shah, 2025-06-24:
Hi @MartyG-RealSense, yes, I'm still working on the suggestions you gave me, as well as coming up with a minimal code example that I can show you to reproduce the error I've been getting!

MartyG-RealSense, 2025-06-25:
Thanks very much for the update.  I am not able to test scripts on my computer though, unfortunately.

I look forward to your next update after your tests.  Good luck!

MartyG-RealSense, 2025-07-18:
It's totally fine to close this issue for now and re-open it at a future date when you are able to work on it.