realsenseai/librealsense #14085: Random Segmentation faults when using RealSense SDK

GirishiF, 2025-06-19:
**Required Info:**
Camera Model [D450]
Firmware Version (Open RealSense Viewer --> Click info)
Operating System & Version {Win (8.1/10) / Linux (Ubuntu 14/16/17)
Kernel Version (Linux Only) (e.g. 4.14.13)
Platform PC/Raspberry Pi/ NVIDIA Jetson / Qualcomm/etc..
SDK Version {legacy / 2.. }
Language {C/C#/LabVIEW/NodeJS/OpenCV/pcl/python/unity }
Segment {Robot/Smartphone/VR/AR/others }

**Description:**
Hello @MartyG-RealSense, we are experiencing random **Segmentation Faults** when using RealSense SDK on a battery-powered Arm system.

**Some details:**
- SDK version 2.38.1 and 2.54.1
- FW version 5.15.000.2
- OS: Ubuntu 20.04
- Arch: Arm64

We are getting only the following information in stack trace:

```
libusb: debug [handle_events] poll() returned 1

[Caught signal 11 (Segmentation fault)]

Registers:

  RIP = 0x7f000002f0

  RSP = 0x7f6bffbf90

[CrashReporter] Stack trace (symbol → file:line):

  [STACK 00] 

  [STACK 01]

  [STACK 02] linux-vdso.so.1(__kernel_rt_sigreturn+0) [0x7faf2205b4]

  [STACK 03] [0x7f000002f0]
```

Please let us know your views and/or suggestions to debug and resolve this. Thanks.

MartyG-RealSense, 2025-06-19:
Hi @GirishiF  The log line `linux-vdso.so.1(__kernel_rt_sigreturn+0)` suggests that you might have a **realtime** (RT) kernel.  If you do - such kernels typically have 'rt' in their version name - then these can cause problems when used with librealsense.  https://github.com/IntelRealSense/librealsense/issues/2395#issuecomment-426536906 has advice from one of my Intel RealSense colleagues about this.

MartyG-RealSense, 2025-07-05:
Hi @GirishiF  Do you require further assistance with this case, please?  Thanks!

GirishiF, 2025-07-06:
Hello @MartyG-RealSense 
We are trying the libuvc backend installation approach for realsense sdk. We are trying to check if this fixes the segfault issue. We will keep you posted.

MartyG-RealSense, 2025-07-06:
Thanks very much, @GirishiF - I look forward to your next report.  Good luck!

MartyG-RealSense, 2025-07-28:
You are very welcome.  Please do feel free to re-open the thread at a future date if you need to.  Good luck!

closed 2025-07-28 by MartyG-RealSense