realsenseai/librealsense #14088: 18:34:45 [None] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\common\notifications.cpp:538 - Frame Metadata is a device feature allowing software synchronization between different camera streams. It must be explicitly enabled on Windows OS

bolt2012-china, 2025-06-20:
| | |
|---|---|
|**librealsense**|2.56.3 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense D435I|
|**Serial Number**|043322074365|
|**Firmware Version**|5.16.0.1|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|041323050750|
|**Firmware Update Id**|041323050750|
|**Dfu Device Path**||

Please provide a description of the problem

MartyG-RealSense, 2025-06-20:
Hi @bolt2012-china  You can easily enable frame metadata on Windows in the realsense-viewer tool.  When realsense-viewer is launched, the top corner of the program window should have a message in a pop-up box that appears and asks if you want to enable metadata.  Click the button on the pop-up box to enable metadata.

https://github.com/IntelRealSense/librealsense/issues/6433 has further information about this subject.

***********

您好 @bolt2012-china。您可以在 Windows 系统中通过 realsense-viewer 工具轻松启用帧元数据。启动 realsense-viewer 后，程序窗口顶部会弹出一个消息框，询问您是否要启用元数据。点击弹出框上的按钮即可启用元数据。

https://github.com/IntelRealSense/librealsense/issues/6433 提供了更多关于此主题的信息。

bolt2012-china, 2025-06-20:
> Hi [@bolt2012-china](https://github.com/bolt2012-china) You can easily enable frame metadata on Windows in the realsense-viewer tool. When realsense-viewer is launched, the top corner of the program [...]

I'm so sorry but I failed to see such pop-up box in my program, should I re-settup?

MartyG-RealSense, 2025-06-20:
This feature is only in the realsense-viewer tool.  The pop-up box should look like this:

[image]

If it does not appear in the top corner of the realsense-viewer window under the exit X, you can try unplugging and plugging the camera a few times to see if it appears.

You could also try downloading and running the standalone **Intel.RealSense.Viewer.exe** version of realsense-viewer from the 'Assets' file list at the link below and see if the metadata pop-up appears when you use that version of the Viewer.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.55.1

[image]

*********

If the RealSense SDK is built from source code then it is possible to enable metadata by editing the registry values of Windows.  I would not recommend it though unless you are confident in registry editing, otherwise Windows could stop working if something goes wrong.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_windows.md#enabling-metadata-on-windows

********

However, it is not necessary to enable frame metadata in order for RealSense to work.  It just means that frame timestamps will be "software timestamps" that have times that are based on the computer's time clock if metadata is not enabled.

closed 2025-07-05 by MartyG-RealSense