realsenseai/librealsense #14092: On Ultra-wide displays, GUI does not render properly when opened.

omnisurfer, 2025-06-23:
| | |
|---|---|
|**librealsense**|2.55.1 RELEASE|
|**OS**|Linux|
|**Name**|Intel RealSense D455|
|**Serial Number**|037522250816|
|**Firmware Version**|5.16.0.1|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|030223060173|
|**Firmware Update Id**|030223060173|
|**Dfu Device Path**||

On an ultra-wide Samsung display set at a resolution of 5120x1440, when starting the RealSense viewer, the GUI fails to render at all. Instead a window with a plain gray canvas displays. Interacting with the canvas does slightly change the color to a lighter gray color but none of the GUI controls renders at all.

To get the GUI to render, I must change my resolution to 1920x1080, open the viewer (which renders properly), then change my resolution back to 5120x1440.

PRETTY_NAME="Ubuntu 22.04.5 LTS"
NAME="Ubuntu"
VERSION_ID="22.04"
VERSION="22.04.5 LTS (Jammy Jellyfish)"

omnisurfer, 2025-06-28:
Hi @MartyG-RealSense ,

Sorry for the late reply, I am playing with my RealSense device off and on so my reply may be delayed. I am also not under any deadline so no rush on this issue, I just wanted to document it in case someone else was seeing it. 

Regarding toggling F8 (full screen), this does not seem to change the behavior I am observing. I am not sure if this helps or is related but I am also noticing similar behavior with the simulation environment called Gazebo that is associated with the Robot Operating System, version 2. The "fix" is to change resolution to a typical aspect ratio (16:9), open the app, then, restore my ultra-widescreen resolution/aspect.

I think maybe this might point to an issue with a common GUI middleware like QT or maybe OpenGL??

MartyG-RealSense, 2025-06-29:
Thanks very much for the information.  I will highlight this issue with high resolution displays to my Intel RealSense colleagues on the development team.