realsenseai/librealsense #14094: Issue when tried to Build librealsense

brostami85, 2025-06-23:
I tried to install librealsense on my MacBook Pro (Sequoia 15.5) based on the link below that was kindly provided by Intel Support Service team: https://lightbuzz.com/realsense-macos/

Everything went well and I could create and see the file "librealsense2.xcodeproj" inside the build folder of the RealSense repository, however the next step failed. In fact, when I followed the rest of the document to change the build settings inside Xcode and then pressed "cmd + B" it showed "Build Failed" and generated the error that I have attached to this message.

[image]

I appreciate if you could help with that please then I can install and run RealSense on my mac. 

Thank you.

brostami85, 2025-06-24:
Thank you for your response, @MartyG-RealSense . 
I understand. By providing this link (https://formulae.brew.sh/formula/librealsense) you meant to simply run this command in my mac terminal: brew install librealsense ? If so, I did it and you could find the outcome in the attached image. Would you please let me know what would be the next steps? Thank you.

[image]

MartyG-RealSense, 2025-06-25:
I have not personally used **brew install librealsense** but a RealSense Mac user provides advice at https://github.com/IntelRealSense/librealsense/issues/2811#issuecomment-443909901 on what to do next after installation.

closed 2025-07-27 by MartyG-RealSense