realsenseai/librealsense #14111: D455 cannot get imu data on jetson orin nano

104kpf, 2025-07-04:
jetpack: R36.4.4（jetpack6.2）
OS: Linux 22.04
SDK: Librealsense v2.55.1

I have already follow this guide https://github.com/IntelRealSense/librealsense/issues/13725#issuecomment-2780697359, and it fix the detection problem, but when I try to display the imu data it will show no frames or motion module force pause, although both RGB and stereo module both work well

[image]

[image]

MartyG-RealSense, 2025-07-04:
Hi @104kpf  If a **Motion Module Force Pause** error occurs in the RealSense Viewer when enabling the Motion Module, enabling it a second time after the error has occurred may sometimes cause the IMU streams to start correctly.  Is the Motion Module able to start if you enable it again immediately after the error, please?

MartyG-RealSense, 2025-07-04:
You are very welcome.  It's great to hear that it worked for you!

Your workflow looks fine.  You can achieve the same outcome in a simpler way though by using the **libuvc_installation.sh** build script at the link below and editing line 46 to add the list of CMake flags.

https://github.com/IntelRealSense/librealsense/blob/master/scripts/libuvc_installation.sh

104kpf, 2025-07-04:
OK, thank you for the suggestion!
I’ve also tested ros2 topic echo, and the IMU data is now streaming smoothly.

Since the issue is resolved, I’ll go ahead and close this. Appreciate your help!