realsenseai/librealsense #14118: About python and sdk video rotation angle problem

Tianhao-Wu7, 2025-07-08:
I am using a D405 depth camera. Due to the depth camera's socket, the image is upside down after I integrate it into the system. I need to rotate it 180° before I can use it directly. How can I achieve this in Python code? Below is my main program code. In addition, do I have a function to rotate the video angle in the SDK?

def main(arguments=None):
# Get connected devices
try:
device = ctx.query_devices()[0]
except IndexError:
print('No device connection detected')
sys.exit(1)

# Configure depth stream
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 848, 480, rs.format.z16, 30)

# Start streaming
profile = pipeline.start(config)
depth_sensor = profile.get_device().first_depth_sensor()
# Auto exposure
depth_sensor.set_option(rs.option.enable_auto_exposure, 1)
temp_filter = rs.temporal_filter(0.5, 20, 0) # default 0.1
spatial = rs.spatial_filter() 
spatial.set_option(rs.option.filter_magnitude, 1) 
spatial.set_option(rs.option.filter_smooth_alpha, 0.5) 
spatial.set_option(rs.option.filter_smooth_delta, 20) 
threshold_filter = rs.threshold_filter(0, 0.8) 
HDRmerge = rs.hdr_merge() 
sequenceid = rs.sequence_id_filter() 

try: 
while True: 

# Wait for depth frame 
frames = pipeline.wait_for_frames() 
# frames = rs.align(align_to).process(frames) 
depth_frame = frames.get_depth_frame() 
print("depth_frame:", depth_frame) 
if not depth_frame: 
continue 
height = depth_frame.get_height() 
width = depth_frame.get_width() 
depth_data = np.frombuffer(depth_frame.get_data(), dtype=np.uint16) 
depth_data = depth_data.reshape((height, width)) 

depth_sensor = profile.get_device().first_depth_sensor() 
depth_scale = depth_sensor.get_depth_scale() 
depth_data = depth_data.astype(np.float32) * depth_scale 

center_y = int(height / 2) 
center_x = int(width / 2) 
center_distance = depth_data[center_y, center_x] 
print("center distance: ", center_distance, "m") 

# Post-processing 
depth_frame = HDRmerge.process(depth_frame) 
depth_frame = sequenceid.process(depth_frame) 
depth_frame = threshold_filter.process(depth_frame) 
depth_frame = spatial.process(depth_frame) 
depth_frame = temp_filter.process(depth_frame) 
# Convert depth image to numpy Array
depth_colormap = np.asanyarray(depth_frame.get_data())

# Histogram equalization
clahe = cv2.createCLAHE(clipLimit=4, tileGridSize=(8, 8))
depth_colormap = clahe.apply(depth_colormap)

# Gamma correction
gamma = 1
# Normalize to [0, 1] range
depth_colormap_normalized = depth_colormap / 255.0
# Apply gamma correction
depth_colormap = np.power(depth_colormap_normalized, gamma)
# Restore to [0, 255] range and convert to uint8 type
depth_colormap = (depth_colormap * 255.0).astype(np.uint16)

# Apply color map to depth image (convert to 8-bit pixels first)
depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(depth_colormap, alpha=0.03), cv2.COLORMAP_JET)

# Display depth color map
cv2.namedWindow('RealSense', cv2.WINDOW_AUTOSIZE)
cv2.imshow('RealSense', depth_colormap)
cv2.waitKey(1)

finally:
# Stop the flow
pipeline.stop()

MartyG-RealSense, 2025-07-08:
Hi @Tianhao-Wu7  A post-processing Rotation filter has been implemented in the **development** branch of the librealsense SDK, as described at https://github.com/IntelRealSense/librealsense/pull/13499

https://github.com/IntelRealSense/librealsense/tree/development

The development branch forms the basis of what will be in the next  release of librealsense, so it should also be in the next SDK version when it is released.

The rotation filter has also been implemented in the RealSense ROS2 wrapper.

The C++ version of the official post-processing example program on the development branch has been updated to demonstrate the rotation filter and shows that the filter's name is **rotation_filter** - so you should be able to use that as a basis for implementing it in your Python post-processing code.

https://github.com/IntelRealSense/librealsense/blob/development/examples/post-processing/rs-post-processing.cpp#L82

In Python, the C++ post-processing code may convert to something like this:

```
rot_filter = rs.rotation_filter()

depth_frame = rot_filter.process(depth_frame)
```

Tianhao-Wu7, 2025-07-08:
There is no rotation_filter function declaration in filter.py of pyrealsense2, nor does it appear in other py scripts.
# encoding: utf-8
# module pyrealsense2.pyrealsense2
# from D:\python310\lib\site-packages\pyrealsense2\pyrealsense2.cp310-win_amd64.pyd
# by generator 1.147
"""
LibrealsenseTM Python Bindings
        ==============================
        Library for accessing Intel RealSenseTM cameras
"""

# imports
import pybind11_builtins as __pybind11_builtins

from .processing_block import processing_block

from .filter_interface import filter_interface

class filter(processing_block, filter_interface):
    """ Define the filter workflow, inherit this class to generate your own filter. """
    def as_decimation_filter(self): # real signature unknown; restored from __doc__
        """ as_decimation_filter(self: pyrealsense2.pyrealsense2.filter) -> rs2::decimation_filter """
        pass

    def as_disparity_transform(self): # real signature unknown; restored from __doc__
        """ as_disparity_transform(self: pyrealsense2.pyrealsense2.filter) -> rs2::disparity_transform """
        pass

    def as_hdr_merge(self): # real signature unknown; restored from __doc__
        """ as_hdr_merge(self: pyrealsense2.pyrealsense2.filter) -> rs2::hdr_merge """
        pass

    def as_hole_filling_filter(self): # real signature unknown; restored from __doc__
        """ as_hole_filling_filter(self: pyrealsense2.pyrealsense2.filter) -> rs2::hole_filling_filter """
        pass

    def as_sequence_id_filter(self): # real signature unknown; restored from __doc__
        """ as_sequence_id_filter(self: pyrealsense2.pyrealsense2.filter) -> rs2::sequence_id_filter """
        pass

    def as_spatial_filter(self): # real signature unknown; restored from __doc__
        """ as_spatial_filter(self: pyrealsense2.pyrealsense2.filter) -> rs2::spatial_filter """
        pass

    def as_temporal_filter(self): # real signature unknown; restored from __doc__
        """ as_temporal_filter(self: pyrealsense2.pyrealsense2.filter) -> rs2::temporal_filter """
        pass

    def as_threshold_filter(self): # real signature unknown; restored from __doc__
        """ as_threshold_filter(self: pyrealsense2.pyrealsense2.filter) -> rs2::threshold_filter """
        pass

    def is_decimation_filter(self): # real signature unknown; restored from __doc__
        """ is_decimation_filter(self: pyrealsense2.pyrealsense2.filter) -> bool """
        return False

    def is_disparity_transform(self): # real signature unknown; restored from __doc__
        """ is_disparity_transform(self: pyrealsense2.pyrealsense2.filter) -> bool """
        return False

    def is_hdr_merge(self): # real signature unknown; restored from __doc__
        """ is_hdr_merge(self: pyrealsense2.pyrealsense2.filter) -> bool """
        return False

    def is_hole_filling_filter(self): # real signature unknown; restored from __doc__
        """ is_hole_filling_filter(self: pyrealsense2.pyrealsense2.filter) -> bool """
        return False

    def is_sequence_id_filter(self): # real signature unknown; restored from __doc__
        """ is_sequence_id_filter(self: pyrealsense2.pyrealsense2.filter) -> bool """
        return False

    def is_spatial_filter(self): # real signature unknown; restored from __doc__
        """ is_spatial_filter(self: pyrealsense2.pyrealsense2.filter) -> bool """
        return False

    def is_temporal_filter(self): # real signature unknown; restored from __doc__
        """ is_temporal_filter(self: pyrealsense2.pyrealsense2.filter) -> bool """
        return False

    def is_threshold_filter(self): # real signature unknown; restored from __doc__
        """ is_threshold_filter(self: pyrealsense2.pyrealsense2.filter) -> bool """
        return False

    def __bool__(self): # real signature unknown; restored from __doc__
        """ __bool__(self: pyrealsense2.pyrealsense2.filter) -> bool """
        return False

    def __init__(self, filter_function, *args, **kwargs): # real signature unknown; NOTE: unreliably restored from __doc__ 
        """ __init__(self: pyrealsense2.pyrealsense2.filter, filter_function: Callable[[pyrealsense2.pyrealsense2.frame, pyrealsense2.pyrealsense2.frame_source], None], queue_size: int = 1) -> None """
        pass

    def __nonzero__(self): # real signature unknown; restored from __doc__
        """ __nonzero__(self: pyrealsense2.pyrealsense2.filter) -> bool """
        return False