realsenseai/librealsense #14129: RuntimeError: No device connected

zhang-qihao, 2025-07-10:
| Required Info              |                |
|----------------------------|----------------|
| Camera Model               | D435i      |
| Firmware Version           | 5.55.1       |
| Operating System & Version | Ubuntu 20.04  |
| Kernel Version (Linux Only)|  |
| Platform                   | LubanCat-4 V1 |
| SDK Version                | 2.55.1         |
| Language                   | Python         |

## Issue Description
Hello, I only wrote the following code and ran it,
```python
import pyrealsense2 as rs

pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 15)
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 15)

profile = pipeline.start(config)
```
 but it gave me an error . How can I troubleshoot and resolve this?
```bash
Traceback (most recent call last):
  File "/home/noetix/work/dev/realsense_project/test.py", line 13, in <module>
    profile = pipeline.start(config)
RuntimeError: No device connected
```

MartyG-RealSense, 2025-07-10:
Hi @zhang-qihao  Is the camera detected if you launch your program in **sudo** admin permissions mode, please?  For example:

`sudo python3 test.py`

MartyG-RealSense, 2025-07-10:
What method did you use to install librealsense, please?  On non-mainstream computing devices, such as your LubanCat, it is recommended to build librealsense from source code using the **libuvc backend** installation method at the link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

The libuvc method provides a pre-made build script called **libuvc_installation.sh**

https://github.com/IntelRealSense/librealsense/blob/master/scripts/libuvc_installation.sh

A RealSense user at https://github.com/IntelRealSense/librealsense/issues/14047 recently used a LubanCat 4 and was able to achieve a working pyrealsense2 setup.

zhang-qihao, 2025-07-12:
Thank you for your reply.  

Yes, I used the **libuvc backend** installation method.  

I will try to clear the previously configured environment and recompile the source code, as I didn't directly install the libuvc version at the first time.  

May I ask if there is any way to completely clean up all the configured SDK environments?

MartyG-RealSense, 2025-07-12:
For package installations, you can remove them with the command below.

`dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge`

To remove a source code installation, you can go to the 'build' directory of the librealsense source code folder that was used to install that SDK and input the CMake build command below to uninstall that SDK and clean up CMake:

`sudo make uninstall && make clean`

zhang-qihao, 2025-07-12:
Unfortunately, after attempting to clean the previous environment and recompile the source code, pyrealsense2 still cannot detect the camera.
I would greatly appreciate it if you could let me know what specific information I should provide to help with debugging. Thank you in advance!🙏

zhang-qihao, 2025-07-12:
Thank you very much for your response.

I will try downgrading the firmware tomorrow to see if it helps resolve the issue.

In addition, I have a couple of questions regarding the installation process.

I installed the `pyrealsense2` package using `pip`. What is the correct and recommended way to install the `pyrealsense2` package?

When building the SDK from source, does specifying the CMake flags `-DBUILD_PYTHON_BINDINGS=ON -DPYTHON_EXECUTABLE=$(which python3)` automatically generate the `pyrealsense2` Python bindings?

I sincerely appreciate your guidance on this.

MartyG-RealSense, 2025-07-12:
It appears that the LubanCat CPU chip is ARM architecture, so this means that the Python versions supported by the **pip install pyrealsense2** installation method will be 3.7 to 3.10.  If your Python 3 version is newer than 3.10 then the pyrealsense2 wrapper would have to be built from source code.

Yes, when building the SDK from source, specifying those CMake flags causes the pyrealsense2 wrapper to automatically be built from source code at the same time as the librealsense SDK.  The built pyrealsense2 files should then be able to be found in the **librealsense > build >wrappers > python** directory of the librealsense source code folder.

zhang-qihao, 2025-07-13:
> It appears that the LubanCat CPU chip is ARM architecture, so this means that the Python versions supported by the **pip install pyrealsense2** installation method will be 3.7 to 3.10. If your [...]

I do find the `pyrealsense2.cpython-38-aarch64-linux-gnu.so` file under the `build/release` directory, but I couldn't find any `.so` file under the `build/wrappers/python directory`.

Should I copy this `.so` file to the `site-packages` directory of Python 3? 

After copying it there, I tried running my test script again, but I still encountered the error: `RuntimeError: No device connected.`

MartyG-RealSense, 2025-07-13:
You are very welcome.  It's great to hear that you achieved a solution.  Thanks very much for the update and best wishes to you too!