realsenseai/librealsense #14141: intel RealSense d435 can`t run opencv_viewer_example.py

Willy1999927, 2025-07-18:
| Field                 | Value                           |
|----------------------|----------------------------------|
|Name                          |     Intel RealSense D435|
|    Serial Number                 |     317422076018|
|    Firmware Version              |     05.16.00.01|
|    Advanced Mode                 |     YES|
|    Usb Type Descriptor           |     3.2|
|    Product Line                  |     D400|
|    Asic Serial Number            |     320143021868|
|    Firmware Update Id           |      320143021868|
| Operating System     | Linux linaro-alip 6.1.99|
| Kernel Version       | 6.1.99                           |
| Programming Language | Python 3.10.18                       |

Hello,
I am currently running the example program opencv_viewer_example.py, but no matter what, I always get the following error:
```
Traceback (most recent call last):
  File "/userdata/camera_example/examples/opencv_viewer_example.py", line 39, in <module>
    frames = pipeline.wait_for_frames()
RuntimeError: Frame didn't arrive within 5000
```
I have confirmed that the connection is using USB 3.0:
```
/:  Bus 07.Port 1: Dev 1, Class=root_hub, Driver=ohci-platform/1p, 12M
/:  Bus 06.Port 1: Dev 1, Class=root_hub, Driver=ehci-platform/1p, 480M
    |__ Port 1: Dev 4, If 0, Class=Hub, Driver=hub/4p, 480M
/:  Bus 05.Port 1: Dev 1, Class=root_hub, Driver=ohci-platform/1p, 12M
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=ehci-platform/1p, 480M
    |__ Port 1: Dev 4, If 0, Class=Hub, Driver=hub/4p, 480M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 5000M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 5000M
        |__ Port 4: Dev 5, If 0, Class=Video, Driver=uvcvideo, 5000M
        |__ Port 4: Dev 5, If 1, Class=Video, Driver=uvcvideo, 5000M
        |__ Port 4: Dev 5, If 2, Class=Video, Driver=uvcvideo, 5000M
        |__ Port 4: Dev 5, If 3, Class=Video, Driver=uvcvideo, 5000M
        |__ Port 4: Dev 5, If 4, Class=Video, Driver=uvcvideo, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 480M
```

MartyG-RealSense, 2025-07-18:
Hi @Willy1999927  I note that you are using a type of Linux OS called **Linaro**.  When using an 'exotic' non-Ubuntu OS, problems are less likely to occur if the librealsense SDK is built from source code with the **libuvc backend** installation method.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

This method provides a pre-made installation script called **libuvc_installation.sh**

https://github.com/IntelRealSense/librealsense/blob/master/scripts/libuvc_installation.sh

To install the Python pyrealsense2 wrapper at the same time as librealsense, edit line 46 of the script to add the flag `-DBUILD_PYTHON_BINDINGS:bool=true` to the line.

***********

Before installing the SDK again though, you could first try **rebinding** the XHCI driver using the method described at https://github.com/IntelRealSense/librealsense/issues/6899#issuecomment-671984909

Another RealSense Pyrthon user who experienced the 'Frame didn't arrive within 5000' error also reported at https://github.com/IntelRealSense/librealsense/issues/14047#issuecomment-2955160963 that cleaning up their computer of previous installations by wiping and doing a fresh install of everything resolved the error in their particular situation.

Willy1999927, 2025-07-18:
@MartyG-RealSense  Thank you for your reply. I will first try rebinding the XHCI driver.

Willy1999927, 2025-07-21:
I have already run libuvc_installation.sh, but it doesn't work.
I also modified libuvc_installation.sh to make sure it runs inside the Conda environment.
'''
#!/bin/bash -xe

ENV_NAME="py3_10"

#Locally suppress stderr to avoid raising not relevant messages
exec 3>&2
exec 2> /dev/null
con_dev=$(ls /dev/video* | wc -l)
exec 2>&3

if [ $con_dev -ne 0 ];
then
        echo -e "\e[32m"
        read -p "Remove all RealSense cameras attached. Hit any key when ready"
        echo -e "\e[0m"
fi

lsb_release -a
echo "Kernel version $(uname -r)"
sudo apt-get update
cd ~/
sudo rm -rf ./librealsense_build
mkdir librealsense_build && cd librealsense_build

if [ $(sudo swapon --show | wc -l) -eq 0 ];
then
        echo "No swapon - setting up 1Gb swap file"
        sudo fallocate -l 2G /swapfile
        sudo chmod 600 /swapfile
        sudo mkswap /swapfile
        sudo swapon /swapfile
        sudo swapon --show
fi

echo Installing Librealsense-required dev packages
sudo apt-get install git cmake libssl-dev freeglut3-dev libusb-1.0-0-dev pkg-config libgtk-3-dev unzip -y
rm -f ./master.zip

wget https://github.com/IntelRealSense/librealsense/archive/master.zip
unzip ./master.zip -d .
cd ./librealsense-master

PY_EXEC=$(conda run -n "$ENV_NAME" which python)

echo Install udev-rules
sudo cp config/99-realsense-libusb.rules /etc/udev/rules.d/
sudo cp config/99-realsense-d4xx-mipi-dfu.rules /etc/udev/rules.d/
sudo udevadm control --reload-rules && sudo udevadm trigger
mkdir build && cd build
cmake ../ -DFORCE_LIBUVC=true -DCMAKE_BUILD_TYPE=release -DBUILD_PYTHON_BINDINGS:bool=true -DPYTHON_EXECUTABLE="$PY_EXEC"
make -j2
sudo make install
echo -e "\e[92m\n\e[1mLibrealsense script completed.\n\e[0m"

'''
I found another problem my system does not have the uvcvideo module installed.
I'm going to try installing it next. Is there any installation guide I can follow?

'''
(py3_10) linaro@linaro-alip:/userdata/librealsense/scripts$ dmesg | grep uvc
[    3.462161] usbcore: registered new interface driver uvcvideo
[    4.381197] uvcvideo 2-1.4:1.1: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[    4.490274] uvcvideo 2-1.4:1.4: Unknown video format 36315752-1a66-a242-9065-d01814a8ef8a
[    7.981794] uvcvideo 2-1.4:1.1: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[    7.985276] uvcvideo 2-1.4:1.4: Unknown video format 36315752-1a66-a242-9065-d01814a8ef8a
[   11.597712] uvcvideo 2-1.4:1.1: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[   11.600469] uvcvideo 2-1.4:1.4: Unknown video format 36315752-1a66-a242-9065-d01814a8ef8a
[ 4608.909935] uvcvideo 2-1.4:1.1: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[ 4608.913243] uvcvideo 2-1.4:1.4: Unknown video format 36315752-1a66-a242-9065-d01814a8ef8a
[235472.477747] uvcvideo 2-1.4:1.1: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[235472.480764] uvcvideo 2-1.4:1.4: Unknown video format 36315752-1a66-a242-9065-d01814a8ef8a
[238642.669514] uvcvideo 2-1.4:1.1: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[238642.672785] uvcvideo 2-1.4:1.4: Unknown video format 36315752-1a66-a242-9065-d01814a8ef8a
[238790.724419] uvcvideo 2-1.4:1.1: Non-zero status (-71) in video completion handler.
[238996.591239] uvcvideo 2-1.4:1.4: Non-zero status (-71) in video completion handler.
[239033.740901] uvcvideo 2-1.4:1.1: Non-zero status (-71) in video completion handler.
[239033.742465] uvcvideo 2-1.4:1.4: Non-zero status (-71) in video completion handler.
[239038.743626] uvcvideo 2-1.4:1.1: Failed to resubmit video URB (-1).
[239164.870608] uvcvideo 2-1.4:1.1: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[239164.890901] uvcvideo 2-1.4:1.1: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[239164.923920] uvcvideo 2-1.4:1.1: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[239164.962502] uvcvideo 2-1.4:1.1: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[239165.006431] uvcvideo 2-1.4:1.4: Unknown video format 36315752-1a66-a242-9065-d01814a8ef8a
[239273.789867] uvcvideo 2-1.4:1.1: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[239273.793851] uvcvideo 2-1.4:1.4: Unknown video format 36315752-1a66-a242-9065-d01814a8ef8a
'''

Willy1999927, 2025-07-21:
"I tried running dmeg -w to check where the problem is, and it seems to be related to the USB."
```
  801.028643] uvcvideo 2-1.4:1.4: Non-zero status (-71) in video completion handler.
[  801.333294] usb 2-1.4: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[  801.344175] usb 2-1.4: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[  801.354749] usb 2-1.4: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).

```

Willy1999927, 2025-07-22:
Alright, I understand. Thank @MartyG-RealSense

Willy1999927, 2025-07-22:
I have recently tried running version 2.54.1 and found that it works, but the newer version does not. For now, I will continue using version 2.54.1 as a temporary workaround. Thank you.

closed 2025-07-22 by Willy1999927