realsenseai/librealsense #14146: Compatibility with CMake < 3.10 will be removed from a future version of cmake

rhabacker, 2025-07-21:
Building librealsense with cmake version 4.0.3 gives many messages like this:

```
[cmake] CMake Deprecation Warning at CMakeLists.txt:2 (cmake_minimum_required):
[cmake] Compatibility with CMake < 3.10 will be removed in a future version of
 [cmake] CMake.
``'

Since some included cmake scripts already depend on cmake 3.10 or a newer version, see

```
.github/workflows/rs-all-client/CMakeLists.txt:3:cmake_minimum_required( VERSION 3.15 )
unit-tests/unit-test-config.py:120:cmake_minimum_required( VERSION 3.10.0 )
```

it would be helpful to update the cmake build system to use at least version 3.10.0.

Translated with DeepL.com (free version)

MartyG-RealSense, 2025-07-21:
Hi @rhabacker  The librealsense SDK typically updates its minimum requirements for software components once a certain version of that component officially becomes End of Life (EOL).  An example of this is removing support for specific old versions of ROS2 or Python 3 as they reach EOL.

MartyG-RealSense, 2025-07-21:
My understanding is that having a suitable CMake version is left to the user.  For example, the main CMakeLists.txt build file of the SDK in its root folder lists CMake 3.1.0 as its minimum but accompanies it with a note that some Ubuntu versions require a newer CMake.

```
# Xenial requires 3.5, Bionic 3.10... 
cmake_minimum_required(VERSION 3.1.0)
```

MartyG-RealSense, 2025-07-30:
My research of the `Compatibility with CMake < 3.10 will be removed from a future version of CMake.` message indicates that it is not a RealSense-specific error but occurs in numerous different projects that are compiled with CMake.  Updating the minimum value parameter in CMakeLists.txt files can resolve it.

There has not been any other requests from the RealSense user community to change the minimum CMake version though, so it may not be a high development priority for the librealsense SDK.

MartyG-RealSense, 2025-08-30:
Hi @rhabacker  As the RealSense development team have begun the process of updating the SDK's minimal CMake to 3.10, do you require further assistance with this case please?  Thanks!

closed 2025-09-07 by MartyG-RealSense