realsenseai/librealsense #14160: Build failure with default settings on openSUSE Leap 15.6

rhabacker, 2025-07-23:
I tried to compile librealsense with the default settings on openSUSE Leap 15.6, which resulted in the error:

```
/usr/lib64/gcc/x86_64-suse-linux/7/../../../../x86_64-suse-linux/bin/ld: ../../libcurl/libcurl_install/lib/libcurl.a(psl.c.o): in function `Curl_psl_destroy':
psl.c:(.text+0x24): undefined reference to `psl_free'
/usr/lib64/gcc/x86_64-suse-linux/7/../../../../x86_64-suse-linux/bin/ld: ../../libcurl/libcurl_install/lib/libcurl.a(psl.c.o): in function `Curl_psl_use':
psl.c:(.text+0xb9): undefined reference to `psl_latest'
/usr/lib64/gcc/x86_64-suse-linux/7/../../../../x86_64-suse-linux/bin/ld: psl.c:(.text+0xfb): undefined reference to `psl_free'
/usr/lib64/gcc/x86_64-suse-linux/7/../../../../x86_64-suse-linux/bin/ld: psl.c:(.text+0x16f): undefined reference to `psl_builtin'
/usr/lib64/gcc/x86_64-suse-linux/7/../../../../x86_64-suse-linux/bin/ld: ../../libcurl/libcurl_install/lib/libcurl.a(cookie.c.o): in function `Curl_cookie_add':
cookie.c:(.text+0x79b): undefined reference to `psl_is_cookie_domain_acceptable'
/usr/lib64/gcc/x86_64-suse-linux/7/../../../../x86_64-suse-linux/bin/ld: ../../libcurl/libcurl_install/lib/libcurl.a(idn.c.o): in function `Curl_idn_decode':
idn.c:(.text+0x5d): undefined reference to `idn2_check_version'
/usr/lib64/gcc/x86_64-suse-linux/7/../../../../x86_64-suse-linux/bin/ld: idn.c:(.text+0x74): undefined reference to `idn2_lookup_ul'
/usr/lib64/gcc/x86_64-suse-linux/7/../../../../x86_64-suse-linux/bin/ld: idn.c:(.text+0x91): undefined reference to `idn2_free'
/usr/lib64/gcc/x86_64-suse-linux/7/../../../../x86_64-suse-linux/bin/ld: idn.c:(.text+0xbe): undefined reference to `idn2_lookup_ul'
/usr/lib64/gcc/x86_64-suse-linux/7/../../../../x86_64-suse-linux/bin/ld: ../../libcurl/libcurl_install/lib/libcurl.a(idn.c.o): in function `Curl_idn_encode':
idn.c:(.text+0x10c): undefined reference to `idn2_to_unicode_8z8z'
/usr/lib64/gcc/x86_64-suse-linux/7/../../../../x86_64-suse-linux/bin/ld: idn.c:(.text+0x145): undefined reference to `idn2_free'
/usr/lib64/gcc/x86_64-suse-linux/7/../../../../x86_64-suse-linux/bin/ld: ../../libcurl/libcurl_install/lib/libcurl.a(idn.c.o): in function `Curl_idnconvert_hostname':
idn.c:(.text+0x1e2): undefined reference to `idn2_check_version'
/usr/lib64/gcc/x86_64-suse-linux/7/../../../../x86_64-suse-linux/bin/ld: idn.c:(.text+0x1f9): undefined reference to `idn2_lookup_ul'
/usr/lib64/gcc/x86_64-suse-linux/7/../../../../x86_64-suse-linux/bin/ld: idn.c:(.text+0x216): undefined reference to `idn2_free'
/usr/lib64/gcc/x86_64-suse-linux/7/../../../../x86_64-suse-linux/bin/ld: idn.c:(.text+0x24e): undefined reference to `idn2_lookup_ul'
collect2: error: ld returned 1 exit status
```

# How to reproduce
```
git clone  git@github.com:IntelRealSense/librealsense.git
git -C librealsense checkout development
cmake -B librealsense-build  -S librealsense
cmake --build librealsense-build 
```
cmake --version
cmake version 4.0.3

# logs

[cmake.log](https://github.com/user-attachments/files/21387651/cmake.log)
[build.log](https://github.com/user-attachments/files/21387652/build.log)

MartyG-RealSense, 2025-07-23:
Hi @rhabacker  libcurl errors should be avoided if you compile from source with the CHECK_FOR_UPDATES build flag disabled.

`cmake ../ -DCHECK_FOR_UPDATES=OFF`

The consequence of disabling this will simply be that you will not receive suggestions for online updates.

rhabacker, 2025-07-23:
Thanks for this pointer. 

And to build with "suggestion for online updates" included the following patch may help:

```
diff --git a/CMake/external_libcurl.cmake b/CMake/external_libcurl.cmake
index 61f804814..733c8b258 100644
--- a/CMake/external_libcurl.cmake
+++ b/CMake/external_libcurl.cmake
@@ -57,7 +57,7 @@ if(CHECK_FOR_UPDATES)
         target_link_libraries(curl INTERFACE ws2_32.lib crypt32.lib)
     else()
         find_package(OpenSSL REQUIRED)
-        target_link_libraries(curl INTERFACE OpenSSL::SSL OpenSSL::Crypto)
+        target_link_libraries(curl INTERFACE OpenSSL::SSL OpenSSL::Crypto idn2 psl)
         if (APPLE)
             target_link_libraries(curl INTERFACE "-framework CoreFoundation -framework SystemConfiguration")
         endif()
```

MartyG-RealSense, 2025-07-23:
Thanks very much for the information!

closed 2025-08-08 by MartyG-RealSense