realsenseai/librealsense #14162: pyrealsens2 align problem

Willy1999927, 2025-07-24:
|Field | Value|
|-- | --|
|Name | Intel RealSense D435
|Serial Number | 317422076018
|Firmware Version | 05.16.00.01
|Advanced Mode | YES
|Usb Type Descriptor | 3.2
|Product Line | D400
|Asic Serial Number | 320143021868
|Firmware Update Id | 320143021868
|Operating System | Linux linaro-alip 6.1.99
|Kernel Version | 6.1.99
|Programming Language | Python 3.10.18
l|ibrealsense SDK | 2.54.1

I set the camera to 
```
self.align = rs.align(rs.stream.depth) 
```
and enabled the align function, but I noticed that the resulting color image appears distorted. Is there a way to fix this?

詢問 ChatGPT

Willy1999927, 2025-07-24:
[image]

MartyG-RealSense, 2025-07-24:
If you use `self.align = rs.align(rs.stream.depth)`  then you are performing **color to depth** alignment instead of the usual depth to color.  This means that the color image will be stretched in size to match the larger field of view that the D435's depth sensor has.  This occurs because on the D435 camera model the RGB color sensor has a smaller field of view than the depth sensor.

Does your image display correctly if you align depth to color with `self.align = rs.align(rs.stream.color)`

closed 2025-08-08 by MartyG-RealSense