realsenseai/librealsense #14164: D415 suddenly showing up as D410

MathijsNL, 2025-07-24:
Hello,

So my D415 suddenly showed up as D410. How do I know what the cause of this issue is? And is there a solution?

Today out of nowhere my camera did not give rgb output anymore.

I found these issues from some time ago. I tried most of the things mentioned in #11963 and #13785 

I upgraded and downgraded the firmware a few times, this did not resolve the issue.

Finally this https://github.com/IntelRealSense/librealsense/issues/3785#issuecomment-488639272 comment mentioned an rs-terminal command but that also didn't work.

```
rs-terminal -d 0 -s "14 00 ab cd 22 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00"

Device ID 0 has loaded.
 24/07 22:34:18,226 ERROR [127046536451968] (rs.cpp:256) [rs2_send_and_receive_raw_data( device:0x5815a8490f00 ) Invalid Value] get_xu(ctrl=1) failed! Last Error: Device or resource busy
 24/07 22:34:18,227 ERROR [127046536451968] (rs.cpp:256) [rs2_delete_raw_data( buffer:nullptr ) UNKNOWN] null pointer passed for argument "buffer"

get_xu(ctrl=1) failed! Last Error: Device or resource busy
```

The camera was originally attached on a Linux pc with Ubuntu 24.04 LTS kernel 6.8 but it has also been tested on a Windows laptop and there it shows up as D410.

I also read somewhere that this might be a ribbon cable, but I am not sure if looking inside will void warranty.

[fw_log.txt](https://github.com/user-attachments/files/21418798/fw_log.txt)

```
rs-enumerate-devices
Device info:
    Name                          :     Intel RealSense D410
    Serial Number                 :     319522065373
    Firmware Version              :     5.16.0.1
    Recommended Firmware Version  :     5.16.0.1
    Physical Port                 :     /sys/devices/pci0000:00/0000:00:14.0/usb4/4-1/4-1:1.0/video4linux/video0
    Debug Op Code                 :     15
    Advanced Mode                 :     YES
    Product Id                    :     0AD2
    Camera Locked                 :     YES
    Usb Type Descriptor           :     3.2
    Product Line                  :     D400
    Asic Serial Number            :     402443022346
    Firmware Update Id            :     402443022346
    Dfu Device Path               :
```
Full output: [enum.txt](https://github.com/user-attachments/files/21418743/enum.txt)

MartyG-RealSense, 2025-07-25:
Hi @MathijsNL   Yes, opening up the camera would void the warranty if you purchased it within the last 12 months.  If it was bought longer than a year ago then it will be outside of the warranty period and so opening the camera would be an option.  If the camera was working normally up until now though then it is more likely to be a firmware issue rather than a hardware one.

On the D415 model the RGB sensor is integrated on the camera's circuit board and not attached separately via a cable, so it would not be an internal RGB sensor cable problem.  

If there was a firmware related problem then the issue would affect all computers that the camera was used on, as the firmware driver is stored inside the camera hardware and not on individual computers.

Does the RGB start working if you downgrade the firmware to version **5.13.0.50**

As you have a D415, there is an alternative RGB mode available from the depth sensor called 'RGB from left infrared sensor' that does not require the detection of the RGB sensor.  To use it, you select the Infrared stream and set its format to RGB8 or BGR8 instead of the default Y8 format of infrared.

MathijsNL, 2025-07-25:
Thanks for your reply. The camera is about 2 weeks in my possesion. I tried downgrading the firmware but I think I have hit the firmware downgrade limit so I can't test the 5.13 firmware anymore.

The firmware has been updated to 5.16 as soon as I got it, and it has been used a lot the last 2 weeks without any changes to the firmware.

As the camera is still under warranty I will send it back to the supplier, but I would really like to know why this happened.

MartyG-RealSense, 2025-07-25:
A new SDK version (2.56.4) has just been released, so you could try installing **Intel.RealSense.Viewer.exe** on Windows and updating the camera to the latest **5.17.0.9** firmware in the Viewer (you can still upgrade to a newer firmware even when downgrades are locked after 20 downgrades).

https://github.com/IntelRealSense/librealsense/releases/tag/v2.56.4

[image]

If the camera is still detected as D410 then you could return it under warranty to the supplier that you purchased it from.

MathijsNL, 2025-07-25:
Updating to 5.17.0.9 is not possible:

```
UNKNOWN in rs2_update_firmware_cpp(device:00000228C2EE4190,
fw_image:00000228C79AE060):
Device: 402443022346 is locked for update. Use firmware version higher than:
5.17.0.9
```

I already tried this version and now this version is locked as well. In my opinion updating to highest version should always be possible.