realsenseai/librealsense #14169: realsense-viewer error

wzhaofeng, 2025-07-28:
When installing using the Debian method in the installation approach for Nvidia Jetson Devices, after the installation is completed, realsense-viewer can be started normally and the D457 can be recognized as the D455, but the following error is displayed:
17:03:25 [Info]  - librealsense version: 2.56.4

17:03:25 [Info] ./src/rs.cpp:3200 - Framebuffer size changed to 1792 x 1008
17:03:25 [Info] ./src/rs.cpp:3200 - Scale Factor is now 1
17:03:25 [Info] ./src/linux/backend-v4l2.cpp:675 - Cannot access /sys/class/d4xx-class
17:03:25 [Info] ./src/linux/backend-v4l2.cpp:675 - Cannot access /sys/class/d4xx-class
17:03:26 [Info] ./src/linux/backend-v4l2.cpp:675 - Cannot access /sys/class/d4xx-class
17:03:26 [Info] ./src/linux/backend-v4l2.cpp:675 - Cannot access /sys/class/d4xx-class
17:03:26 [Info] ./src/linux/backend-v4l2.cpp:675 - Cannot access /sys/class/d4xx-class
17:03:26 [Info] ./src/linux/backend-v4l2.cpp:675 - Cannot access /sys/class/d4xx-class
17:03:26 [Info] ./src/linux/backend-v4l2.cpp:675 - Cannot access /sys/class/d4xx-class
17:03:26 [Info] ./src/linux/backend-v4l2.cpp:675 - Cannot access /sys/class/d4xx-class
17:03:26 [Info] ./src/linux/backend-v4l2.cpp:675 - Cannot access /sys/class/d4xx-class
17:03:26 [Info] ./src/linux/backend-v4l2.cpp:675 - Cannot access /sys/class/d4xx-class
17:03:26 [Info] ./src/linux/backend-v4l2.cpp:675 - Cannot access /sys/class/d4xx-class
17:03:26 [Info] ./src/linux/backend-v4l2.cpp:675 - Cannot access /sys/class/d4xx-class
17:03:27 [Info] ./src/linux/backend-v4l2.cpp:675 - Cannot access /sys/class/d4xx-class
17:03:27 [Info] ./src/linux/backend-v4l2.cpp:675 - Cannot access /sys/class/d4xx-class
17:03:27 [Info] ./src/linux/backend-v4l2.cpp:675 - Cannot access /sys/class/d4xx-class
17:03:27 [Info] ./src/linux/backend-v4l2.cpp:675 - Cannot access /sys/class/d4xx-class
17:03:50 [Info] ./src/linux/backend-v4l2.cpp:675 - Cannot access /sys/class/d4xx-class
17:03:50 [Info] ./src/linux/backend-v4l2.cpp:775 - Enumerating UVC video0 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-3/2-3.3/2-3.3:1.0/video4linux/video0
17:03:50 [Info] ./src/linux/backend-v4l2.cpp:775 - Enumerating UVC video1 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-3/2-3.3/2-3.3:1.0/video4linux/video1
17:03:50 [Info] ./src/linux/backend-v4l2.cpp:775 - Enumerating UVC video2 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-3/2-3.3/2-3.3:1.0/video4linux/video2
17:03:50 [Info] ./src/linux/backend-v4l2.cpp:775 - Enumerating UVC video3 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-3/2-3.3/2-3.3:1.0/video4linux/video3
17:03:50 [Info] ./src/linux/backend-v4l2.cpp:775 - Enumerating UVC video4 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-3/2-3.3/2-3.3:1.3/video4linux/video4
17:03:50 [Info] ./src/linux/backend-v4l2.cpp:775 - Enumerating UVC video5 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-3/2-3.3/2-3.3:1.3/video4linux/video5
17:03:50 [Info] ./src/backend-device-factory.cpp:155 - Device connected: /sys/devices/platform/bus@0/3610000.usb/usb2/2-3/2-3.3/2-3.3:1.0/video4linux/video0
17:03:50 [Warn] ./src/ds/ds-motion-common.cpp:452 - No HID info provided, IMU is disabled
17:03:50 [Error] ./src/rs.cpp:256 - [rs2_create_device( info_list:0xffff9c0262b0, index:0 ) UNKNOWN] bad optional access
17:03:50 [Error] ./src/rs.cpp:256 - [rs2_delete_device( device:nullptr ) UNKNOWN] null pointer passed for argument "device"
17:03:50 [Warn] ./src/rs.cpp:392 - null pointer passed for argument "device"
17:03:50 [Warn] ./src/rs.cpp:3203 - Couldn't refresh devices - bad optional access

What's the reason for this and what are the solutions?

wzhaofeng, 2025-07-29:
My device is Jetson agx orin developer kit, LT4 version is 36.4.4, can I use the above method with this configuration?

MartyG-RealSense, 2025-07-29:
L4T 36.4.4 corresponds to JetPack version 6.2.1.

Yes, you can use the libuvc installation method with this configuration.

MartyG-RealSense, 2025-08-08:
Hi @wzhaofeng  Do you require further assistance with this case, please?  Thanks!

skandvarma, 2025-08-13:
Alright let me try the firmware downgrade , my current camera is 435i and firmware version is Intel RealSense D435I firmware: 5.16.0.1

Also i would be really delighted if you guided me through the firmware downgrading process ! 

Thanks!

MartyG-RealSense, 2025-08-13:
https://github.com/IntelRealSense/realsense-ros/issues/3416#issuecomment-3182546563 has instructions for downgrading to 5.13.0.50.  It's the step I always recommend taking if someone has installed with libuvc backend and the camera still isn't detected (and if their camera model is suitable for the downgrade), as taking the additional step of the firmware downgrade resolves the detection problem for libuvc backend installations in most cases.

The firmware driver is stored inside the camera hardware and not on the computer, so you can perform the firmware update on a Windows PC or non-Jetson Linux computer and the firmware will be carried back to the Jetson inside the camera.

MartyG-RealSense, 2025-08-14:
That's great to hear that 5.13.0.50 worked for you too - thanks very much for the update about your success!