realsenseai/librealsense #14171: Realsense D-455 refuses to connect over USB

perseusdg, 2025-07-28:
I am unsure if this is the right platform for this , but my lab got a realsense D455 from amazon(us) sometime in late feb/early march ,but recently the realsense refuses to connect over USB at all (either usb 2 or 3) and sometimes it doesn't even show up in lsusb. 
In the offchance it works , we have to physically hold the realsense at odd angles and ensure it doesn't move at all otherwise the usb connection fails and we likely won't be able to reconnect with it. 

I tried contacting intel support over phone but I was told for realsense products I had to fill out a support form - the link which was provided to me doesn't work.

Is there a way for me to get in touch with someone at the support team of realsense and figure out how to get the realsense fixed since I believe the realsense is under warranty still.

perseusdg, 2025-07-29:
Hi @MartyG-RealSense , Thanks for the prompt response! 

With regards to the USB connection problem - I have been using the official cable that came with the box. I did try the realsense on multiple laptops and servers and the connection issue is pretty consistent. 

I also tried it with other cables and none of them seem to work.

perseusdg, 2025-07-29:
I believe it was the latest firmware, I remember updating to the version suggested by the realsense-viewer - I tried connecting the device now , but doesn't seem like it is responding at all

MartyG-RealSense, 2025-07-30:
In the last 2 / 3 weeks, has the camera been stored unused in a place like a cupboard or desk drawer before you tried connecting it again now and finding that it was not responding?

perseusdg, 2025-07-31:
nope , it was always on top of my desk and I believe i did use it once or twice in between

MartyG-RealSense, 2025-07-31:
I will discuss your case with my RealSense colleagues.  Thanks very much for your patience.

closed 2025-08-03 by perseusdg