realsenseai/librealsense #14182: Obtaining RAW image from Realsense D405

ShutongJIN, 2025-07-30:
Hi, I need access to RAW images from the RealSense D405. Since the RGB sensor used is the OV9782, it should theoretically support RAW10 output according to its documentation. I’ve previously obtained RAW16 from the D435 using a specific firmware.
Is there any firmware version for the D405 that also enables RAW10 output?

ShutongJIN, 2025-07-31:
Thanks for the reply. I'm getting a bit confused, isn't OV9782 used in D405 a color sensor? https://www.ovt.com/products/ov9782/

MartyG-RealSense, 2025-07-31:
The particular way in which RealSense cameras implement the sensor in its design enables it to be used for depth and infrared too.

The infrared image essentially resembles a monochrome RGB image.  And some RealSense models (including D405) are able to provide an alternative RGB image to the main one by setting the Infrared stream to RGB8 or BGR8 format instead of infrared's default Y8 format.  So thinking about it like that makes it easier to see how a sensor originally designed by its manufacturer for RGB could be adapted for other purposes.

ShutongJIN, 2025-07-31:
Thanks, is it possible to provide a firmware that supports output in RAW10 for D405?

ShutongJIN, 2025-07-31:
Thanks a lot!

closed 2025-08-14 by MartyG-RealSense