realsenseai/librealsense #14184: Introducing the RealSense D555 PoE Ethernet Camera

MartyG-RealSense, 2025-08-01:
Hi everyone,

The new generation of RealSense stereo depth camera, the **D555**, is now listed on the official RealSense Store for registering of notification of when it becomes available, and its data sheet document is available for download.

**RealSense Store**
https://store.realsenseai.com/buy-intel-realsense-depth-camera-d555.html

**Product information**
https://realsenseai.com/ruggedized-industrial-stereo-depth/d555-poe/

**Datasheet**
https://dev.realsenseai.com/docs/intel-realsense-d400-series-product-family-datasheet

Key features of D555 include 

-  The long depth sensing range of the D455 model thanks to its use of the D450 depth module
-  Power over Ethernet (PoE) connectivity
-  New-generation ASIC, the RealSense Vision SoC V5, whose features include RGB enhancement with Geometric Distortion Correction (GDC) and Temporal Noise Reduction (TNR)
-  Plug-and-play integration with ROS2 via Data Distribution Service (DDS)
-  Protective IP65 casing that is resistant to entry of dust and projected water jets
- IMU

ROliveira-lab, 2025-08-08:
Already out of stock 
There is a huge demand/curiosity for these cameras.

closed 2025-08-17 by MartyG-RealSense

d4n13l-k, 2025-08-28:
Hi @MartyG-RealSense 

Is there a 3D model of the module?

MartyG-RealSense, 2025-08-28:
Hi @d4n13l-k  Although there is not yet a CAD file available for the D555 or the new  RealSense Vision SoC V5ASIC board at the time of writing this, the D450 depth camera module does have a CAD file in .SLDPRT format.  

If you visit the link below and click on the **D400 Depth Modules** link, a download of a zip file called D400_Module.zip will begin in your browser.  When the contents of the zip file are extracted you can obtain the D450 CAD file 'D450_Module.SLDPRT'.

[image]

tovtpr, 2025-08-29:
Does anyone know if there if it's available anywhere other than the official Realsense online store ? To ship it to Europe currently adds a staggering 160$ to the 499$ base price ...

MartyG-RealSense, 2025-08-29:
Hi @tovtpr  Whilst the official RealSense Store is currently the only supplier for the D555, other retailers do stock our cameras after the launch period.  

A highly reliable European stockist of the RealSense product range is the Germany-based Rutronik24.

https://www.rutronik24.com/search-result/qs:Realsense/

They also have a United Kingdom version of their site.

https://www.rutronik24.co.uk/search-result/qs:Realsense/

Another leading European RealSense product stockist, also a German company, is FRAMOS.  Their RealSense page at the link below states that they will be carrying D555 in their product portfolio.

https://framos.com/product-catalog/3d-depth-sensing/intel-realsense/

v-lopez, 2025-09-10:
Yeah, but I am looking for the equivalent of this combo:
https://www.rutronik24.com/product/realsense/999wcm/27778206.html
https://www.rutronik24.co.uk/product/realsense/952019/10164045.html

So I can build my own board + sensor integration, with different mechanical design.

MartyG-RealSense, 2025-09-10:
@v-lopez  The D450 module is of course easily available.  The new **Vision SoC V5** ASIC board is not yet sold separately though and I do not have information at the time of writing this about when that board will be available for individual purchase.

The V5 that you mentioned is the Vision Processor D4 V5 for use with the D450 to create the equivalent of a D457 GMSL / Fakra camera.  Only the Vision SoC V5 board has the PoE ethernet connector.

MartyG-RealSense, 2025-09-10:
Hi @jrecasens  D555 has the same 87° × 58° depth field of view as the D455 as, like the D455, it has the D450 depth camera module inside it.  So the baseline distance between the left and right stereo sensors will not have changed.

D555 appears to have a larger, wider heatsink radiator component on the back of the D450 module compared to the smaller D456 that helps to account for the larger casing size.  The heatsink draws heat away from the circuit boards to help keep them cool.  The larger the heatsink, the more effective they are.

**D555 heatsink**

[image]

**D456 heatsink**

[image]

MartyG-RealSense, 2025-09-15:
@subbubhat99  D555 should be able to work fine with Jetson boards.

One of my RealSense colleagues at https://github.com/IntelRealSense/librealsense/issues/14281#issuecomment-3283824946 offers advice for a problem with displaying streams with D555.

subbubhat99, 2025-09-15:
@MartyG-RealSense Yes your reference to #14281 really helped to now echo all topics with continous data streaming that were previously not returning anything (except for the topic /realsense/D555_344522302142_Motion). 

The problem still persists with the D555 not being detected when I run rs-enumerate-devices or through the rs_launch.py file of the SDK. What could the problem then be? I would like to able to see the depth-map, point-cloud, RGB-images, and IMU movements in the realsense-viewer but can't right now.

NOTE: I initially installed the SDK using the native backend. After reading up on one of your comments in #13892 to install the libuvc backend built from source since that has proven to work for some realsense cameras with the Jetson nano, I also followed that. Could that have lead to a dual backend problem/mixup?

MartyG-RealSense, 2025-09-15:
With the ROS wrapper, D555 is supported in wrapper version **4.56.4** or newer.

The IMU topics of the ROS wrapper are disabled by default.  Are they published if you enable them in your ros2 launch instruction?

`ros2 launch realsense2_camera rs_launch.py enable_accel:=true enable_gyro:=true unite_imu_method:=2`

With rs-enumerate-devices, is the D555 detected if you launch the program in **sudo** admin permissions mode with `sudo rs-enumerate-devices`

I would always highly recommend uninstalling previous installations of the librealsense SDK before installing a new one in order to avoid the possibility of conflicts between two separate SDK installations on the same computer.

subbubhat99, 2025-09-17:
@MartyG-RealSense Even if I run the launch file with the gyro, accel, and unite_imu_method parameters enabled, since the camera cannot be found, I don't see them. But what's strange is that the camera is already publishing to 5 topics. And they're all echoing and I could visualize them in Foxglove as well!

But rs-enumerate-devices or sudo-rs-enumerate-devices returns: 

dev@0001-nano:~$ sudo rs-enumerate-devices
[sudo] password for dev: 
 17/09 14:12:20,790 ERROR [281473507584064] (context.cpp:40) No valid configuration file found at : /root/.realsense-config.json loading defaults
No device detected. Is it plugged in?

I am a bit skeptical to remove anything since the camera does publish its topics and I am getting the depth map, color-RGB image, the 2 IR and the IMU (Motion) inputs just fine. But I know there are 2 backends installed i.e. the native and the libuvc backend built from source. A bit confused how to proceed right now.

subbubhat99, 2025-09-17:
For now, the depth camera should only be needed for integrating obstacle avoidance into our nav_stack, and for running visual odometry to facilitate navigation in GNSS-denied areas when our UBLOX-DGNSS module may not be usable for localization.

It would of course have been ideal to get rs-enumerate-devices to detect it. Also the realsense-viewer GUI didn't detect it so that prevents us from visualizing point clouds. Is there an alternative to quick p.cl visualization?

subbubhat99, 2025-09-17:
@MartyG-RealSense Tried both these but returned that no device is connected. The rs-sensor-control detected an RGB platform camera instead. rs-pointcloud opens the viewer GUI but then shuts down and returns that no device is connected.

RasmusGrift96, 2025-11-06:
> Hi [@RasmusGrift96](https://github.com/RasmusGrift96) Other D555 users have reported that LEDs do not illuminate even when the camera is operating normally, so this is not necessarily an indication that there is a problem.
> 
> To use the camera in ethernet mode, you need to make sure that the USB cable is disconnected from the camera.

@MartyG-RealSense I do not have the USB cable connected. I was expecting the RealSense to appear in the Intel RealSense Viewer (v2.56.5) or to be visible on my network using sudo arp-scan -l. I also tried querying all available devices using the RealSense SDK, but none of these methods detect the device.

How can I verify if the RealSense is available for use?