realsenseai/librealsense #14207: No RealSense devices were found  D455

Frank-mirle, 2025-08-13:
### Hardware
NVIDIA Jetson NX
RealSense cameras: D435 and D455

### Software
Ubuntu 22.04
ROS2 Humble
JetPack 36.4.4

RealSense ROS v4.56.0
Built with Librealsense v2.57.2
Running with Librealsense v2.57.2

### Issue
The D435 camera can display images through RealSense Viewer or RViz, but when I use the D455, the device cannot be detected. Running rs-enumerate-devices also cannot find the D455. However, running lsusb can detect the D455.

1. How can I resolve this issue?
2. How can I use both the D435 and D455 cameras simultaneously?

詢問 ChatGPT

MartyG-RealSense, 2025-08-13:
Hi @Frank-mirle  36.4.4 is part of JetPack 6.2.1.  RealSense camera models equipped with an IMU, such as D455, can have problems with being detected when using JetPack 6.  Non-IMU models such as D435 are unaffected.  https://github.com/IntelRealSense/librealsense/issues/14169#issuecomment-3127182150 has more information about this and a solution that can work.

In regard to using D455 and D435 simultaneously in ROS2 once both cameras are detectable, the RealSense ROS wrapper provides a two-camera launch file called rs_multi_camera_launch.py

https://github.com/IntelRealSense/realsense-ros/blob/ros2-master/realsense2_camera/launch/rs_multi_camera_launch.py

Frank-mirle, 2025-08-13:
Hi @MartyG-RealSense 
When I followed these steps, an error occurred.

1. Make sure no RealSense device is connected
2. Open the terminal, run:
```
$ wget https://github.com/IntelRealSense/librealsense/raw/master/scripts/libuvc_installation.sh
$ chmod +x ./libuvc_installation.sh
$ ./libuvc_installation.sh
```

```
Reading package lists... Done                                     
N: Skipping acquire of configured file 'main/binary-arm64/Packages' as repository 'http://dl.google.com/linux/chrome/deb stable InRelease' doesn't support architecture 'arm64'
E: Failed to fetch http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo/dists/jammy/InRelease  403  Forbidden [IP: 52.218.45.129 80]
E: The repository 'http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo jammy InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
E: Failed to fetch http://realsense.intel.com/Debian/apt-repo/dists/jammy/InRelease  403  Forbidden [IP: 2001:b034:1:23a4::1ea2 443]
E: The repository 'http://realsense.intel.com/Debian/apt-repo jammy InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
W: https://librealsense.intel.com/Debian/apt-repo/dists/jammy/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
```

MartyG-RealSense, 2025-08-15:
For your D455 to be detected, its firmware driver may need to be downgraded to version **5.13.0.50**.  This enables a D455 to be successfully detected in most cases on a Jetson where a libuvc installation has been performed but the D455 still isn't detected afterwards.

If you have access to a Windows PC or a non-Jetson Linux PC then the camera should be detectable on that so that the firmware change can be performed, as the non-detection is related to JetPack 6 on Jetson boards specifically.  PCs are not affected. The firmware is stored inside the camera hardware and not on the computer, so after updating the firmware to version 5.13.0.50 on a PC the camera can be taken back to the Jetson, where the D455 should then be detectable.

Frank-mirle, 2025-08-21:
Hi @MartyG-RealSense Thank you for your help. After downgrading the firmware driver to version 5.13.0.50, the RealSense Viewer was able to detect the D455.

MartyG-RealSense, 2025-08-21:
You are very welcome, @Frank-mirle - I'm pleased to hear that the firmware downgrade worked for you.  Thanks very much for the update!