realsenseai/librealsense #14210: Android Project Error：[rs2_pipeline_wait_for_frames( pipe:0xb4000073b8b32110 ) UNKNOWN] Frame didn't arrive within 5000

naiyi, 2025-08-13:
### Hardware
Camera: D415
Firmware: 5.17.0.10

### Android Device
Brand: HW Honor
Android version:15

### Software
RealSense SDK version: v 2.56.5
Example Project: examples.capture

### Description

```
2025-08-13 20:37:03.286 22886-22983 alsense.capture         com.intel.realsense.capture          E  open file error
2025-08-13 20:37:03.286 22886-22983 alsense.capture         com.intel.realsense.capture          E  open file error
2025-08-13 20:37:03.286 22886-22983 QT                      com.intel.realsense.capture          E  [QT]file does not exist
2025-08-13 20:37:03.287 22886-22983 QT                      com.intel.realsense.capture          E  [QT]file does not exist
2025-08-13 20:37:05.456 22886-22886 librs                   com.intel.realsense.capture          E  No valid configuration file found at : /data/.realsense-config.json loading defaults
2025-08-13 20:37:05.541 22886-22886 librs                   com.intel.realsense.capture          E  No valid configuration file found at : /data/.realsense-config.json loading defaults
2025-08-13 20:37:07.387 22886-22886 [FASProxy:]             com.intel.realsense.capture          E  failed to open /proc/fas/ioctl, errno=2, No such file or directory
2025-08-13 20:37:07.519 22886-22886 RtgSchedIpcFile         com.intel.realsense.capture          E  RtgSchedIpcFile xxx failed to open /proc/22886/rtg
2025-08-13 20:37:07.519 22886-22886 RtgSchedIpcFile         com.intel.realsense.capture          E  RtgSchedIpcFile xxx failed to open /proc/rtg_gki/ioctl.
2025-08-13 20:37:07.519 22886-22886 RtgSchedIpcFile         com.intel.realsense.capture          E  RtgSchedIpcFile xxx failed to open /proc/22886/rtg
2025-08-13 20:37:07.519 22886-22886 RtgSchedIpcFile         com.intel.realsense.capture          E  RtgSchedIpcFile xxx failed to open /proc/rtg_gki/ioctl.
2025-08-13 20:37:07.519 22886-22886 RtgSchedIpcFile         com.intel.realsense.capture          E  RtgSchedIpcFile xxx failed to open /proc/22886/rtg
2025-08-13 20:37:07.519 22886-22886 RtgSchedIpcFile         com.intel.realsense.capture          E  RtgSchedIpcFile xxx failed to open /proc/rtg_gki/ioctl.
2025-08-13 20:37:07.519 22886-22886 RtgSchedIpcFile         com.intel.realsense.capture          E  setCommandByIoctl not opened
2025-08-13 20:37:07.521 22886-22886 RtgSchedIpcFile         com.intel.realsense.capture          E  RtgSchedIpcFile xxx failed to open /proc/22886/rtg
2025-08-13 20:37:07.521 22886-22886 RtgSchedIpcFile         com.intel.realsense.capture          E  RtgSchedIpcFile xxx failed to open /proc/rtg_gki/ioctl.
2025-08-13 20:37:07.521 22886-22886 RtgSchedIpcFile         com.intel.realsense.capture          E  RtgSchedIpcFile xxx failed to open /proc/22886/rtg
2025-08-13 20:37:07.521 22886-22886 RtgSchedIpcFile         com.intel.realsense.capture          E  RtgSchedIpcFile xxx failed to open /proc/rtg_gki/ioctl.
2025-08-13 20:37:07.521 22886-22886 RtgSchedIpcFile         com.intel.realsense.capture          E  setCommandByIoctl not opened
2025-08-13 20:37:12.522 22886-22886 librs                   com.intel.realsense.capture          E  [rs2_pipeline_wait_for_frames( pipe:0xb4000073b8b2db30 ) UNKNOWN] Frame didn't arrive within 5000
2025-08-13 20:37:12.522 22886-22886 librs capture example   com.intel.realsense.capture          E  streaming, error: Frame didn't arrive within 5000
2025-08-13 20:37:12.918 22886-22886 FBI                     com.intel.realsense.capture          E  Can't load library: dlopen failed: library "libmagtsync.so" not found

```

naiyi, 2025-08-13:
I tried to switch to the 'development' branch and rebuild the project. Unfortunately, the same errors still persist. # @MartyG-RealSense @

naiyi, 2025-08-13:
Anyway, Thank you for your guidance. This issue can be closed now. @MartyG-RealSense

MartyG-RealSense, 2025-08-13:
You are very welcome.  I'm pleased to hear that you had a phone with an Android version that worked with the application. 

As you suggested, I will close the issue.  Thanks again, @naiyi

closed 2025-08-13 by MartyG-RealSense