realsenseai/librealsense #14211: Creating Pseudo RGB HDR with two manual exposure values; results are slow

JHershberger-GK, 2025-08-13:
**Camera Hardware:** D435e FRAMOS
**Camera Image Info:** 30fps 1280x720 for both Depth and RGB
**Librealsense Version:** FRAMOS Modified 2.55.1 (I'm aware of the potential issues of a modified library not acting as expected with the OG)
**Computer Hardware:** Jetson Orin NX 16GB
**Computer OS:** Jetson Jetpack 6.1
**Prog Language:** C++17

**Background information:**
With the current project I'm working on, i'm dealing with bright daylight and shadows created from the machine frame. I'm using the camera for visual processing to collect target locations. Since the RGB sensor doesn't have any HDR like capabilities, I needed to make a pseudo version of it so that way I can find targets in two extreme lights (i'm really trying to avoid adding shading to the entire machine frame). My plan is to get the RGB sensors' exposures for very bright areas and very dark areas and then have each frame switch between those exposures (Camera uses 30 fps, i'm doing 4 fps). 
I have already collected the exposure values needed; a guessing game with ROI Auto exposure and comparing ROI pixels brightness with a manually set exposure throughout a range.

**The Problem:**
The problem is that when I do the 'hotswap', the frames are inconsistent. Either every wait_for_frames() frame i'm hot swapping, or every 2 wait_for_frames() frames. A reminder that these frames are being called rather slowly, 4 frames per second are being wait_for_frames() where as the camera is actually targeting 30 frames normally.

Barely sometimes the pictures are taken consistently as expected, but majority of the time it is stuck with one exposure for **seconds** before swapping to the next then sometimes instantly swapping back.

Down below is the code that does hot swap. the DarkExposure and LightExposure are just floats that contains the appropriate values for the RGB sensor
```c++
     . . . . . . . . . . . . .
    color_sensor.set_option(RS2_OPTION_ENABLE_AUTO_EXPOSURE, false);
    color_sensor.set_option(RS2_OPTION_EXPOSURE, DarkExposure);
}

//in a seperate class function . . .
{
    if(this->HotSwap){ //Value determines whether its turned on or off
        //Exposure hotswap
        static int counter{0}; //To make it so that it takes two frames before manually swapping exposure
        if(currentExposure != DarkExposure && counter < 2){
            currentExposure = DarkExposure; //DarkExposure is used for darkest areas
            color_sensor.set_option(RS2_OPTION_EXPOSURE, currentExposure);
            
            counter++;
        }
        else{
            currentExposure = BrightExposure; //BrightExposure is used for brightest areas
            color_sensor.set_option(RS2_OPTION_EXPOSURE, currentExposure);
            
            counter++;
            if(counter > 3) counter = 0;  //Resets the counter so that it now does dark areas
        }
    }
    //This is a custom timer that puts the thread to sleep for so many milliseconds. Right now, jetson->getFPS() is set to 4
    //frameTimer calles .finish() at the end of the function. This makes it wait out the rest of the timer 
    //so that way it should hopefully always be that targeted framerate
    Rtimer frameTimer(int(1000 / jetson->getFPS()));  

    rs2::frameset frameSet = cameraPipe.wait_for_frames(5000); //Pulls both Depth and RGB frames
    frameSet = align_to_color->process(frameSet); //Aligns the Depth to RGB for coordination

    /*Remainder of the function processes visual targeting*/
}
```

The questions I have: 
- Is there a better way to do this to get faster/consistent exposure results? 
- Is there a better way to setting the RGB sensor's exposure?
- Is there something i'm not understanding properly how frames work?
- Am I expecting too much out of the camera?

MartyG-RealSense, 2025-08-14:
Hi @JHershberger-GK  Thanks very much for your questions.  I'm pleased to hear that you were able to achieve a code modification that seemed to help with the inconsistency.

We do not provide technical support for the RealSense-compatible 'e' industrial cameras by FRAMOS unfortunately as they are not manufactured by us.  FRAMOS should be contacted on their website for support advice.

https://framos.com/resources/support/

In general though, if you have RGB issues relating to light and dark areas in the same scene then you could try fixing a physical ND (Neutral Density) filter product over the RGB sensor on the outside of the camera to reduce the amount of light that enters the sensor - please see https://github.com/IntelRealSense/librealsense/issues/7155#issuecomment-691356656 for more information.

JHershberger-GK, 2025-08-14:
@MartyG-RealSense 
Okay, I'll look into the ND filter. I was hoping to at least get a bit of insight on some of the functions related to exposure timings. I know it is a modified library but i'm hoping that some insight on how setting exposure works on the official product. That way i have an idea what i should be looking for from the FRAMOS product.

Just suggestions is all i need, i'm not expecting full on answers because i know they won't be available for a 3rd party products. If still a no, then let me know and feel free to close the issue.

JHershberger-GK, 2025-08-15:
**Answer this first:** So by official intel hardware, should that recent manual exposure value represent the last value the Auto Exposure was used for the RGB sensor, or the last time user input an exposure? After you turn off auto exposure of course. 

I'm wanting to condense a bunch of the RGB Exposure problems and suggestions so that way it's easier for people to find. I've spent far too many hours looking around to get a form of "not currently possible".

On FRAMOS Cameras, no metadata related to exposure or gain gets sent back from RGB, and neither does Auto Exposure set the software's EXPOSURE value when it gets deactivated. Also after contacting them about this issue they have said it wasn't a planned feature to send back the EXPOSURE value; neither was it possible due to hardware limitations.

### Issue pretty much solved:
Currently it looks like after having the camera run at 30 frames with the listed resolution and placing them in a frameset queue with a capacity of 1 to constantly drop unused frames, then pulling the frames at 6 fps; with these settings, swapping the manual exposure every 3 pulled frames works consistently for the projects purpose.

### Documentation for future individuals, making it easier to find:
**Keywords:** Manual Exposure, RGB HDR, RGB Exposure, Exposure Value, Auto Exposure

For trying to get that exposure value; it is a bit hacky, isn't really efficient, and could definitely be better. The process is as follows:
- Gather the color_sensor as an roi_sensor
- Set the roi on the roi_sensor to either your darkest area(shadows) or brightest area(daylight)
- Get the auto exposure over a second or two; hold that last frame for use, convert the last auto exposure(AE) color frame into HLS with openCV (do not modify, you need this as your reference for finding the exposure)
- Turn off auto exposure and begin the loop to "guessing" the manual exposure by comparing images. 
-   Decide on a range of Manual Exposure(ME) values to try and guess, (i'm doing a form of divide and conquer, so I take halves of my min and max, which get changed if my error rate is better; until i reach my tolerated brightness error)
-   Set the roi_sensor to the brightest region or darkest region
-   Set the manual exposure value, pull half your camera's frame rate, convert the last frame collected into HLS as well
-   Iterate through each pixel that is in your ROI; with your ME and your AE pictures, get the difference of their luminosity values H(L)S and add them up to a total, compare this ABS(total) with a tolerance. if it doesn't reach under the tolerance, then change your manual exposure to the next one and repeat. {Keep in mind, this doesn't work well with blacks and whites. But for the purpose of this project; I don't need to worry about that}

One you've collected the manual exposure value that closely matches the auto exposure's picture, you can use that value for what light level you have. Do this again for the opposite side of the brightness and then you can manually swap between those exposures. If you're dealing with the sun or changing lights, you will need to periodically rerun this.

Indoor and clouded day values, you're expecting roughly 100-250 units.  Sunny days are 10-60 units. For dark shadows produced 450-700 units. FRAMOS uses 100 microseconds per unit for exposure times on the RGB sensor; not sure if the official intel hardware does too. 
Depth sensors I'm not worried about setting manual exposure as they set each pixel's value as millimeters for distance from camera. I needed the RGB photos for visual software to detect my targets.

Like I said, there is probably a better way to compare the images and have a range that could give you an accurate exposure value, but I need fast and approximate.

**TL;DR**
Get a picture of your auto exposure, then get a picture of some manual exposure value you set. Compare the two images, change the manual exposure and repeat until you're satisfied. That is the RGB's "auto exposure" value. For my jetson it takes roughly 5-10 seconds to find that exposure value but it's possibly inefficient code for majority of it. Most of the time is spent getting the auto exposure image and and then getting your manual exposure image.

MartyG-RealSense, 2025-08-16:
If a manual RGB value has not been input by the user then the official camera would use the default RGB manual exposure value when auto-exposure is disabled.

The official cameras are no longer Intel hardware as the RealSense division has, as of July 11 2025, spun off from Intel to become its own separate company. 

https://realsenseai.com/

The first new camera product released after the spin-off is the RealSense D555, which has PoE ethernet connectivity.

https://realsenseai.com/ruggedized-industrial-stereo-depth/d555-poe/

Thanks so much for sharing your solution in such detail for the benefit of the RealSense community!

JHershberger-GK, 2025-08-22:
Closing issue since it's basically been resolved

TIP: Make sure that your exposure value never exceeds the range of RS2_OPTION_EXPOSURE. Or else you might for example get the following error: out of range value for argument "0"

I kept getting it randomly, didn't realize it was realsense throwing it, it caused a bit of pain trying to figure out why.