realsenseai/librealsense #14214: D415 Manual Exposure Wraparound

dgcole, 2025-08-14:
### HW / SW
- OS: Ubuntu 20.04
- Camera: D415
- Camera FW Version: 5.15.0.2
- librealsense2 Version: 2.55.1

### Problem

I am experiencing some weird exposure behavior in the color images returned from a D415 camera. When the camera initializes, pick the highest exposure value (lets call it X) for which the RGB image is almost completely black. This value seems to change every time the camera is reset.

Here is an example where X is 343. The solid white sections are not part of the image, they have been added to obscure the background. The RGB values in the bottom left correspond to the color of the area behind the central grid. The RS2_* values come from calling `get_option_range` / `get_option`. The metadata values come from calling `get_frame_metadata` on the color frame in question. First, an image where the exposure is 343:

[image]

```
INFO:  RS2_OPTION_EXPOSURE Range: [1.000000, 10000.000000] step 1.000000 def 166.000000
INFO:  RS2_OPTION_EXPOSURE: 343.000000
INFO:  RS2_OPTION_ENABLE_AUTO_EXPOSURE: 0.000000
INFO:  Color metadata auto exposure: 0
INFO:  Color metadata exposure: 343
INFO:  Depth metadata auto exposure: 1
INFO:  Depth metadata exposure: 59514
```

Next, if the lower the exposure value to X-1, 342 in this case, we get this clearly incorrect image:
[image]
```
INFO:  RS2_OPTION_EXPOSURE Range: [1.000000, 10000.000000] step 1.000000 def 166.000000
INFO:  RS2_OPTION_EXPOSURE: 342.000000
INFO:  RS2_OPTION_ENABLE_AUTO_EXPOSURE: 0.000000
INFO:  Color metadata auto exposure: 0
INFO:  Color metadata exposure: 342
INFO:  Depth metadata auto exposure: 1
INFO:  Depth metadata exposure: 59514
```

I would expect the second image to be darker than, or at least equally dark (if the RGB values can't go any lower) as, the first.

MartyG-RealSense, 2025-08-15:
Hi @dgcole  Is the black RGB frame captured before or after auto-exposure is set to 0 to disable it?

When the camera starts streaming, if auto-exposure is enabled then the first several frames may have incorrect exposure because it takes that time for auto-exposure to settle down.  This can be avoided by adding an instruction to a script to skip the first several frames before the first frame capture is taken.  

There is no need to perform a frame skip if a manual exposure value has been set (doing so automatically disables auto-exposure) before the first frame capture is taken.

dgcole, 2025-08-15:
Thank you for the prompt response. The black RGB frame was captured after `RS2_OPTION_ENABLE_AUTO_EXPOSURE` was set to 0. Auto exposure has always been off for me when encountering this issue.

With regards to the frame timing, this is not just happening in the first few frames. The images in the issue description were captured after around 15 minutes of constant streaming at 15fps.

MartyG-RealSense, 2025-08-15:
Typically when RGB exposure is manually increased above the default of 156, the image brightens until it becomes white, not darken to black.  A black RGB image can be produced by a minimum exposure of 1.

How does your program behave if you set RGB exposure as 312 instead of 342 / 343 please?

dgcole, 2025-08-19:
Using firmware version `5.16.0.1` did not fix the weird exposure behavior I am seeing. I do think I have narrowed down the circumstances that cause the issue:
* Scenario A (**CAUSES ISSUE**): Enable device -> Set auto exposure to false -> Set manual exposure
* Scenario B (**DOES NOT CAUSE ISSUE**): Enable device -> Set auto exposure to true -> Some time later, set auto exposure to false -> Set manual exposure

I will try to create a minimal reproduction of the issue to share.

dgcole, 2025-08-19:
> Auto-exposure will automatically disable when a manual exposure value is set. So what happens if the manual exposure value is set first, so that when the device enables, the auto-exposure disables without you having to tell it to disable with an instruction.

I also tried not explicitly setting `RS2_OPTION_ENABLE_AUTO_EXPOSURE` to off when setting `RS2_OPTION_EXPOSURE`, and that did not make a difference. I have decided to temporarily work around this issue by always leaving auto exposure on after enabling the camera, capturing 1s of frames, and then turning auto exposure back off. This seems to eliminate the problem for me.

That being said, this does seem like a bug with either `librealsense2` or the D415 firmware, as I would expect to be able to set manual exposure immediately without first capturing any frames.

MartyG-RealSense, 2025-08-19:
I'm pleased to hear that you achieved a solution.  Thanks very much for the update and for providing details of your solution!

If your script uses the **wait_for_frames()** instruction then you could try placing the manual exposure value configuration line after the pipeline start line but before the wait_for_frames() line, as the program should not start capturing frames until it reaches the wait_for_frames() line.

MartyG-RealSense, 2025-08-22:
BTW, the .bin file download links on the firmware releases page, including 5.16.0.1, have now been restored.  Thanks for highlighting that!

dgcole, 2025-08-22:
> If your script uses the **wait_for_frames()** instruction then you could try placing the manual exposure value configuration line after the pipeline start line but before the wait_for_frames() line, as the program should not start capturing frames until it reaches the wait_for_frames() line.

Initially I was configuring the exposure both before starting the pipeline and before waiting for any frames. My workaround mentioned above was to move the manual exposure `set_option` call to happen after waiting for a few frames.

Per your suggestion, I also tried moving my `set_option` call to be between starting the pipeline and waiting for frames, but that still seems to trigger this bug. The only order I have found that avoids hitting this bug seems to be waiting a few frames before calling `set_option` to set the manual exposure.

MartyG-RealSense, 2025-08-22:
Which method did you use to install the librealsense SDK, please?  If it was built from source code with the libuvc backend or RSUSB backend installation methods then calls to SET_OPTION instructions have a higher risk of being missed by the SDK.  Using 'native' installation methods such as DKMS packages or building from source with libuvc / RSUSB false and applying a kernel patch can provide faster response times by the SDK to SET_OPTION instructions, reducing the risk of them being missed and not activated.

dgcole, 2025-08-22:
I am using a from-source build with the rsusb back-end disabeld, here are my CMake arguments:
```
cmake -DCMAKE_BUILD_TYPE=RELEASE \
	-D CMAKE_INSTALL_PREFIX=../../install/usr/local \
	-D FORCE_RSUSB_BACKEND=false \
	-D CMAKE_BUILD_TYPE=release \
	-D BUILD_EXAMPLES=false \
	-D BUILD_GRAPHICAL_EXAMPLES=false \
	-D BUILD_PYTHON_BINDINGS=false
```

MartyG-RealSense, 2025-08-22:
Thanks very much.  Have you applied the librealsense patch script to your kernel, please?  The patch script used will depend on the kernel version that you are using with Ubuntu 20.04.

Kernel 5.15, 5.19, 6.5 - **./scripts/patch-realsense-ubuntu-lts-hwe.sh**

https://github.com/IntelRealSense/librealsense/blob/master/scripts/patch-realsense-ubuntu-lts-hwe.sh

Kernel 5.13 or older - **./scripts/patch-realsense-ubuntu-lts.sh**
https://github.com/IntelRealSense/librealsense/blob/master/scripts/patch-realsense-ubuntu-lts.sh