realsenseai/librealsense #14240: realsense-viewer: Cannot access /sys/class/d4xx-class

coollingomg, 2025-08-25:
|---------------------------------|------------------------------------------- |
| Camera Model | D435i |
| Operating System & Version | Ubuntu 22.04|
| device | nvida jetson orin nx super 16GB |
| SDK Version | v2.56.2 |

After installation of SDK from packages according to
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md
realsense-viewer does not connect to the camera and prints the following error to the console
(backend-v412.cpp:675) Cannot access /sys/class/d4xx-class

MartyG-RealSense, 2025-08-25:
Hi @coollingomg  There is another case of this error message at https://github.com/IntelRealSense/librealsense/issues/14169 where installing the librealsense SDK from source code with the **libuvc backend** installation method and then dowgrading the camera firmware driver to version **5.13.0.50** resolved the problem.

On Jetsons that have JetPack 6 installed and are using a RealSense camera model equipped with an IMU (like your D435i), the libuvc backend and the 5.13.0.50 are the solution that enables the camera to be detected in most cases.

coollingomg, 2025-08-25:
Thanks for such a quick reply, but after I tried it, I still have the problem that it is not readable.

Here's how I do it:
First, I uninstalled the pass using "sudo apt-get autoremove librealsense"
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md the software installed in Debian Packages.

And then I go through what you mentioned
https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md built librealsense. But this still doesn't make me detect the camera. After I entered the "rs-enumerate-devices" command, my feedback was: "No device detected. Is it plugged in?”。 The same goes for using "realsense-viewer".

Subsequently, I downgraded to 5.13.0.50 on my Windows PC through the firmware you mentioned. After repeatedly entering the "rs-enumerate-devices" command, my feedback was still "No device detected. Is it plugged in?”.

I looked at my USB device via "lsusb" which showed the camera device "Bus 002 Device 021: ID 8086:0b3a Intel Corp. Intel(R) RealSense(TM) Depth Camera 435D”

At the moment, I don't know what to use to make my camera work properly on the Jetson NX.

I would be very happy if you could guide me on how to use my camera on the Jetson NX normally!

Thank you!

MartyG-RealSense, 2025-08-25:
I would recommend uninstalling RealSense SDK packages using the command below:

`dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge`

Is the camera detected if you launch programs using the **sudo** admin permissions mode?  For example:

`sudo realsense-viewer`

`sudo rs-enumerate-devices`

MartyG-RealSense, 2025-08-25:
Does anything change if you unplug the micro-sized end of the USB cable from the side of the camera, turn the connector around the other way and re-insert it into the side of the camera (USB-C cables are two-way insertion at the micro-sized end).

coollingomg, 2025-08-25:
I tried this operation, but nothing changed.